realsenseai/realsense-ros #1761: filters:=colorizer Not Working

yashvakil, 2021-03-17:
I am trying to run the following command:
`roslaunch realsense2_camera rs_camera.launch filters:=colorizer`

But the camera wont start stating the error:
``` diff
process[camera/realsense2_camera_manager-1]: started with pid [10059]
process[camera/realsense2_camera-2]: started with pid [10060]
[ INFO] [1616008562.201374762]: Initializing nodelet with 6 worker threads.
[ INFO] [1616008562.543015850]: RealSense ROS v2.2.22
[ INFO] [1616008562.543178218]: Built with LibRealSense v2.42.0
[ INFO] [1616008562.543280650]: Running with LibRealSense v2.42.0
[ INFO] [1616008562.609286090]:  
[ INFO] [1616008562.980004362]: Device with serial number 045322075034 was found.

[ INFO] [1616008562.980174122]: Device with physical ID 2-3.4-4 was found.
[ INFO] [1616008562.980251658]: Device with name Intel RealSense D435I was found.
[ INFO] [1616008562.982316618]: Device with port number 2-3.4 was found.
[ INFO] [1616008562.982436522]: Device USB type: 3.2
[ INFO] [1616008562.988902250]: getParameters...
[ INFO] [1616008563.151626282]: setupDevice...
[ INFO] [1616008563.151742218]: JSON file is not provided
[ INFO] [1616008563.151809386]: ROS Node Namespace: camera
[ INFO] [1616008563.151873162]: Device Name: Intel RealSense D435I
[ INFO] [1616008563.151967594]: Device Serial No: 045322075034
[ INFO] [1616008563.152028010]: Device physical port: 2-3.4-4
[ INFO] [1616008563.152127594]: Device FW version: 05.12.10.00
[ INFO] [1616008563.152237002]: Device Product ID: 0x0B3A
[ INFO] [1616008563.152327434]: Enable PointCloud: Off
[ INFO] [1616008563.152401610]: Align Depth: On
[ INFO] [1616008563.152482282]: Sync Mode: On
[ INFO] [1616008563.152663018]: Device Sensors: 
[ INFO] [1616008563.191850442]: Stereo Module was found.
[ INFO] [1616008563.242133898]: RGB Camera was found.
[ INFO] [1616008563.243991818]: Motion Module was found.
[ INFO] [1616008563.244183786]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1616008563.244299882]: Add Filter: colorizer
- OpenCV Error: Assertion failed (s >= 0) in setSize, file /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 307 terminate called after throwing an instance of 'cv::Exception'
-  what():  /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:307: error: (-215) s >= 0 in function setSize
- [camera/realsense2_camera_manager-1] process has died [pid 10059, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/rover/.ros/log/bf05d96c-8753-11eb-a3ce-1b64057f7c91/camera-realsense2_camera_manager-1.log].
- log file: /home/rover/.ros/log/bf05d96c-8753-11eb-a3ce-1b64057f7c91/camera-realsense2_camera_manager-1*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/rover/.ros/log/bf05d96c-8753-11eb-a3ce-1b64057f7c91/camera-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
```

doronhi, 2021-03-18:
It is fixed in #1714 and will be released in the next release.
In the meanwhile, specify images width, height, and fps manually to avoid the issue.
`roslaunch realsense2_camera rs_camera.launch filters:=colorizer depth_width:=848 depth_height:=480 depth_fps:=30`

MartyG-RealSense, 2021-03-18:
Thanks very much @doronhi for your advice to @yashvakil    :)

closed 2021-03-19 by yashvakil