realsenseai/realsense-ros #1763: Unable to get Infrared streams on D435i

Ravi3191, 2021-03-18:
I am  running a D435i on Ubuntu 18.04. I was initially able to get the infra streams for the camera but then suddenly they stopped streaming. When I run the launch file by enabling the infra nodes I get the following:

```
[ INFO] [1616039497.934390633]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1616039497.934411614]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
```
I am not sure why this is happening. I am using a USB 3 port and I am able to visualize rgb and also the depth streams. I tried reinstalling the librealsense2 and have also updated the firmware on the camera. I am attaching complete log from the launch file. 

Is there any other way to resolve the issue? Thanks!

```
$ roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true enable_infra:=true enable_infra_rgb:=true unite_imu_method:=linear_interpolation
... logging to /home/ravi/.ros/log/3d2dcf86-879d-11eb-a799-087190addca2/roslaunch-ravi-HP-Spectre-x360-Convertible-15-df1xxx-27516.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ravi-HP-Spectre-x360-Convertible-15-df1xxx:37447/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
[... 78 lines cut ...]
[ INFO] [1616039497.934390633]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1616039497.934411614]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1616039497.934440793]: num_filters: 0
[ INFO] [1616039497.934457172]: Setting Dynamic reconfig parameters.
[ INFO] [1616039498.331835250]: Done Setting Dynamic reconfig parameters.
[ INFO] [1616039498.332194789]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1616039498.332982777]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1616039498.333005736]: gyro stream is enabled - fps: 200
[ INFO] [1616039498.333017934]: accel stream is enabled - fps: 63
[ INFO] [1616039498.333037942]: setupPublishers...
[ INFO] [1616039498.334250196]: Expected frequency for depth = 30.00000
[ INFO] [1616039498.335812946]: Expected frequency for color = 30.00000
[ INFO] [1616039498.336551087]: Start publisher IMU
[ INFO] [1616039498.337097790]: setupStreams...
[ INFO] [1616039498.347515460]: insert Depth to Stereo Module
[ INFO] [1616039498.347553293]: insert Color to RGB Camera
[ INFO] [1616039498.347569495]: insert Gyro to Motion Module
[ INFO] [1616039498.347584861]: insert Accel to Motion Module
[ WARN] [1616039498.361215162]: 
[ WARN] [1616039498.361263070]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 17/03 23:51:38,358 WARNING [140410916316928] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ INFO] [1616039498.409960559]: SELECTED BASE:Depth, 0
[ INFO] [1616039498.420888253]: RealSense Node Is Up!

```

doronhi, 2021-03-18:
The following line will enable the infrared streams:
`roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true enable_infra1:=true enable_infra2:=true unite_imu_method:=linear_interpolation infra_width:=848 infra_height:=480 infra_fps:=30`

In D435i, there is no infra_rgb option so no infrared stream with index 0. Only infra1 and infra2. So `enable_infra:=true` does not help. Only `enable_infra1:=true` and/or `enable_infra2:=true`. Also, there is no point in adding `enable_infra_rgb:=true`
As opposed to the color and depth streams, librealsense2 does not define a default profile for the infra stream and in the launch file, it is undefined (values are set to -1). You need to either modify the launch file or set them like so in the command line. Make sure the profile selected for the infrared streams matches the one selected for the depth stream (`infra_width:=848 infra_height:=480 infra_fps:=30`)

Ravi3191, 2021-03-18:
Thanks you. I am able to stream it now. It is however surprising as I was able to get infrared using my previous command a week ago.

closed 2021-03-18 by Ravi3191

doronhi, 2021-03-18:
To avoid this issue I just added some default values for the infrared streams in the rs_camera.launch file. This is not a permanent solution but it may help confusion in the meanwhile.