realsenseai/realsense-ros #1766: realsense-viewer: command not found on Raspberry Pi 4B

locvu98, 2021-03-18:
My 2Gb Raspberry Pi 4B uses Ubuntu 18.04.5 LTS and ROS Melodic. I have installed realsense2_camera for D435 from the ROS distribution. But the realsense-viewer command is not found. I try to create an udev rules as suggested but still encounter the same problem. When I try to execute the launch file, the camera automatically shuts down. How can I fix the problem? Thank you very much.
[image]

MartyG-RealSense, 2021-03-18:
Hi @locvu98   The **realsense-viewer** program is not part of RealSense ROS, it is part of the **librealsense** SDK.  

When you say that you installed realsense2_camera from the ROS distribution, do you mean that you installed with **Method 1** please (which installs the librealsense SDK and ROS wrapper together using packages)?

https://github.com/IntelRealSense/realsense-ros#method-1-the-ros-distribution

If you are installing the ROS wrapper on Raspberry Pi 4 with Ubuntu, the installation process that has been most successful has been to use **Ubuntu Server**.

If you would like to install Melodic on Ubuntu Server 18.04 on your Pi 4, the guidance in the link below from a RealSense user who succeeded in doing so will hopefully be helpful to you.

https://support.intelrealsense.com/hc/en-us/community/posts/360050092674-Realsense-D435i-with-Ubuntu-18-server-on-Raspberry-Pi4-for-Slam

locvu98, 2021-03-19:
Thank you for replying. Yes I installed with **Method 1**. 
Here's my set up:
- I flashed my Raspberry Pi 4 with Ubuntu 18.04.5 preinstalled server arm64 image 
- Then I installed ROS melodic desktop
- I installed realsense2_camera with **Method 1**
- Finally I included the 99-realsense-libusb.rules in /etc/udev/rules.d as suggested in https://github.com/IntelRealSense/realsense-ros/issues/1408
However:
- realsense-viewer command is not found
- camera node shuts down immediately after starting up
Can you show me what is missing from my set-up, please? I tried to follow **Method 2** but I could not install librealsense-dkms package.

MartyG-RealSense, 2021-03-19:
The 'Method 2' librealsense packages do not work with computers / computing devices with Arm processors, which includes Raspberry Pi, which is likely why that method failed for you.  They require an x86 or x64 processsor like those in laptops and desktops.  So I would suggest trying Method 1 again to install librealsense and the RealSense ROS wrapper together.

I would though recommend not installing the udev file at first, and only consider it if you find that librealsense will not work without it (e.g **realsense-viewer** will not launch and the command **rs-enumerate-devices** does not work).  May I confirm please that you are entering these launch commands into the **Ubuntu terminal** and not the ROS terminal?

locvu98, 2021-03-20:
Yes I ran realsense-viewer command in an Ubuntu terminal.

MartyG-RealSense, 2021-03-20:
Are you able to launch realsense-viewer or rs-enumerate devices in the Ubuntu terminal now?

And when you run a roslaunch instruction in the ROS termonal, do you still receive **control_transfer returned** errors?

locvu98, 2021-03-22:
I still couldn't launch realsense-viewer in the Ubuntu terminal.
However I installed Raspbian on Pi and managed to install librealsense by following the acrobatic guide: https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435
Now I can run realsense-viewer successfully.
How can I install ROS package of D435 from here and which version of ROS should I use?
Thank you very much.

MartyG-RealSense, 2021-03-22:
As far as I know, the best guide to installing the RealSense ROS wrapper on Raspbian is the **eleccelerator** guide.  Before that though, ROS should be installed on Raspbian first (Noetic is the most recent version of ROS1 and is supported by the RealSense ROS wrapper).

**Step 1 - install ROS Noetic on Raspbian*
https://varhowto.com/install-ros-noetic-raspberry-pi-4/

 **Step 2 - install the RealSense ROS wrapper on Raspbian**
https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Installing_ROS_Package:_realsense-ros

locvu98, 2021-03-27:
For some reasons ROS Noetic didn't work for me. But I managed to ROS Melodic and RealSense ROS Wrapper. Thank you for your help!

closed 2021-03-27 by locvu98

MartyG-RealSense, 2021-03-27:
Great news that your build was successful @locvu98 - thanks for the update!