realsenseai/realsense-ros #1767: Is it possible to convert 3d coordinate to yaw pitch and roll.

ranjitkathiriya, 2021-03-18:
Hello,

I have a 3d coordinate of an object, and I want its pose to mak a complete Quaternion. I am planning to obtain Eular(Yaw, pitch and roll) angle, and from that, I will convert to Quaternion rotation. 

`tf.transformations.quaternion_from_euler(0, 0, 0)`

[image]

Is there any way that I can obtain the pose of the detected object?

MartyG-RealSense, 2021-03-18:
Hi @ranjitkathiriya   As you have already made a start on the project by obtaining the 3D coordinate, the best way to begin investigating your case may be to look at a range of options of obtaining the pose.  If there is an option in that list that you prefer then we can focus on that.

1.  An approach involving starting the project from the beginning instead of building upon what you have already created would be to use a **DOPE** system that identifies objects and works out their pose.

https://github.com/pauloabelha/Deep_Object_Pose

2.  Alternatively, **trigonometry** maths could be used to calculate the angles of objects with your RealSense camera.

https://answers.ros.org/question/300639/how-to-get-the-distance-and-the-angle-distance-and-bearing-from-a-depth-camera/

3.  Further possibilities for obtaining object pose with ROS are discussed here:

https://support.intelrealsense.com/hc/en-us/community/posts/360051876853-Using-D435-for-tracking-3D-color-objects

ranjitkathiriya, 2021-03-19:
Hello Marty,

I cannot move to 1 option because from the below reason:
  1. 3D model training is to time taking.
  2. 3D model data generation in blender, unity, unreal engine will take time for my use-case.
  3. Now, I have achieved this much, and I cannot afford to start again from the stretch. 

I think option 2 and 3 will be helpful for me. I have checked both links but could not understand how I can get Eular angle from point cloud? Is it possible for you to explain this both ways in more depth?

I have vector point (X, Y, Z) of point cloud and I have obtained it from the pixel point(Image U, V point) of a detected object. I got my Z point from point cloud over here.

It will be great for me; you can explain point 2 and 3 in-depth. 

Thanks and regards,
Ranjit

MartyG-RealSense, 2021-03-19:
I could not find further information about obtaining pose from trigonometry.

There is a previous discussion involving a RealSense user who wanted to obtain euler angles from a point cloud because they were manipulating a D435i camera with a robot.  

https://github.com/IntelRealSense/librealsense/issues/7884

Is the content of this discussion relevant to your needs please?

closed 2021-03-25 by ranjitkathiriya

MartyG-RealSense, 2021-03-25:
Thanks very much @ranjitkathiriya for the update!

ranjitkathiriya, 2021-03-25:
Any time, I have not checked this link because I was busy with other work. I Will check afterward, and can I provide you an update on this thread?

MartyG-RealSense, 2021-03-25:
Yes, please do update here when you are able to.  Good luck!