realsenseai/realsense-ros #1769: Use Realsense D435i camera on Turtlebot3 in Gazebo simulation

MART98, 2021-03-22:
Hi friends...
I want to use Realsense D435i camera on Turtlebot3 for SLAM and Navigation in Gazebo and RViz . (I dont buy this camera yet and I want to simulating  turtlebot3 with this camera in Gazebo initially)
How can i do that?
How can i change the camera of turtlebot3 and use Realsense D435i camera instead?

ROS 1 Melodic
Ubuntu 18.04

Thanks...

MartyG-RealSense, 2021-03-22:
Hi @MART98   The link below has details of a RealSense user who used Turtlebot3 Waffle Pi with a D435 and Gazebo.

https://github.com/IntelRealSense/realsense-ros/issues/1170#issuecomment-620132790

Their work was based on a D435 Gazebo plugin by **pal-robitics**:

https://github.com/pal-robotics/realsense_gazebo_plugin

MART98, 2021-03-23:
@MartyG-RealSense Thanks. I save it in src and built it. I use this command :
roslaunch turtlebot3_gazebo turtlebot3_with_realsense_d435.launch

but there is this Error:

Resource not found: turtlebot3_description_with_d435
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/mart/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

MartyG-RealSense, 2021-03-23:
It looks as though **turtlebot3_description_with_d435** is a xacro that should be in the urdf folder of this project.  I could not locate that file though.

I note that in the urdf folder of this project's current repository, the xacro file **turtlebot3_waffle_pi.urdf.xacro** has a "D435 worked" comment beside it and contains D435 references in the file.  So I wonder whether editing the launch file to reference turtlebot3_waffle_pi.urdf.xacro instead would work.

[image]

MartyG-RealSense, 2021-03-23:
It may be worth trying the **pal_robotics** D435 gazebo plugin instead.

https://github.com/pal-robotics/realsense_gazebo_plugin

MartyG-RealSense, 2021-03-30:
Hi @MART98  Do you require further assistance with this case, please?  Thanks!

MART98, 2021-04-03:
@MartyG-RealSense
Thank you for your pursuit. Since I'm new to the ROS, I couldn't do it. I need a clear narrative :I . I meet many errors that I have told you.

MART98, 2021-04-03:
@MartyG-RealSense I despair of it.

MartyG-RealSense, 2021-04-03:
Hi @MART98  I'm sorry to hear that you are having problems with your installation.  Please let me know what the results were of your most recent re-installation attempt when you have performed it, please.  Thanks very much.

MART98, 2021-04-04:
@MartyG-RealSense  Thanks .... Now i reinstall these packages : 

https://github.com/pal-robotics/realsense_gazebo_plugin
https://github.com/IntelRealSense/realsense-ros
https://github.com/ROBOTIS-GIT/turtlebot3_simulations
https://github.com/ROBOTIS-GIT/turtlebot3

What should I do Next?

MartyG-RealSense, 2021-04-04:
Looking at the package list above, the thoughts that I have are:

1.  Whilst it is fine to have realsense-ros installed, it is not required in order to use the gazebo plugins.
2.   The **pal-robotics** and the **ROBOTIS-GIT** packages are two separate gazebo plugins by different creators that do not depend on each other.

As the pal-robotics plugin - https://github.com/pal-robotics/realsense_gazebo_plugin - is designed specifically for use with D435, I would recommend focusing solely on that link.  Instructions for installing it can be found in the discussion at the pal-robotics forum at the link below.

https://github.com/pal-robotics/realsense_gazebo_plugin/issues/7

I will provide an edited version of those instructions below.

*******************************

1.  Download the **realsense_gazebo_plugin** package and put it in your catwkinws (catkin workspace) folder.
 
2.  Compile the catkin package.  This will generate a shared library called **librealsense_gazebo_plugin.so** that will be used in the files that will be downloaded in the next step.

3.  Visit the page at the link below and download from the page's file list the two files **_d435.gazebo.xacro** and **_d435.urdf.xacro**

https://github.com/pal-robotics-forks/realsense/tree/upstream/realsense2_description/urdf

[image]

4.  Put those two files in the package inside the urdf folder where you want the RealSense camera to be simulated 

5.  Edit the file **_d435.urdf.xacro** and change line 13 from this:

`<xacro:include filename="$(find realsense2_description)/urdf/urdf_d435.gazebo.xacro"/>`

To this:

`<xacro:include filename="$(find packagename)/urdf/urdf_d435.gazebo.xacro"/>`

**packagename** is the name of the package.

6.  In the urdf file where you want to use the simulated RealSense camera, add the following code, while again replacing **packagename** with the name of the package you are using.  Also replace **baselink** with the link where you want the RealSense camera to be joint on.  
 
**Note:** I am not certain of the original instructions' meaning of 'to be joint on' in regard to the baselink.

```
<xacro:include filename="$(find packagename)/urdf/_d435.urdf.xacro" />
<sensor_d435 parent="${prefix}base_link">
</sensor_d435>
```

7.  When you launch the file which uses this urdf model, you will find the simulated RealSense and you should see rostopics like **/camera/color/image_raw**

MART98, 2021-04-06:
@MartyG-RealSense thanks...
1. you said : Put those two files in the package inside the urdf folder where you want the RealSense camera to be simulated.
What do you mean? I maked folder (urdf) in catkin_ws/src/turtlebot3_simulation/turtlebot3_gazebo and put _d435.gazebo.xacro and _d435.urdf.xacro in there. Is it ture?

2. you said: In the urdf file where you want to use the simulated RealSense camera, add the following code, while again replacing packagename with the name of the package you are using. Also replace baselink with the link where you want the RealSense camera to be joint on.
Note: I am not certain of the original instructions' meaning of 'to be joint on' in regard to the baselink.

<xacro:include filename="$(find packagename)/urdf/_d435.urdf.xacro" />
<sensor_d435 parent="${prefix}base_link">
</sensor_d435>

what do you mean? In which file i havo to write : 
<xacro:include filename="$(find packagename)/urdf/_d435.urdf.xacro" />
<sensor_d435 parent="${prefix}base_link">
</sensor_d435>

3. What is the launch comment for launching turtlebot3 with D435 after doing these changes?

MartyG-RealSense, 2021-04-07:
Sincere apologies for the delay in responding further.

Unfortunately I cannot offer any further suggestions about this Gazebo plugin after providing the edited instructions above as I do not have any knowledge about the subject of Gazebo simulation of RealSense.  I am sorry.

MART98, 2021-04-08:
@MartyG-RealSense Thanks for helping

MartyG-RealSense, 2021-04-14:
Case closed due to no further comments received.

closed 2021-04-14 by MartyG-RealSense

Selim-Savas-Oplog, 2023-11-15:
@MART98 Does anyone have a roadmap for a solution to this issue?