realsenseai/realsense-ros #1771: L515 - Control transfer returned error

htawadegetample, 2021-03-22:
Hello Everyone,
I am seeing the "Control transfer returned error" printed every second for l515 camera. I installed from source and used RSUSB_BACKEND flag during compilation. I have used initial_reset=true during launch. I am getting point cloud and image data.
I am running the l515 using ROS melodic on ubuntu 18.04 using a docker container. The kernel version is 5.4.0-67-generic. Following are the logs.

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.100.0.1:40529/

SUMMARY
========

PARAMETERS
 * /camera_0/realsense2_camera/accel_fps: 250
 * /camera_0/realsense2_camera/accel_frame_id: camera_0_accel_frame
 * /camera_0/realsense2_camera/accel_optical_frame_id: camera_0_accel_op...
 * /camera_0/realsense2_camera/align_depth: False
 * /camera_0/realsense2_camera/aligned_depth_to_color_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_0_aligned_...
 * /camera_0/realsense2_camera/allow_no_texture_points: False
 * /camera_0/realsense2_camera/base_frame_id: camera_0_link
 * /camera_0/realsense2_camera/calib_odom_file: 
 * /camera_0/realsense2_camera/clip_distance: -2.0
 * /camera_0/realsense2_camera/color_fps: 30
 * /camera_0/realsense2_camera/color_frame_id: camera_0_color_frame
 * /camera_0/realsense2_camera/color_height: 480
 * /camera_0/realsense2_camera/color_optical_frame_id: camera_0_color_op...
 * /camera_0/realsense2_camera/color_width: 640
 * /camera_0/realsense2_camera/confidence_fps: 30
 * /camera_0/realsense2_camera/confidence_height: 480
 * /camera_0/realsense2_camera/confidence_width: 640
 * /camera_0/realsense2_camera/depth_fps: 30
 * /camera_0/realsense2_camera/depth_frame_id: camera_0_depth_frame
 * /camera_0/realsense2_camera/depth_height: 240
 * /camera_0/realsense2_camera/depth_optical_frame_id: camera_0_depth_op...
 * /camera_0/realsense2_camera/depth_width: 320
 * /camera_0/realsense2_camera/device_type: 
 * /camera_0/realsense2_camera/enable_accel: False
 * /camera_0/realsense2_camera/enable_color: True
 * /camera_0/realsense2_camera/enable_confidence: True
 * /camera_0/realsense2_camera/enable_depth: True
 * /camera_0/realsense2_camera/enable_fisheye1: False
 * /camera_0/realsense2_camera/enable_fisheye2: False
 * /camera_0/realsense2_camera/enable_fisheye: False
 * /camera_0/realsense2_camera/enable_gyro: False
 * /camera_0/realsense2_camera/enable_infra1: False
 * /camera_0/realsense2_camera/enable_infra2: False
 * /camera_0/realsense2_camera/enable_infra: False
 * /camera_0/realsense2_camera/enable_pointcloud: True
 * /camera_0/realsense2_camera/enable_pose: False
 * /camera_0/realsense2_camera/enable_sync: False
 * /camera_0/realsense2_camera/filters: pointcloud
 * /camera_0/realsense2_camera/fisheye1_frame_id: camera_0_fisheye1...
 * /camera_0/realsense2_camera/fisheye1_optical_frame_id: camera_0_fisheye1...
 * /camera_0/realsense2_camera/fisheye2_frame_id: camera_0_fisheye2...
 * /camera_0/realsense2_camera/fisheye2_optical_frame_id: camera_0_fisheye2...
 * /camera_0/realsense2_camera/fisheye_fps: 30
 * /camera_0/realsense2_camera/fisheye_frame_id: camera_0_fisheye_...
 * /camera_0/realsense2_camera/fisheye_height: 480
 * /camera_0/realsense2_camera/fisheye_optical_frame_id: camera_0_fisheye_...
 * /camera_0/realsense2_camera/fisheye_width: 640
 * /camera_0/realsense2_camera/gyro_fps: 400
 * /camera_0/realsense2_camera/gyro_frame_id: camera_0_gyro_frame
 * /camera_0/realsense2_camera/gyro_optical_frame_id: camera_0_gyro_opt...
 * /camera_0/realsense2_camera/imu_optical_frame_id: camera_0_imu_opti...
[... 91 lines cut ...]
[ INFO] [1616440472.893600417]: Expected frequency for depth = 30.00000
[ INFO] [1616440472.956843910]: Expected frequency for color = 30.00000
[ INFO] [1616440472.979632769]: setupStreams...
[ INFO] [1616440473.002963848]: insert Depth to L500 Depth Sensor
[ INFO] [1616440473.003167843]: insert Color to RGB Camera
 22/03 19:14:33,299 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1616440473.307428513]: SELECTED BASE:Depth, 0
 22/03 19:14:33,349 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:33,401 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1616440473.456672468]: RealSense Node Is Up!
 22/03 19:14:34,011 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:34,354 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:34,407 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:35,609 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:38,064 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:39,318 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:40,574 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:41,828 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:43,082 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:44,335 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:45,588 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:46,841 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:49,299 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:50,553 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/03 19:14:51,806 WARNING [139796798228224] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

RealSenseSupport, 2021-03-23:
Hi @htawadegetample 

You mention using docker container, are you able to see if you get the same error outside of using a docker container?
Also is there a reason why you're using RSUSB?

htawadegetample, 2021-03-23:
> Hi @htawadegetample
> 
> You mention using docker container, are you able to see if you get the same error outside of using a docker container?
> Also is there a reason why you're using RSUSB?

@RealSenseSupport I tried without the docker container and the issue persists. 
The reason for using RSUSB is that initially I was getting "Failed to query (GET_CUR) UVC control 1 on unit 3: -32" on my dmesg logs. This was similar to [issue7099](https://github.com/IntelRealSense/librealsense/issues/7099) to which the solution mentioned by @MartyG-RealSense was issue6368[[https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279](url)

RealSenseSupport, 2021-04-12:
Are you experiencing any actual functionality problems or just the messages? There's a good discussion here with one of our RealSense engineers related to the control_transfer error messages that might be useful.

https://github.com/IntelRealSense/realsense-ros/issues/1162

Thanks

htawadegetample, 2021-05-05:
> Are you experiencing any actual functionality problems or just the messages? There's a good discussion here with one of our RealSense engineers related to the control_transfer error messages that might be useful.
> 
> #1162
> 
> Thanks

htawadegetample, 2021-05-05:
@RealSenseSupport there are no functionality issues. But it spams the logs.

Schweizliu, 2023-09-22:
> Are you experiencing any actual functionality problems or just the messages? There's a good discussion here with one of our RealSense engineers related to the control_transfer error messages that might be useful.
> 
> #1162
> 
> Thanks

I found that using the download source code compilation method git clone https://github.com/IntelRealSense/realsense-ros Run roslaunch realsene2_ Camera rs_ (messenger library. cpp: 42) control will not appear when camera. launch_ Transfer returned error, and when I use another direct download method: sudo apt-get install ros-noetic-realsense2-camera, the above warning will appear, and modifying any parameters is useless

MartyG-RealSense, 2023-09-22:
Hi @Schweizliu  If you have used the **sudo apt-get install ros-noetic-realsense2-camera** installation instruction after having already installed the librealsense SDK then you may have installed librealsense twice and have a conflict between the multiple librealsense versions.  This is because the sudo apt-get instruction installs both librealsense and the RealSense ROS wrapper from packages at the same time.  So if librealsense was installed first and then sudo apt-get was used then a second installation of librealsense would be created.

closed 2023-10-05 by MartyG-RealSense