realsenseai/realsense-ros #1772: [Report] `OpenCV` path problem in building `realsense-ros`

bigbellmercy, 2021-03-23:
In `catkin_make clean` and `catkin_make` commands for building `realsense-ros` for the first time in the system (Jetson Nano),
the errors such as the following happen much recently. But, it succeeded ago in the same environment.

### Error
```
/opt/ros/melodic/include/cv_bridge/cv_bridge.h:43:10: fatal error: opencv2/core/core.hpp: No such file or directory
 #include <opencv2/core/core.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~
```

### Solution
There is no `/usr/include/opencv` but there is `/usr/include/opencv4`. So, after making a link file as following command, the problem disappeared:
In `/usr/include` folder, `sudo ln -s opencv4 opencv`

### Environment
realsense-ros v2.2.22
ROS melodic
Ubuntu 18.04
Jetson Nano (Arm86)

MartyG-RealSense, 2021-03-23:
Thanks very much @bigbellmercy for sharing your solution with the RealSense ROS community!

MartyG-RealSense, 2021-03-30:
@doronhi Do you have any thoughts about why the above problem with **/usr/include/opencv** may have occurred on Nano for @bigbellmercy please?  Thanks!

MartyG-RealSense, 2021-04-06:
Hi @bigbellmercy   Are you still having the problem described in this case, please?

bigbellmercy, 2021-04-07:
Hi~ @MartyG-RealSense, this problem must happens all the recent users of Jetson Nano because it happens when using the Jetson Nano SD card latest image file `jetson-nano-jp451-sd-card-image.zip`. In this `jp451` version, `opencv` is not installed `/usr/include/opencv` but `/usr/include/opencv4` because the recent version of `opencv` use the folder `opencv4` as I heard, but `realsense-ros` uses the old path `opencv`. So, it seems better that `realsense-ros` coup with this new `opencv` path problem.

MartyG-RealSense, 2021-04-07:
Thanks very much @bigbellmercy - I think it will be useful to keep this case open for a couple of weeks to wait and see if any other Nano users report a similar experience to yours.  Thanks again!

doronhi, 2021-04-08:
Hi, sorry for failing to notice this issue.
opencv is not a direct requirement of realsense2_camera. cv_bridge is. It seems that the issue is with the cv_bridge package, rather than the realsense2_camera package.
Looking at the [source code of cv_bridge](https://github.com/ros-perception/vision_opencv/blob/fc782bbf9d1d02afe497fae7b0e15951163e4af6/cv_bridge/CMakeLists.txt#L21) they seem to address the issue.
Until further investigation on my part, I cannot say how best realsense2_camera can deal with this situation.
I opened an inner ticket: DSO-16886 but I expect it will take a while.

huangwen0907, 2021-04-28:
I have the similar problem and modified the file `cv_bridgeConfig.cmake`.
change the old path from: `  set(_include_dirs "include;/usr/include;/usr/include/opencv")`
to: `  set(_include_dirs "include;/usr/include;/usr/include/opencv4;/usr/include/opencv4/opencv2")`
and can fix the problem.  
I know this is not good for the system, but I do not know the other ways

MartyG-RealSense, 2021-04-28:
Thanks very much @huangwen0907  for sharing your method!

JoKoll, 2021-10-01:
Has there been an update to this problem? I'm currently encountering the same issue and haven't been able to build the realsense-ros package.

**Environment**
realsense-ros branch 3e569d8
Intel Realsense SDK (v2.49.0)
ROS melodic
Ubuntu 18.04
Jetson TX2

**Error**
`In file included from /home/jackaltx2/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/realsense_node_factory.h:16:0,
                 from /home/jackaltx2/catkin_ws/src/realsense-ros/realsense2_camera/src/realsense_node_factory.cpp:4:
/opt/ros/melodic/include/cv_bridge/cv_bridge.h:43:10: fatal error: opencv2/core/core.hpp: No such file or directory
 #include <opencv2/core/core.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:62: recipe for target 'realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o' failed
make[2]: *** [realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/realsense_node_factory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/jackaltx2/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:16:0,
                 from /home/jackaltx2/catkin_ws/src/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
                 from /home/jackaltx2/catkin_ws/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/opt/ros/melodic/include/cv_bridge/cv_bridge.h:43:10: fatal error: opencv2/core/core.hpp: No such file or directory
 #include <opencv2/core/core.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
`

I first encountered the "opencv_bridge" error when attempting to use "catkin_make clean" in Step 2 of Method 2 in the realsense-ros github page. Creating the symbolic link similar to the OP of this thread solved the issue. However when compiling the catkin_ws I got the same error. I performed the symbolic linking as instructed above, but it did not solve the problem for me. I'm hesistant to change the cv_bridgeConfig.cmake file as huangwen0907 suggested since I am running other programs with ROS.

Have other solutions been found?

MartyG-RealSense, 2021-10-02:
Hi @JoKoll  A RealSense ROS user who also recently had problems with cv_bridge during installation achieved a successful build by using **sudo apt-get install ros-melodic-realsense2-camera** and also performing "a clean catkin_make of all the workspace".  https://github.com/IntelRealSense/realsense-ros/issues/2078#issuecomment-926456214

dnovischi, 2023-01-08:
Hi @JoKoll,

 You can try to follow the tutorials I just put up right now [here](https://github.com/MartyG-RealSense). Most likely it will solve many of the problems you encounter.

Cheers.

MartyG-RealSense, 2023-07-18:
Case closed due to RealSense ROS1 wrapper no longer being updated.