realsenseai/realsense-ros #1773: missing Ros topic D435i

Majed-Alsubaie, 2021-03-24:
Hello,,,
I am trying to subscribe these topics   

`/camera/infra1/image_rect_raw \
   /camera/infra2/image_rect_raw \
   /camera/infra1/camera_info \
   /camera/infra2/camera_info \
`
I launched the file with enabling both infra1/2 , by using 

`roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true unite_imu_method:=linear_interpolation enable_gyro:=true enable_accel:=true enable_sync:=true initial_reset:=true`

but i got only these topic without infra1 / infra2

```

//camera/accel/imu_info
/camera/color/camera_info
/camera/color/image_raw
/camera/color/image_raw/compressed
/camera/color/image_raw/compressed/parameter_descriptions
/camera/color/image_raw/compressed/parameter_updates
/camera/color/image_raw/compressedDepth
/camera/color/image_raw/compressedDepth/parameter_descriptions
/camera/color/image_raw/compressedDepth/parameter_updates
/camera/color/image_raw/theora
/camera/color/image_raw/theora/parameter_descriptions
/camera/color/image_raw/theora/parameter_updates
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/extrinsics/depth_to_color
/camera/gyro/imu_info
/camera/imu
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
```

**why that happened and how can i solve it ?** thanks 
and yes there are several  warning but i read that i can safley ignore them 

**Device information :**
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
kernel : 5.8.0-48-generic
ROS : Noetic

[... 179 lines cut ...]
 24/03 08:36:27,531 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:33,585 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:36,639 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:36,689 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:38,742 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:38,792 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:39,843 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:39,894 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:40,946 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:40,996 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:42,048 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:36:42,098 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:37:03,167 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:37:03,464 ERROR [140056132056832] (ds5-options.cpp:86) Asic Temperature value is not valid!
 24/03 08:37:04,466 ERROR [140056132056832] (ds5-options.cpp:86) Asic Temperature value is not valid!
 24/03 08:37:05,467 ERROR [140056132056832] (ds5-options.cpp:86) Asic Temperature value is not valid!
 24/03 08:37:25,234 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:38:33,323 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:38:59,389 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:39:44,470 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:40:15,547 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:40:20,599 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 24/03 08:40:20,650 WARNING [140057004472064] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

doronhi, 2021-03-24:
See #1763

Majed-Alsubaie, 2021-03-24:
@doronhi  Thanks , its work 

Now , i am facing #1310  , https://github.com/introlab/rtabmap_ros/issues/517

MartyG-RealSense, 2021-03-24:
@doronhi  Thanks very much for the advice to @Majed-Alsubaie 

@Majed-Alsubaie  Can you confirm please if your new problem is with the **stereo baseline** like in the two links that you kindly provided, please?  Thanks!

Majed-Alsubaie, 2021-03-25:
@MartyG-RealSense 
Hi,,
Yes it is with  **stereo baseline**  like the two links  above

Doing:
`roslaunch rtabmap_ros rtabmap.launch    rtabmap_args:="--delete_db_on_start"    left_image_topic:=/camera/infra1/image_rect_raw    right_image_topic:=/camera/infra2/image_rect_raw    left_camera_info_topic:=/camera/infra1/camera_info    right_camera_info_topic:=/camera/infra2/camera_info    stereo:=true    wait_imu_to_init:=true    imu_topic:=/rtabmap/imu
`
what i got:

**SUMMARY**
========
```

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.9
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: False
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 1.0
 * /rtabmap/rtabmap/odom_tf_linear_variance: 1.0
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: False
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: True
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmapviz/approx_sync: False
 * /rtabmap/rtabmapviz/frame_id: camera_link
 * /rtabmap/rtabmapviz/odom_frame_id: 
 * /rtabmap/rtabmapviz/queue_size: 10
 * /rtabmap/rtabmapviz/subscribe_depth: False
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/rtabmapviz/subscribe_rgbd: False
 * /rtabmap/rtabmapviz/subscribe_scan: False
 * /rtabmap/rtabmapviz/subscribe_scan_cloud: False
 * /rtabmap/rtabmapviz/subscribe_scan_descriptor: False
 * /rtabmap/rtabmapviz/subscribe_stereo: True
 * /rtabmap/rtabmapviz/wait_for_transform_duration: 0.2
 * /rtabmap/stereo_odometry/approx_sync: False
 * /rtabmap/stereo_odometry/config_path: 
 * /rtabmap/stereo_odometry/expected_update_rate: 0.0
 * /rtabmap/stereo_odometry/frame_id: camera_link
 * /rtabmap/stereo_odometry/ground_truth_base_frame_id: 
 * /rtabmap/stereo_odometry/ground_truth_frame_id: 
 * /rtabmap/stereo_odometry/guess_frame_id: 
 * /rtabmap/stereo_odometry/guess_min_rotation: 0.0
 * /rtabmap/stereo_odometry/guess_min_translation: 0.0
 * /rtabmap/stereo_odometry/max_update_rate: 0.0
 * /rtabmap/stereo_odometry/odom_frame_id: odom
 * /rtabmap/stereo_odometry/publish_tf: True
 * /rtabmap/stereo_odometry/queue_size: 10
 * /rtabmap/stereo_odometry/subscribe_rgbd: False
[... 61 lines cut ...]
/rtabmap/stereo_odometry subscribed to (exact sync):
   /camera/infra1/image_rect_raw \
   /camera/infra2/image_rect_raw \
   /camera/infra1/camera_info \
   /camera/infra2/camera_info
[ WARN] [1616632155.931834508]: odometry: Initialized odometry with IMU's orientation (rpy = -0.146117 -0.139200 -1.528787).
[ INFO] [1616632156.018109425]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1616632156.018271631]: rtabmap: Deleted database "/home/majed/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1616632156.018301491]: rtabmap: Using database from "/home/majed/.ros/rtabmap.db" (0 MB).
[ERROR] [1616632156.031004060]: The stereo baseline (-0.049927) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1616632156.060412698]: The stereo baseline (-0.049927) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1616632156.094383915]: The stereo baseline (-0.049927) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1616632156.127131895]: The stereo baseline (-0.049927) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1616632156.160430436]: The stereo baseline (-0.049927) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1616632156.193588046]: The stereo baseline (-0.049927) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ INFO] [1616632156.223265282]: rtabmap: Database version = "0.20.7".
[ERROR] [1616632156.227400662]: The stereo baseline (-0.049927) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1616632156.255023726]: The stereo baseline (-0.049927) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.

```

how did i install realsense-ros ?:
using `sudo apt-get install ros-noetic-realsense2-camera `

Thanks for help

MartyG-RealSense, 2021-03-25:
Hi @Majed-Alsubaie   ROS wrapper **2.2.22** introduced a new system where the default profile of the particular camera model being used is applied near the end of the roslaunch process (under the **Setting dynamic config parameters** heading) if an invalid configuration request is detected.  This is why a lot of the settings appear as -1 in the upper half of the launch log.  

On the D435i, the default camera configuration is **depth + RGB**.

[image]

Under the new system, custom stream definitions in the roslaunch instruction require three factors - stream height, stream width and FPS (expressed as a decimal number, such as 30.0).  

**infra1_width:=640 infra1_height:=480 infra1_fps:=30.0 infra2_width:=640 infra2_height:=480 infra2_fps:=30.0**

The subject is discussed further in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1757

MartyG-RealSense, 2021-03-25:
The nformation I posted was in relation to your original question, where you stated that the infra1 and infra2 topics were not being published after roslaunch.

MartyG-RealSense, 2021-03-25:
Essentially, your problem seems to be that RTABMAP is reporting this:

**The stereo baseline ... should be positive (baseline=-Tx/fx).  We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg**

The problem, as you hve mentioned, is referred to in RealSense ROS case https://github.com/IntelRealSense/realsense-ros/issues/1310

I do not have knowledge of this subject unfortunately, so I will refer it to @doronhi who commented on the above case.

Majed-Alsubaie, 2021-03-26:
Hi @doronhi , Regrading the **baseline sign convention  ** issue 
i am thinking of changing the sign of p(0,3) at rostopic
`/camera/infra2/camera_info`
where the p(0,3) value is the issue' source  as suggested by https://github.com/introlab/rtabmap_ros/issues/517

So , is this a good approach ? +  how and where can i find this node/script that publish the the rostopic ?

thanks

MartyG-RealSense, 2021-03-31:
Reminder to @doronhi about the question of @Majed-Alsubaie in the comment above.  Thanks!

MartyG-RealSense, 2021-04-06:
Hi @Majed-Alsubaie  Do you still require assistance with this case, please?  Thanks!

Majed-Alsubaie, 2021-04-06:
Hi @MartyG-RealSense ,,
yes , i am looking for you help
@doronhi  , could you please help me?

doronhi, 2021-04-07:
The current positive value is set, as mentioned in [#1310](https://github.com/IntelRealSense/realsense-ros/issues/1310#issuecomment-758496209) in [here](https://github.com/IntelRealSense/realsense-ros/blob/c11f8ff2db6a334bedd555e41f9d556a85b9b2bf/realsense2_camera/src/base_realsense_node.cpp#L1878).
You can change the sign, I can't tell yet if this is a good idea or not. As you can see from the open discussion in #1310.

Majed-Alsubaie, 2021-04-08:
Thank you @doronhi  ,,  So far it is working well after i change the sign,, 

Exactly in this line:
https://github.com/IntelRealSense/realsense-ros/blob/c11f8ff2db6a334bedd555e41f9d556a85b9b2bf/realsense2_camera/src/base_realsense_node.cpp#L1913 

Thank you @MartyG-RealSense , i would like to close the case

closed 2021-04-08 by Majed-Alsubaie

MartyG-RealSense, 2021-04-08:
Thanks very much @Majed-Alsubaie - it is great to hear that the advice of @doronhi is working for you.  Thanks for the update!