realsenseai/realsense-ros #1774: [camera/realsense2_camera_manager-2] process has died on Melodic with kernel 5.4

w407022008, 2021-03-24:
Hello @MartyG-RealSense ,

Similar to #1656 , when I try to run the `rs_camera.launch` file I run into the issue that the camera manager just dies.

however, this is run on Ubuntu 18.04 - 5.4.0-65-generic ROS Melodic. The camera is a D435I. Looking at the camera feed etc. with `realsense-viewer` works without any problems.

The output is:
`~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/sique/.ros/log/521908cc-8cc3-11eb-93b9-000c291428f4/roslaunch-sique-11561.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sique:40805/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
[... 291 lines cut ...]
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1095, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 898, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 860, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 837, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
error: [Errno 111] Connection refused

[rosmaster.master][INFO] 2021-03-24 18:14:35,038: -PUB [/rosout_agg] /rosout http://sique:46783/
[rosmaster.master][INFO] 2021-03-24 18:14:35,038: -SUB [/rosout] /rosout http://sique:46783/
[rosmaster.master][INFO] 2021-03-24 18:14:35,039: -SERVICE [/rosout/get_loggers] /rosout rosrpc://sique:56907
[rosmaster.master][INFO] 2021-03-24 18:14:35,039: -SERVICE [/rosout/set_logger_level] /rosout rosrpc://sique:56907
[rosmaster.master][INFO] 2021-03-24 18:14:35,040: -CACHEDPARAM [/rosout/omit_topics] by /rosout
[rosmaster.main][INFO] 2021-03-24 18:14:35,093: keyboard interrupt, will exit
[rosmaster.main][INFO] 2021-03-24 18:14:35,093: stopping master...
[rospy.core][INFO] 2021-03-24 18:14:35,093: signal_shutdown [atexit]`

Thx a lot.

MartyG-RealSense, 2021-03-24:
Hi @w407022008   I researched your problem carefully.  In the majority of similar cases that I reviewed,  building librealsense  again from **source code** - like in https://github.com/IntelRealSense/realsense-ros/issues/1656#issuecomment-770348121 - was the solution.  If librealsense is rebuilt then the RealSense ROS wrapper should also be rebuilt from source code afterwards using the **Method 2** installation process in the RealSense ROS documentation.

w407022008, 2021-03-24:
Thanks, It's solved !

building librealsense again from source code

closed 2021-03-24 by w407022008

MartyG-RealSense, 2021-03-25:
I'm very pleased to hear that you were successful - thanks for the update!