realsenseai/realsense-ros #1775: Unable to set D435 Realsense camera beyond 30FPS

leopck, 2021-03-25:
Currently I'm using D435 Realsense Camera on Ubuntu 18.04 on librealsense v2.39.0 on Intel TigerLake Core i7-1185G7. I've used the realsense-viewer and I'm able to obtain 60FPS on the RGB camera, however, using ROS, no matter what configuration I've set, I can't go beyond 30 FPS...

I've referred to past issues relating to this and they recommended to **disable Auto-Exposure Priority while enabling auto-exposure to get 60FPS**, I've tried that but it still doesn't show beyond 30FPS.

https://github.com/IntelRealSense/realsense-ros/issues/1338#issuecomment-676128422
https://github.com/IntelRealSense/realsense-ros/issues/1040#issuecomment-569606085

My configuration
```
rosparam set /camera/rgb_camera/auto_exposure_priority false
rosparam set /camera/rgb_camera/enable_auto_exposure true

```

I've also changed the rs_camera.launch file to 60 fps for color and depth:

```
+  <arg name="depth_fps"            default="60"/>
+  <arg name="color_fps"             default="60"/>
```

How I'm running:
```
roslaunch realsense2_camera rs_camera.launch
```

How do I get the ros to configure D435 camera to 60FPS?

MartyG-RealSense, 2021-03-25:
Hi @leopck   Could you look near the bottom of your launch log, under the **Setting dynamic config parameters** heading, to check the configuration that is being applied to the camera when the launch process completes, please?  For example:

[image]

leopck, 2021-03-26:
@MartyG-RealSense here you go

```
[ INFO] [1616750291.889583421]: Setting Dynamic reconfig parameters.
[ WARN] [1616750291.908158976]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1616750291.909281011]: Done Setting Dynamic reconfig parameters.
[ INFO] [1616750291.909574034]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1616750291.909678891]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1616750291.909777596]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1616750291.910057462]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
```

MartyG-RealSense, 2021-03-26:
Thank you very much.  Which RealSense ROS wrapper version are you using with librealsense 2.39.0 please?  It is important that the wrapper version is paired as closely as possible with a particular librealsense version listed in the wrapper release notes.  The notes recommend that SDK 2.39.0 is paired with ROs wrapper version **2.2.18**

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.18

closed 2021-04-07 by MartyG-RealSense