realsenseai/realsense-ros #1776: Param '/camera/rgb_camera/power_line_frequency'

987428377, 2021-03-26:
When I use command  roslaunch realsense2_camera rs_camera.launch  there is a warn:
Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options
And I can not find the param.
Who can help me ? 
Thanks

MartyG-RealSense, 2021-03-26:
Hello, this error is discussed in the link below.  It is caused by having an unpatched kernel. 

https://github.com/IntelRealSense/realsense-ros/issues/607

**Doronhi** the RealSense ROS wrapper developer suggests in that linked case to patch the kernel using the instructions in point 4 of the section of librealsense installation instructions linked to below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites

987428377, 2021-03-26:
> Hello, this error is discussed in the link below. It is caused by having an unpatched kernel.
> 
> [#607 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/607#issuecomment-547856923)
> 
> **Doronhi** the RealSense ROS wrapper developer suggests in that linked case to patch the kernel using the instructions in point 4 **(Build and apply patched kernel modules**) of the section of the librealsense installation guide linked to below.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites
> 
> Does the ROS wrapper still work normally for you despite the warning, please?

YES I can use rostopic list find some topics

987428377, 2021-03-26:
> YES I can use rostopic list find some topics

And  an example of how to start the camera node and make it publish the point cloud using the pointcloud option.I used 
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud ，but rviz,did not open

MartyG-RealSense, 2021-03-26:
Does RViz launch if you input the command **rosrun rviz rviz**

987428377, 2021-03-26:
> Build and apply patched kernel module

> Does RViz launch if you input the command **rosrun rviz rviz**
yes I can open it

MartyG-RealSense, 2021-03-26:
Thank you very much.  Do you still have a problem that you require assistance with, please?  Thanks!

987428377, 2021-03-26:
> Thank you very much. Do you still have a problem that you require assistance with, please? Thanks!

 I want to use realsene D435i run orb-slam3, but I am a  beginning student. Thank you very much !

MartyG-RealSense, 2021-03-26:
There was a RealSense user who used a D455 camera with ORB-SLAM3 in October 2020, which suggests that you should be able to also do so with your D435i.

https://github.com/IntelRealSense/librealsense/issues/7568

If you are at a beginner level though, Intel's D435i SLAM guide that uses RTABMAP may be better for learning if you are able to use a SLAM system other than ORB-SLAM3.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

MartyG-RealSense, 2021-04-02:
Hello, do you require further assistance with this case, please?  Thanks!

987428377, 2021-04-02:
> Hello, do you require further assistance with this case, please? 
Thanks！I I will try your suggestion ！

MartyG-RealSense, 2021-04-02:
Thanks very much for the update - good luck!

closed 2021-04-06 by 987428377

jiakang, 2022-09-15:
> Hello, this error is discussed in the link below. It is caused by having an unpatched kernel.
>
> [#607 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/607#issuecomment-547856923) [...]

I am on jetson xavier NX board,  the scripts returns this: 
Unsupported kernel version 4.9.140-tegra . The Bionic patches are maintained for Bionic Beaver LTS kernels 4.1[5/8], 5.[0/3/4]

What should I do? Thanks, please...

MartyG-RealSense, 2022-09-15:
Hi @jiakang  Jetson boards have their own dedicated patching script instead of the one in the installation.md instructions.

`./scripts/patch-realsense-ubuntu-L4T.sh`

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#building-from-source-using-native-backend