realsenseai/realsense-ros #1777: Unable to change paramateres from rs_camera.launch args with L515

mmonto95, 2021-03-27:
Currently I'm using L515 Realsense Camera on Ubuntu 18.04. I've used the realsense-viewer and I'm able to change multiple parameteres on the RGB camera, however, using ROS, no matter what configuration I've set, the parameters remain with default values.

For example, if I run:
`roslaunch realsense2_camera rs_camera.launch align_depth:=true color_width:=1920  color_height:=1080`

This is the output. As you can see, the color stream remains at 1280x720:
```
process[camera/realsense2_camera_manager-1]: started with pid [12285]
process[camera/realsense2_camera-2]: started with pid [12286]
[ INFO] [1616863189.675458415]: Initializing nodelet with 4 worker threads.
[ INFO] [1616863189.800272208]: RealSense ROS v2.2.22
[ INFO] [1616863189.800335251]: Built with LibRealSense v2.42.0
[ INFO] [1616863189.800356012]: Running with LibRealSense v2.42.0
[ INFO] [1616863189.845088989]:
27/03 16:39:50,012 WARNING [140044200290048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1616863190.692319646]: Device with serial number f0264464 was found.

[ INFO] [1616863190.692387441]: Device with physical ID 4-1-7 was found.
[ INFO] [1616863190.692403073]: Device with name Intel RealSense L515 was found.
[ INFO] [1616863190.692896445]: Device with port number 4-1 was found.
[ INFO] [1616863190.693008227]: Device USB type: 3.2
[ INFO] [1616863190.695256159]: getParameters...
[ INFO] [1616863190.793418358]: setupDevice...
[ INFO] [1616863190.793500468]: JSON file is not provided
[ INFO] [1616863190.793517001]: ROS Node Namespace: camera
[ INFO] [1616863190.793547016]: Device Name: Intel RealSense L515
[ INFO] [1616863190.793556998]: Device Serial No: f0264464
[ INFO] [1616863190.793567835]: Device physical port: 4-1-7
[ INFO] [1616863190.793576815]: Device FW version: 01.05.05.00
[ INFO] [1616863190.793587019]: Device Product ID: 0x0B64
[ INFO] [1616863190.793595130]: Enable PointCloud: Off
[ INFO] [1616863190.793604210]: Align Depth: On
[ INFO] [1616863190.794120213]: Sync Mode: On
[ INFO] [1616863190.794246764]: Device Sensors:
[ INFO] [1616863190.800018362]: L500 Depth Sensor was found.
[ INFO] [1616863190.807980449]: RGB Camera was found.
[ INFO] [1616863190.808696665]: Motion Module was found.
[ INFO] [1616863190.809077967]: num_filters: 0
[ INFO] [1616863190.809185254]: Setting Dynamic reconfig parameters.
[ INFO] [1616863191.127801529]: Done Setting Dynamic reconfig parameters.
[ INFO] [1616863191.128550553]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1616863191.130455433]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1616863191.131965434]: setupPublishers...
[ INFO] [1616863191.135199292]: Expected frequency for depth = 30.00000
[ INFO] [1616863191.178978770]: Expected frequency for color = 30.00000
[ INFO] [1616863191.204952600]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1616863191.236566909]: setupStreams...
[ INFO] [1616863191.265021006]: insert Depth to L500 Depth Sensor
[ INFO] [1616863191.265229819]: insert Color to RGB Camera
27/03 16:39:51,656 WARNING [140044200290048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1616863191.685152045]: SELECTED BASE:Depth, 0
27/03 16:39:51,709 WARNING [140044200290048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
27/03 16:39:51,762 WARNING [140044200290048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
27/03 16:39:51,815 WARNING [140044200290048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1616863191.869216892]: RealSense Node Is Up!
[ WARN] [1616863191.872358343]:
[ WARN] [1616863191.872458948]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
27/03 16:39:52,383 WARNING [140044200290048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
27/03 16:39:52,718 WARNING [140044200290048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
27/03 16:39:52,873 WARNING [140044200290048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
27/03 16:39:53,974 WARNING [140044200290048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

RealSenseSupport, 2021-03-31:
@mmonto95 Please configure the resolution following the reply in https://github.com/IntelRealSense/realsense-ros/issues/1751#issuecomment-798834284 

ros2 launch realsense2_camera rs_launch.py color_width:=1920 color_height:=1080 color_fps:=30.0

mmonto95, 2021-04-02:
Hello, what I am trying to achieve is to lower the resolution because I have limited processing power in my final application. Are there any other ways to save processing power using the roslaunch parameters? Is there a way to receive data in grayscale, for example? Anyway none of the parameters I have tried has shown a different output

mmonto95, 2021-04-02:
Thanks a lot! I will give it a try and get back with the results

RealSenseSupport, 2021-04-09:
@mmonto95 Any update from your side? Thanks!

mmonto95, 2021-04-12:
Yes, It did work this way. Thank you very much! Issue solved

closed 2021-04-12 by mmonto95