realsenseai/realsense-ros #1778: Pointcloud and aligned depth to color image not published from .bag file in ROS2 Foxy

ISim000nI, 2021-03-29:
## Infos

- LibRealSense v2.43.0 
- ROS2 Foxy 
- ROS2 Foxy Wrapper build 3.1.5
- Ubuntu 20.04
- Kernel 5.4.0-050400-generic
- D435i

## Problem
When playing the intel realsense .bag file in the ros2 node with the command 
```bash
ros2 launch realsense2_camera rs_launch.py camera_name:=cam1 filters:=pointcloud rosbag_filename:=walk1.bag align_depth:=true
```
neither the pointcloud or the aligned depth to color image is published even though it is listed on the ros2 topic list.
I can play the bagfile without a problem in the realsense-viewer as well as in custom python code using pyrealsense and can use the pointcloud as well as the aligned depth to color image.

The interesting thing is as soon i add the parameter `pointcloud_texture_stream:=RS2_STREAM_ANY`, so that my command now looks as follows
```bash 
ros2 launch realsense2_camera rs_launch.py camera_name:=cam1 filters:=pointcloud rosbag_filename:=walk1.bag align_depth:=true pointcloud_texture_stream:=RS2_STREAM_ANY
```
the pointcloud is being published but the aligned depth to color image is still not being published.
When not using the bag file the node runs and publishes the topics.

In the release [ROS2-Foxy Wrapper for Intel® RealSense™ Devices (build 3.1.5)](https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.5)  there is a note: 

> following tests were removed for known playback issue with librealsense2 version 2.43.0: align_depth_color_1, align_depth_ir1_1, align_depth_ir1_decimation_1.

is this directly related to my problem ?

MartyG-RealSense, 2021-03-29:
Hi @ISim000nI   I wonder whether your problem may be related to the discussion in the link below about aligned depth point clouds in the ROS2 wrapper.

https://github.com/IntelRealSense/realsense-ros/issues/1680

ISim000nI, 2021-03-30:
What i meant was the following:

When launching 
```bash
ros2 launch realsense2_camera rs_launch.py filters:=pointcloud align_depth:=true
```
using `ros2 topic list `gives me the topics
```bash
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/color/points
/camera/depth/image_rect_raw
/camera/extrinsics/depth_to_color
/parameter_events
/rosout
/tf
/tf_static
```

and from the topics **/camera/depth/color/points** and **/camera/aligned_depth_to_color/image_raw** data can be recieved:
```bash
ros2 topic hz /camera/aligned_depth_to_color/image_raw
average rate: 30.258
	min: 0.023s max: 0.042s std dev: 0.00355s window: 32
average rate: 30.092
	min: 0.022s max: 0.042s std dev: 0.00385s window: 62

ros2 topic hz /camera/depth/color/points
average rate: 30.150
	min: 0.027s max: 0.043s std dev: 0.00385s window: 32
average rate: 30.070
	min: 0.027s max: 0.043s std dev: 0.00350s window: 62
```
but when you specify to read from a bag file with the param `rosbag_filename`
```bash
ros2 launch realsense2_camera rs_launch.py rosbag_filename:=walk1.bag filters:=pointcloud  align_depth:=true
```
the topics **/camera/depth/color/points** and **/camera/aligned_depth_to_color/image_raw** are listed but no data is being published from these topics. Neither `ros2 topic hz /camera/aligned_depth_to_color/image_raw` and `ros2 topic hz /camera/depth/color/points` show something; the same for `ros2 topic echo` and `rviz2`.

MartyG-RealSense, 2021-03-30:
Typically when specifying a file in the launch instruction, whether a bag file or a json camera configuration file, the ROS wrapper wants a full **absolute path** to the location where the file is stored instead of just the name of the file.

https://github.com/IntelRealSense/realsense-ros/issues/1180#issuecomment-619168976

ISim000nI, 2021-03-30:
@MartyG-RealSense Im sourcing my wokrspace (where i put the camera node) and then navigating to the file via the terminal and then launching the node. All topics from the bagfile are published but not **camera/depth/color/points** and /**camera/aligned_depth_to_color/image_raw** even when given the absolute path

MartyG-RealSense, 2021-03-31:
In your launch instruction, I see that you have met the conditions for the /camera/depth/color/points and /camera/aligned_depth_to_color/image_raw topics to be published by defining a pointcloud filter and setting align_depth to True respectively. 

I also note that in your topics list, the individual topics for both color and depth are published successfully, suggesting that aligned topics should be able to be generated.

Bag files created in the RealSense SDK do not save aligned data, only the individual stream data (e.g depth stream and color stream), and alignment of the bag data is typically performed as a post-process in real-time once the bag file has been loaded into memory.   I will seek clarification from the RealSense ROS wrapper developer.

Hi @doronhi  Can aligned topics be generated from the depth and color data in a bag file or does it require live camera data?  Thanks!

ISim000nI, 2021-03-31:
I just tested the wrapper from the [Foxy branch](https://github.com/IntelRealSense/realsense-ros/tree/foxy) which works with LibRealSense version: v2.42.0 and data is available from the named topics when reading from a bag file but this is not the case for   [ROS2-Foxy Wrapper for Intel® RealSense™ Devices (build 3.1.5)](https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.5) which seems to indicate that there may be indeed something not working properly but i do not have enough inside into the code so i leave it up to @doronhi.

For now i will switch to the wapper from the [Foxy branch](https://github.com/IntelRealSense/realsense-ros/tree/foxy).

MartyG-RealSense, 2021-03-31:
Thanks very much @ISim000nI for the update!

doronhi, 2021-04-06:
Hi @ISim000nI 
I think there is an issue with playback in librealsense2 version 2.43.0. I mentioned something of it in the release notes for version 3.1.5:

> Known issues:
> The following tests were removed for known playback issue with librealsense2 version 2.43.0: align_depth_color_1, align_depth_ir1_1, align_depth_ir1_decimation_1.

Sorry for joining in so late.

ISim000nI, 2021-04-06:
Hi @doronhi

is there any timeframe when this issue will be resolved ?

doronhi, 2021-04-08:
I hope soon.
I have created a test and issued an inner ticket: DSO-16888

ISim000nI, 2021-04-14:
Hi @doronhi ,

since you are working on that problem i will close this issue for now. If this shouldn't be closed just open it again.

doronhi, 2021-04-18:
I'll keep it open to be closed when fixed.

MartyG-RealSense, 2021-12-22:
Hi @doronhi  Is this case able to be closed yet or does it need to still be kept open in order for a fix to be made?  Thanks!

MartyG-RealSense, 2022-02-13:
DSO-16888 was closed without a fix in librealsense being implemented.  I will therefore close this case.

closed 2022-02-13 by MartyG-RealSense