realsenseai/realsense-ros #1779: Question: dependencies inside docker

p0rys, 2021-03-29:
Hey guys,

i have a few short questions. I'm working on a ros docker container that is running multiple realsense nodes. 

**First:**
Right now i'm installing the suggested packages (librealsense2-udev-rules librealsense2-dkms librealsense2 librealsense2-utils librealsense2-dev) via apt from realsense package repo (http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo) AND also "ros-foxy-librealsense2" via rosdep before compiling this repo (branch foxy).

Is this really necessary to install both or is it maybe enough to just install the "ros-foxy-librealsense2" package?  

**Second:**
Do i need the lib and udev rules on the docker host when running the container in privileged mode? Most of the time i get a few cameras / nodes working (not always the same ones) but the other ones complaining about not find the device (mapped via serialnumber).

Thanks!

Greetings,
p0rys

p0rys, 2021-04-06:
Hi @MartyG-RealSense , really appreciate your help!

1)
I think i just missed "**sudo apt purge ros-$ROS_DISTRO-librealsense2 -y**" from install method 2 which led to the confussion... Is it right that i can't use install method 1 on Ubuntu 20.04 and ROS foxy? Apt does not find the package "**ros-foxy-realsense2-camera**" even with ROS and Intel Realsense sources.

Currently i'm using the "foxy"-branch of this repository. If i understand it correctly, i could also use the "ros2"-branch as it is compatible with dashing, eloquent and foxy, right? 

2)
It seems that all cameras are working out of the box (with docker privileges) without installing something on the host maschine but the connection is not consistent (sometimes it find more cameras, sometimes less [out of four]) . I will read through the mentioned comments and try a few things if it is not getting better. Maybe it is only something with our current setup (host is Ubuntu 18.04 and docker Ubuntu 20.04). I will write a comment if i find a solution (or if it magically works ;D )

MartyG-RealSense, 2021-04-06:
Hi @p0rys  You are very welcome!

1. The first step of the ROS2 instructions, which comes before choosing Method 1 or 2, recommends using Ubuntu 18.04 with Dashing and Ubuntu 20.04 with Foxy.

https://github.com/IntelRealSense/realsense-ros/tree/ros2#step-1-install-the-ros2-distribution

Since Method 1 simply asks for the instruction **sudo apt-get install ros-$ROS_DISTRO-realsense2-camera** to be input without having to define the $ROS-DISTRO part of the instruction first as required in the ROS1 wrapper, I would assume (not certain) that the choice of 18.04 or 20.04 in the preceding step pre-determines the ROS version that is substituted in for $ROS-DISTRO

If you chose to use the older unsupported ROS2 wrapper then you would not be able to use it with more recent RealSense camera models such as D455 and L515, as it would not have the support for these that the current ROS2 wrapper does.

2.  I have a recollection of a RealSense Docker user recommending that the same kernel version be used for both the host and Docker, though that was the opinion of one person.

p0rys, 2021-04-06:
Thanks @MartyG-RealSense !

Yes, right now i'm on Ubuntu 20.04 with Foxy. I tried the mentioned command and even **sudo apt-get install ros-foxy-realsense2-camera** directly. It still does not find the package in my apt sources (ubuntu, ROS and Intel-Realsense). 
I do **not** want to use the old wrapper and right now i'm totally fine with cloning this repository (branch ros2) and compile it if there is no deb-package available. I was just not sure if i'm doing something wrong because the install guide recommends installing the deb-package...

For the different kernel versions on host and docker -> i will try a few different combinations when i have a little bit more time and post the results here.

eholmes13, 2021-04-07:
@p0rys I'm in a similar boat to you, I'm attempting to create a docker container on Ubuntu 18.04 with Dashing. I'm running into the same issue of not finding a ros-dashing-realsense2-camera package in my apt sources, have you made any progress with figuring out the issue?

MartyG-RealSense, 2021-04-13:
Hi @p0rys  Do you have an update about this case that you can provide, please?  Thanks!

p0rys, 2021-04-13:
Hey @MartyG-RealSense 

Sorry for the late response. Indeed i tried a few different configurations. It seems that it is enough to install the [librealsense2](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages), this repository (branch ros2) and remove _ros-foxy-librealsense2_ manually to get it to work constantly. It is NOT needed to install anything on the host maschine (maybe it is better, but i did not notice any problems. Even with the different kernel versions on host and guest). 

I think the main problem was that rosdep is automatically installing ros-foxy-librealsense2. After removeing it and installing the version from [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) everything seems fine.

@eholmes13 
Right now i did not find a deb package for this wrapper. It seems that it is not available in ros deb sources. There is nothing I (or we) can do about that. Someone with the needed access rights need to add it. 

For now i just download these repository inside my dockerfile, install all dependencies and compile it manually. If it helps, i can provide you a few code snippts.

MartyG-RealSense, 2021-04-14:
Thanks very much @p0rys for the confirmation that your Docker setup is working for you if you make manual adjustments to the setup.

MartyG-RealSense, 2021-04-20:
Hi @p0rys  Do you require further assistance with this case, please?  Thanks!

p0rys, 2021-04-20:
Hey @MartyG-RealSense As everything seems to run fine now, the issue can be closed.  

@eholmes13 Got the confirmation that the deb packages are missing. They should be available in a few weeks. See: #1814

closed 2021-04-20 by p0rys

MartyG-RealSense, 2021-04-20:
Thanks very much for the update @p0rys - great to hear that your system is successful!