realsenseai/realsense-ros #1780: `git checkout` can not switch to any 3.x.x version tag

ravijo, 2021-03-31:
Please see the below snippet taken from [the realsense-ros documentation](https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources).

Clone the latest Intel® RealSense™ ROS from here into 'catkin_ws/src/'
```
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ..
```

Using ```git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1``` , the head now switches to `2.2.23`. Please see below for more info:

```
test@vm:~/temp$ git clone https://github.com/IntelRealSense/realsense-ros.git
Cloning into 'realsense-ros'...
remote: Enumerating objects: 109, done.
remote: Counting objects: 100% (109/109), done.
remote: Compressing objects: 100% (52/52), done.
remote: Total 11413 (delta 56), reused 107 (delta 56), pack-reused 11304
Receiving objects: 100% (11413/11413), 30.08 MiB | 5.36 MiB/s, done.
Resolving deltas: 100% (6893/6893), done.
test@vm:~/temp$ cd realsense-ros/
test@vm:~/temp/realsense-ros$ git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
Note: switching to '2.2.23'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at be1bb6c 2.2.23
test@vm:~/temp/realsense-ros$ git status
HEAD detached at 2.2.23
nothing to commit, working tree clean

```

### Reason
The statement `grep -P "^2.\d+\.\d+" ` is looking for text starting from 2. Therefore version 3.x.x are unrecognized.

### Workaround 
The statement can be modified as `grep -P "^\d+\.\d+"` to search any text starting from a digit. See below for more info:

```
test@vm:~/temp/realsense-ros$ git checkout `git tag | sort -V | grep -P "^\d+\.\d+" | tail -1`
Previous HEAD position was be1bb6c 2.2.23
HEAD is now at 7acba5f 3.1.5
test@vm:~/temp/realsense-ros$ git status 
HEAD detached at 3.1.5
nothing to commit, working tree clean
```

Kindly update the documentation as necessary.

MartyG-RealSense, 2021-03-31:
@ravijo  Thanks very much for your detailed documentation change suggestion.

@doronhi  As this is related to RealSense ROS wrapper documentation rather than librealsense documentation, I will refer the above suggestion of @ravijo to you.  Thanks!

doronhi, 2021-04-06:
Hi @ravijo 
[the realsense-ros documentation](https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources) you mention refers to the ROS1 installation. Versions 2.x.x refers to ROS1 distributions so the script in this document is intentional.
Versions 3.x.x belong to ROS2 distributions. The documentation for those is [here](https://github.com/IntelRealSense/realsense-ros/tree/ros2) and it is, indeed, not updated so for that I thank you for drawing my attention.

ravijo, 2021-04-06:
Ohhh... I understand.

Thank you so much for the update @doronhi 

Closing this issue and pull request too https://github.com/IntelRealSense/realsense-ros/pull/1782

closed 2021-04-06 by ravijo

MartyG-RealSense, 2021-04-06:
@doronhi   Thanks very much for your advice to @ravijo 

@ravijo  I'm pleased that you had a successful outcome to your case.  :)