realsenseai/realsense-ros #1781: Auto Shutdown after launching

loguna123, 2021-03-31:
Hi,

When i launch my real sense D435i using "rs_camera.launch", it runs for a few seconds and shutdown giving me a  message:"process has finished cleanly" on the terminal.

MartyG-RealSense, 2021-03-31:
Hi @loguna123  As a starting point in investigating your case, could you try adding the instruction below to the end of your roslaunch instruction to see whether it improves your situation, please?  The command performs a reset of the camera at launch:

**initial_reset:=true**

MartyG-RealSense, 2021-04-07:
Hi @loguna123  Do you require further assistance with this case, please?  Thanks!

loguna123, 2021-04-07:
Hi, 
Thanks for helping me.

closed 2021-04-07 by loguna123

MartyG-RealSense, 2021-04-07:
You are very welcome @loguna123 - thanks for the update!