realsenseai/realsense-ros #1783: uvc streamer error on Odroid XU4

Daniel-Merizalde, 2021-03-31:
Hello, 

Me and my team have been trying to record a .bag file for RGBD point clouds using the Realsense L515 Camera.

I have had some success using the following commands:

Terminal 1:

`roslaunch realsense2_camera rs_camera.launch align_depth:=true color_width:=1280 color_heigth:=720 `

Terminal 2:

`rosbag record -b 1200 --split --size=100 /rtabmap/odom    /camera/color/image_raw    /camera/aligned_depth_to_color/image_raw    /camera/color/camera_info    /rtabmap/odom_info  /camera/color/image_raw    /camera/aligned_depth_to_color/image_raw    /camera/color/camera_info  /rosout /rosout_agg /tf /tf_static`

Unfortunately with these commands after some time of recording i get the following error:

`ERROR [547113386368] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132`

The endpoint is usually between 130 and 133, after the error happens the camera shuts down and stops recording points into the .bag files.

I am currently using Ubuntu 18.04, with ROS Melodic and Kernel 4.14. 

Any help would be appreciated.

RealSenseSupport, 2021-04-14:
Were you able to get things working?

Daniel-Merizalde, 2021-04-15:
No, i tried to reinstall with ``sudo udevadm control --reload-rules && udevadm trigger`` and `sudo modprobe uvcvideo` and added the 99-realsense-libusb.rules file to /etc/udev/rules.d but the uvc streamer error is still happening.

Any help would be really appreciated,

Daniel.

closed 2021-09-17 by MartyG-RealSense