realsenseai/realsense-ros #1784: can you please add to the rs_camera.launch option to set argument to /camera/color/image_raw/compressed jpeg_quality

YakirHuri, 2021-04-01:


MartyG-RealSense, 2021-04-02:
Hi @YakirHuri   You could configure your own custom launch file with this setting in it.  The **image transport** documentation link below provides an example of configuring a param for it:

**/camera/image/compressed/jpeg_quality (int, default: 80)**

http://wiki.ros.org/image_transport#Parameters

If you prefer not to create a custom launch file, I have seen a couple of examples in the past of this setting being configured dynamically after launch during runtime with a rosrun instruction.  An example of such an instruction, which is input directly **after** the roslaunch instruction, is below.

**rosrun dynamic_reconfigure dynparam set /camera/color/image_raw/compressed jpeg_quality 70**

MartyG-RealSense, 2021-04-08:
Hi @YakirHuri  Do you require further assistance with this case, please?  Thanks!

YakirHuri, 2021-04-08:
Hi, i added the param to the launch file and its work fine, thanks

MartyG-RealSense, 2021-04-08:
That is great to hear @YakirHuri - I will close the case as you had a successful outcome.  Thanks!

closed 2021-04-08 by MartyG-RealSense