realsenseai/realsense-ros #1786: RGB image exposure control

Hiroaki-K4, 2021-04-02:
Hi.
It seems that the depth image can be set from the launch file. Is it possible to control the exposure of the RGB image?
Thanks!

MartyG-RealSense, 2021-04-02:
Hi @Hiroaki-K4   You could try configuring custom RGB exposure dynamically during runtime after launch by inputting the command below directly after the roslaunch instruction:

**rosrun dynamic_reconfigure dynparam set /camera/rgb_camera exposure 100**

Hiroaki-K4, 2021-04-05:
Hi @MartyG-RealSense 
Thanks, I understood.
I found that exposure control is possible when the node is started. Is it possible to dynamically control the exposure of the rgb image while the node is running?

MartyG-RealSense, 2021-04-05:
The rosrun command above does change exposure after launch, as it is a 'dynamic reconfigure' command - changing the setting during runtime after the node has been launched.

Hiroaki-K4, 2021-04-07:
@MartyG-RealSense 
ok.
Is it possible to control the exposure without using the rosrun command?
For example, publish the exposure control value from another node and subscirbe from the dynamic_reconfigure node.

MartyG-RealSense, 2021-04-07:
In the case in the link below, someone set up a listener node and used **query_devices** to change the exposure from within that node.

https://github.com/IntelRealSense/realsense-ros/issues/1206#issuecomment-635179346

Earlier in that discussion they had tried setting exposure (depth exposure in their case) with dynamic reconfigure from within the listener node with the instruction below but found it to be very slow.

**system("rosrun dynamic_reconfigure dynparam set /D435_2/stereo_module exposure 100");**

Hiroaki-K4, 2021-04-07:
ok. Thanks!

closed 2021-04-07 by Hiroaki-K4