realsenseai/realsense-ros #1787: Error extracting usb port from device with physical ID: 2-1.4-3

Andyzug, 2021-04-03:
Hi! 
After running: `roslaunch realsense2_camera rs_camera.launch`
I get this warning: 
`[ WARN] [1617470494.280979430]: Error extracting usb port from device with physical ID: 2-1.4-3
Please report on github issue at https://github.com/IntelRealSense/realsense-ros`
Then:
`03/04 13:21:35,981 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61`

Devices I use:
- camera D435
- Jetson Nano 

kernel:
`uname -a
Linux andy-desktop 4.9.201-tegra #1 SMP PREEMPT Fri Feb 19 08:40:32 PST 2021 aarch64 aarch64 aarch64 GNU/Linux`

Full output:
```
andy@andy-desktop:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/andy/.ros/log/078beca8-94a1-11eb-b9bb-84c5a6ec47c1/roslaunch-andy-desktop-24478.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://andy-desktop:44209/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 1080
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1920
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
[... 90 lines cut ...]
[ INFO] [1617470495.485453693]: Expected frequency for infra1 = 30.00000
[ INFO] [1617470495.540205538]: Expected frequency for infra2 = 30.00000
[ INFO] [1617470495.593690409]: Expected frequency for color = 30.00000
[ INFO] [1617470495.652969872]: setupStreams...
[ INFO] [1617470495.678845166]: insert Depth to Stereo Module
[ INFO] [1617470495.679525556]: insert Color to RGB Camera
[ INFO] [1617470495.679722074]: insert Infrared to Stereo Module
[ INFO] [1617470495.679848016]: insert Infrared to Stereo Module
 03/04 13:21:35,858 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1617470495.909793256]: SELECTED BASE:Depth, 0
[ INFO] [1617470495.942474950]: RealSense Node Is Up!
 03/04 13:21:35,981 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/04 13:21:36,032 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/04 13:21:36,082 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/04 13:21:36,133 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/04 13:21:36,257 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/04 13:21:36,309 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/04 13:21:36,361 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/04 13:21:36,414 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/04 13:21:36,464 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 03/04 13:21:36,819 WARNING [547868369280] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

Can you help me with that?
Thank you!

MartyG-RealSense, 2021-04-04:
Hi @Andyzug   Are the **control_transfer returned** errors only occurring at launch and then stopping after launch, or do they generate continuously after launch.  

If they only appear a small number of times at launch, or once or twice a minute after launch, then they can be safely ignored.  If they are continuous then it can indicate that there is a serious communication problem with the device.

Andyzug, 2021-04-04:
Yes, it's occurring just at the beginning, got it, thanks.
What about "Error extracting usb port from device with physical ID"? It will affect somehow the further usage of the node?

MartyG-RealSense, 2021-04-04:
If the topics are successfully publishing data and can be rendered on RViz then the USB port error can likely be treated as a momentary 'hiccup' that the system has recovered from. 

You can check the state of a topic with the **rostopic echo** instruction.  For example:

**rostopic echo /camera/depth/image_rect_raw/header**

Andyzug, 2021-04-04:
Seems it works just fine.
Thank you.

closed 2021-04-04 by Andyzug

MartyG-RealSense, 2021-04-05:
Great news @Andyzug - thanks very much for the update!