realsenseai/realsense-ros #1788: D435i bad depth image

YunchuZhang, 2021-04-04:
Hi there,
I use this official ros package to get depth stream and check with rqt_image_view.
But it is pretty noisy with a lot of black points. After I use my hand to get closer to the camera lens and overlap the whole camera and then release my hand. The depth could go better and has little black points.

Could someone know what's this problem? And how to solve this?

MartyG-RealSense, 2021-04-04:
Hi @YunchuZhang   If the depth image improves after covering the camera lenses with your hand and then removing the hand, that would suggest to me that the noise may be resulting from light saturation of the infrared sensor (the depth image is generated from the left and right infrared sensors on the 400 Series stereo depth cameras).  

Such saturation may occur if the camera is pointed directly at the sun or the sun is reflected directly into the view of the camera, and be recoverable by the auto-exposure if a hand is covered over the lenses or the camera's view is redirected away from the sun.

A past example of this phenomenon occurring indoors was when a glass door in an office would reflect sunlight from a window into the camera's view when the door was opened. 

These principles are described in the section of Intel's camera tuning guide about use of the camera in sunlight.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-sunlight-but-avoid-glare

Is it possible that the situation described above might be occurring to your camera, please?

YunchuZhang, 2021-04-04:
Hi, thanks for your reply!
We are at indoor setup,  do you think this white board will directly reflect strong light back to camera? To cause this issue?
[image]
[image]
[image]

Thanks

MartyG-RealSense, 2021-04-04:
The strong illumination is useful to produce readable depth information from the black metal sections, which would otherwise likely be unreadable by the camera.  Black or dark grey colours absorb light (a general physics principle not specific to RealSense), making them hard for the camera to read depth detail from.   Strong illumination can bring out depth detail from such surfaces though.

In regard to the white board: although it is not shown clearly on the RGB image, if the strong illumination creates a powerful area of projected light on the board then that could also prevent depth information from being read in that part of the scene.  This could result in an empty black area with no depth, like on the center of the white board on your depth image.

YunchuZhang, 2021-04-05:
Thanks for your reply! Maybe the strong illumination reflected by the white board and cause the black area.
Is there a way we could replace the hand covering movement with some programming process?

MartyG-RealSense, 2021-04-06:
If there is glare from reflection, the negative effect of this on the image can be significantly reduced by applying a physical **optical filter** product called a **linear polarization filter** over the top of the lenses on the outside of the camera.  This is described in Section **4.4  When to use polarizers and waveplates** of the paper.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters

[image]

Alternatively, the section of Intel's white-paper document on camera tuning that covers the effects of sulight and glare suggests that "for many applications, like autonomous robots or cars, it helps to simply change the Region-of-interest of the auto-exposure algorithm to a smaller size, or specifically to the lower half of the image".

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-sunlight-but-avoid-glare

closed 2021-04-19 by MartyG-RealSense