realsenseai/realsense-ros #1789: How to record more than one device?

victorll998, 2021-04-06:
Hi there,

I have got a system that uses D455 and T265. In my application, I need a synchronized data stream from both D455 and T265. I found realsense-viewer can display depth and tracking data seamlessly in one view. 

My question is if I can record D455 and T265 into one bag file in the realsense-viewer. If I can't, does ROS wrapper support synchronized replay from two bag files?  
 
Thank you in advance.

RealSenseSupport, 2021-04-08:
@victorll998 Please note that the bag file recorded via RealSenseViewer should be replayed with RealSenseViewer and can't be replayed under ROS. You need record ROS bag to replay under ROS. For details about ROS bag recording please see https://wiki.ros.org/rosbag/Commandline#rosbag_record

jeffdelmerico, 2021-04-08:
I'm running into the same problem...I connected two sensors (t265 and d415) simultaneously, and recorded two bags through the viewer.  I would like to play both of these back into ROS to recreate the topics, tf tree, etc.  Running the bag for the d415 through `rs_from_file.launch` works as expected, and republishes all of the data in ROS.  With the t265, however, loading the bag fails with the same output as the original poster:
```
[ INFO] [1617910087.390691404]: Initializing nodelet with 8 worker threads.
[ INFO] [1617910087.495343081]: RealSense ROS v2.2.22
[ INFO] [1617910087.495368724]: Built with LibRealSense v2.42.0
[ INFO] [1617910087.495378151]: Running with LibRealSense v2.42.0
[ INFO] [1617910087.512972061]: publish topics from rosbag file: /path/to/bagfile.bag
[ERROR] [1617910099.596389027]: Exception: Could not find requested sensor type!
```

**What is the purpose of `rs_from_file.launch` then?** 

From reading the code, the `realsense_node_factory` appears to follow the same procedure to create a playback pipeline as in [this example](https://github.com/IntelRealSense/librealsense/blob/5ff27fca5aaeec4736d6bb3bfb958fee82ee198b/examples/record-playback/readme.md) from librealsense.  This would make sense if it was designed to replay bags that had been recorded in RS format through the ROS node so that it would behave like a live sensor.  But @RealSenseSupport seems to suggest that a bag recorded from the viewer can't be played back this way.  It seems unintuitive that a ROS bag recorded through ROS would need to be played back _through the device driver_ rather than just with `rosbag play`.

victorll998, 2021-04-09:
@jeffdelmerico Likewise, I can replay the bag for d455 from realsense_viewer through '''rs_from_file.launch''' but fail to load with t265. It is interesting that I can actually load and play both bag file from the viewer through '''rosbag''' even though they say not

@RealSenseSupport I still would like to know if realsense-viewer would support record multiple devices into one bag file. That would be much more convenient if you have a lot of bag files to check and replay when using multiple devices.

RealSenseSupport, 2021-04-09:
@victorll998 Sorry at present we don't have such plan to add new feature to record multiple cameras into one bag file. https://github.com/IntelRealSense/librealsense/issues/8183 mentioned some workaround and please check if it helps. Thanks!

doronhi, 2021-04-20:
The issue of playing back a T265 recording is discussed in #1807 
Considering the answer of @RealSenseCustomerSupport , can this issue be closed?

victorll998, 2021-04-21:
> The issue of playing back a T265 recording is discussed in #1807
> Considering the answer of @RealSenseCustomerSupport , can this issue be closed?

Thank you for the update. I am glad that there is a solution to read t265 bag file on ROS.

- For now I am trying to record 3 devices (2 d455 and 1 t265) that are connected through a 3.0 USB docker. I use all three 3.0 cables and realsense-viewer to obtain the bag format files. I noticed that warning  ```(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61``` keeps occurring. From what I have learned from other posts, it suggests communication problems with the device. I have tried ```initial_reset``` and ```set all stream rate (color and depth fps) to 5``` and these warnings are somehow relieved. But when I move the camera these warnings pop out again.

```
 21/04 14:35:12,548 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:31,617 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:37,674 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:38,726 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,780 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,831 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,881 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,932 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,982 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:43,034 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:43,085 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:45,138 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:46,190 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:48,243 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

```

Though I can visualize related topics in Rviz, I am worried about this issue could affect the quality of the recording (i.e periodic frame missing)

- Besides, the log file shows other warnings when the viewer is recording. I checked the bag file with ```rs-rosbag-inspector``` and there is no missing message from what I have seen. I wonder what are these warnings about and if there is a way to fix them.

```
21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Motion
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Video
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Roi
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Depth Frame
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Advanced Mode
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Video Profile
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Disparity Frame
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Motion Profile
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Pose Frame
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Pose Profile
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Tm2
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Software Device
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Software Sensor
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Decimation Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Threshold Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Disparity Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Spatial Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Temporal Filter
[... cut ...]
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Calibrated Sensor
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Hdr Merge
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Sequence Id Filter
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Max Usable Range Sensor
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Debug Stream Sensor

```

doronhi, 2021-04-21:
I believe your issue now is with recording 3 devices using realsense-viewer and is no longer related to the ROS wrapper. If I am correct, I believe posing an issue in the [librealsense2 repository](https://github.com/IntelRealSense/librealsense/issues) is more likely to produce results.

victorll998, 2021-04-21:
@doronhi  You are correct. I almost forgot that here is for the ROS wrapper and I will post my viewer issue at the neighbor. Would you mind if I leave this thread open when I have some further questions?

doronhi, 2021-04-21:
Personally, I don't mind at all. I find, however, that short issues are much easier to address as one doesn't get confused with irrelevant old stories and scrolling back and forth. Also, the title becomes less informative.
I believe once you have more questions it will be better if you created new issues. After all, there is no limit to the number of issues you can post.

victorll998, 2021-04-21:
@doronhi thank you for your advice. I will be aware of the title when coming up with new issues. I posted the following up issue related to ```rs_from_file.launch``` in the other thread. For now, I will be closing this issue to keep this thread as clean and concise as it should be.

closed 2021-04-21 by victorll998

doronhi, 2021-04-21:
@victorll998 , looked at the issue you have posted in the librealsense2 repository. I would delete the section about the RViz. Unless you think the issue has something to do with ROS or the ROS wrapper, I think it will confuse people. Always try to isolate the problem.
Just my opinion.