realsenseai/realsense-ros #1793: Getting distances of points in pointcloud.

rahulsharma11, 2021-04-07:
Required Info -

Camera Model | D435i
Firmware Version | 05.12.12.100
Operating System & Version | Linux 20.04
Kernel Version (Linux Only) | 5.8.9-050809-generic
Platform | laptop
SDK Version | 2.43.0
Language | C
Segment | others

Hi,
I am using roslaunch command - "roslaunch realsense2_camera opensource_tracking.launch initial_reset:=true is_dense:=true".

I have 4 quick querries.

1) While running the app, rviz shows decent results. But after few seconds like 10-15, it freezes and terminal output is-
[ INFO] [1617772858.545138390]: Odom: quality=520, std dev=0.007428m|0.038960rad, update time=0.534487s
[ WARN] [1617772859.394979272]: Failed to meet update rate! Took 0.010023749000000000201
[ WARN] [1617772859.396603784]: Failed to meet update rate! Took 0.0084350780000000003672
[ WARN] [1617772860.715949553]: Failed to meet update rate! Took 0.0066743310000000003607
[ WARN] [1617772865.583072671]: Still waiting for data on topic /imu/data_raw...

**Then it runs normally. This process happens continously. I checked with lower fps also. Any steps to resolve this?**

2) When i rotate the camera to get the 3d pointcloud, sometime it hangs and i can see IMU drifting starts which sometimes get resolved automatically but sometime not. 
output terminal shows-
[ WARN] (2021-04-07 10:59:16.934) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=34) between -1 and 564"
[ INFO] [1617773356.935316361]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.089860s
[ WARN] (2021-04-07 10:59:17.058) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=24) between -1 and 565"
[ INFO] [1617773357.059841865]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.090116s
[ WARN] (2021-04-07 10:59:17.186) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=28) between -1 and 566"
[ INFO] [1617773357.187354359]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.091475s
[ERROR] (2021-04-07 10:59:17.190) Rtabmap.cpp:1147::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 970 is ignored!

**Does that mean i need to rotate the camera very slowly? and if drift does not autocorrected , then i need to restart it everytime?**

3) **Is there any way i can get point to point distance in pointcloud? like i want to confirm that wall in front is at 2meter should be exactly there in pointcloud? i saw "show_center_depth.py", but i need to see in my desired point to point distance in pointcloud.**

4) **General querry is why IMU is required to get the camera orientation? like in ORB_SLAM, stereo pair is sufficient to get the camera orientation.**

Thanks.

MartyG-RealSense, 2021-04-07:
Hi @rahulsharma11  Does the stability of your IMU results after launch improve if you add the commands below to the end of your roslaunch instruction:

**global_time_enabled:=false unite_imu_method:=linear_interpolation**

Some RealSense ROS users have found that they have better results if they use **unite_imu_method:=copy** instead of linear_interpolation.

Intel's D435i SLAM guide - which also uses **roslaunch realsense2_camera opensource_tracking.launch** - offers the following advice about losing track during movement:

*****************

The built-in IMU can only keep track for a very short time.  Moving or turning too quickly will break the sequence of successful point cloud matches and will result in the system losing track.  It could happen that the system will recover immediately if stopped moving but if not, the longer the time passed since the break, the farther away it will drift from the correct position.  

The odds for recovery get very slim, very quickly.  The parameters set in the launch file are most likely not ideal but this is a good starting point for calibrating.

******************

The D435i SLAM guide can be found here:

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

A RealSense ROS user expanded upon this guide to add information about setting up **localization** with RTABMAP to aid recovery when tracking drifts.

https://shinkansan.github.io/2019-UGRP-DPoom/SLAM.html

In regard to 'point to point' distance measurement with a point cloud, the link below provides guidance about using RViz to measure between two selected points:

https://github.com/IntelRealSense/realsense-ros/issues/553

Similar point to point selection and measurement is also available in the 3D point cloud mode of the RealSenSe Viewer tool.

Regarding use of IMU in SLAM: as you pointed out, IMU-less cameras such as D435 can be used for SLAM in programs such as ORB-SLAM2.  SLAM that makes use of IMU data can be referred to as **Visual-Inertial Odemetry (VIO)** and can utilize the **pose** (position and rotation angle) of the camera.  There are methods that can be used to get the camera pose without an IMU though, such as a **plane-fit** algorithm (which is built into the SDK's **Depth Quality Tool**).

https://github.com/IntelRealSense/librealsense/issues/4440

rahulsharma11, 2021-04-08:
Hi,
Thanks for the information.

**Does the stability of your IMU results after launch improve if you add the commands below to the end of your roslaunch instruction:**
No, it still there. When i rotate the camera with my hand then it happens anytime. May i know what is the max speed i can move the camera? as it has to be put on a ground rover. Is there any recommendations of motion of camera to get the pointcloud to avoid drift?

**There are methods that can be used to get the camera angle without an IMU though, such as a plane-fit algorithm (which is built into the SDK's Depth Quality Tool).**

Do you recommend this? Like getting pointcloud without IMU is better than with IMU? Because if drift does not recover by itself then i need to restart the process again.

Also How many points does the realsense capture at one time/one frame?

rahulsharma11, 2021-04-08:
Also i want to know the reason behind this -

[ERROR] (2021-04-08 12:38:36.042) Rtabmap.cpp:1147::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 865 is ignored!
[ INFO] [1617865716.042676284]: rtabmap (25): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0002s, Maps update=0.0000s pub=0.0000s (local map=4, WM=4)
[ WARN] (2021-04-08 12:38:36.160) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/10 (matches=42) between -1 and 489"
[ INFO] [1617865716.161218580]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.100379s
[ WARN] (2021-04-08 12:38:36.281) OdometryF2M.cpp:557::computeTransform() Registration failed: "Not enough inliers 0/10 (matches=40) between -1 and 490"
[ INFO] [1617865716.282541397]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.101424s

So that i can take precautions.

MartyG-RealSense, 2021-04-08:
1.  I do not have any data relating to how much motion it takes for the D435i's IMU values to drift when used in ROS.  

2.  Generally in the SLAM cases that I have handled, RealSense users with IMU-equipped camera models such as D435i and D455 prefer IMU-based SLAM to IMU-less options.  It is a personal choice for the individual user though.

3.  The RealSense **L515** lidar depth model states the ability to generate 23 million accurate depth points per second.  I do not know of an equivalent depth points per second value for D435i.

Intel's **Which Intel RealSense device is right for you?** camera comparison guide does provide the following information though:

*****************

Looking at the same size area on the pictures, notice the Intel RealSense D415 has a higher pixel density, which means it’s more accurate for any given area than the Intel RealSense D435 at the same points.  At the same resolution, the D415 has more pixels per-degree, since it has the same number of pixels in a narrower field of view. 

While this may not matter in some cases, such as object avoidance or people detection, when accuracy is the number one consideration, for example in 3D scanning, the D415 will provide a better, more accurate scan.  In short, the D415 is more than twice as accurate as the D435.

https://www.intelrealsense.com/which-device-is-right-for-you/

*****************

4.  In regard to the **Failed to meet update rate** warning, I refer you to the information in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1234#issuecomment-641837981

5.  The inlier warning was referenced recently in a reply on another of your cases:

https://github.com/IntelRealSense/librealsense/issues/8729#issuecomment-812850472

closed 2021-04-21 by MartyG-RealSense