realsenseai/realsense-ros #1794: Run realsense D455 on docker

Hiroaki-K4, 2021-04-07:
Hi.
Is there an official image for running realsense D455 on docker? First of all, I want to confirm that realsense-viewer can be run on docker.
Also, please let me know if there is a Dockerfile that you can refer to.

MartyG-RealSense, 2021-04-07:
Hi @Hiroaki-K4   My understanding is that there is not an official Dockerfile for RealSense, though RealSense users have developed their own.

The discussions in the two links below may be useful to you in regard to use of RealSense with Docker on Jetson Xavier.  In the first of the links, from Novemver 2020, the contents of a Dockerfile are shared.

https://github.com/IntelRealSense/librealsense/issues/7849
https://github.com/IntelRealSense/realsense-ros/issues/1287

In the link below, a couple of RealSense ROS users had success with Docker when running it with Privileged Mode = True.

https://github.com/IntelRealSense/realsense-ros/issues/1104

And a link to a Jetson Dockerfile for ROS1 was shared here in September 2020:

https://github.com/IntelRealSense/librealsense/issues/6436#issuecomment-690791576

Hiroaki-K4, 2021-04-08:
ok, Thanks!

closed 2021-04-08 by Hiroaki-K4