realsenseai/realsense-ros #1795: why I get nothing when I echo '/camera/aligned_depth_to_color/image_raw'

wozxfdha, 2021-04-08:
hello,

I try to get the depth image from realsense d435i, so I conduct the commands below:

`roslaunch realsense2_camera rs_rgbd.launch`

when I echo topic `/camera/color/image_raw`, I get data successfully.

However, when I echo topic `/camera/aligned_depth_to_color/image_raw`, nothing output.

Could you tell me what causes this problem and how to solve it?

MartyG-RealSense, 2021-04-08:
Hi @wozxfdha  Did you include the term **align_depth:=true** in your roslaunch instruction to publish additional topics aligned to the depth, please?

**roslaunch realsense2_camera rs_rgbd.launch align_depth:=true**

wozxfdha, 2021-04-08:
@MartyG-RealSense  Thanks for your reply.

After I conduct `roslaunch realsense2_camera rs_rgbd.launch align_depth:=true`

The problem still exists.

wozxfdha, 2021-04-08:
@MartyG-RealSense  More details are below.

> `$ roslaunch realsense2_camera rs_rgbd.launch align_depth:=true
... logging to /home/robot/.ros/log/7a274b0a-984c-11eb-94b9-5cc5d4d4302d/roslaunch-robot-ThinkPad-T440p-11154.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://robot-ThinkPad-T440p:35935/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
[... 67 lines cut ...]
[ INFO] [1617873981.307560337]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1617873981.307739763]: num_filters: 0
[ INFO] [1617873981.307755871]: Setting Dynamic reconfig parameters.
[ INFO] [1617873981.371390619]: Done Setting Dynamic reconfig parameters.
[ INFO] [1617873981.376538297]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1617873981.376768588]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ WARN] [1617873981.376838221]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 640, Height: 480, FPS: 30
[ INFO] [1617873981.383317661]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1617873981.385382014]: setupPublishers...
[ INFO] [1617873981.388424572]: Expected frequency for depth = 30.00000
[ INFO] [1617873981.419272256]: Expected frequency for infra1 = 30.00000
[ INFO] [1617873981.447672667]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1617873981.474283177]: Expected frequency for color = 30.00000
[ INFO] [1617873981.499003461]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1617873981.530063942]: setupStreams...
[ INFO] [1617873981.531578677]: insert Depth to Stereo Module
[ INFO] [1617873981.531833953]: insert Color to RGB Camera
[ INFO] [1617873981.531888972]: insert Infrared to Stereo Module
[ INFO] [1617873981.531926102]: insert Gyro to Motion Module
[ INFO] [1617873981.531963565]: insert Accel to Motion Module
[ INFO] [1617873981.567776507]: SELECTED BASE:Depth, 0
[ INFO] [1617873981.583534475]: RealSense Node Is Up!
 08/04 17:26:21,671 WARNING [140056039118592] (sensor.cpp:962) HID timestamp not found, switching to Host timestamps.
 08/04 17:26:21,707 WARNING [140055662294784] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
 08/04 17:26:22,011 WARNING [140056019859200] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.`

wozxfdha, 2021-04-08:
I figure it out.

This problem is caused by the usb port, when I plug it to usb3.0 port, it is solved (I use usb2.0 port before).

closed 2021-04-08 by wozxfdha

MartyG-RealSense, 2021-04-08:
Great to hear that you found a solution @wozxfdha - thanks for the update!