realsenseai/realsense-ros #1796: In building source of 2.2.23, 'error: ‘RS2_OPTION_SEQUENCE_ID’ was not declared in this scope'

bigbellmercy, 2021-04-08:
Hello, when building the source code for the version of 2.2.23, the errors similar with the following happen.
But it is not happened for the version of 2.2.22.

### Error
```
...
[ 93%] Built target realsense2_camera_generate_messages
/home/user/catkin_ws2/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)’:
/home/user/catkin_ws2/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:564:60: error: ‘RS2_OPTION_SEQUENCE_ID’ was not declared in this scope
                 if (module_name == "stereo_module" && i == RS2_OPTION_SEQUENCE_ID)
                                                            ^~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_ws2/src/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:564:60: note: suggested alternative: ‘RS2_OPTION_SHARPNESS’
                 if (module_name == "stereo_module" && i == RS2_OPTION_SEQUENCE_ID)
                                                            ^~~~~~~~~~~~~~~~~~~~~~
...
```

### Environment
realsense-ros 2.2.23 source build from Github
cakin_make command
Ubuntu 18.04 x86
Laptop PC with Nvidia

MartyG-RealSense, 2021-04-08:
Hi @bigbellmercy  Which librealsense SDK version are you using, please?  ROS wrapper version 2.2.23 should be used with librealsense **2.43.0**, as it is important to match the wrapper version as closely as possible with the specific librealsense SDK version listed in the release notes for a particular ROS wrapper version.

https://github.com/IntelRealSense/realsense-ros/releases

Also, if the librealsense SDK version is changed after the ROS wrapper has been built then the ROS wrapper should be rebuilt using the wrapper version designed for that SDK version.

bigbellmercy, 2021-04-08:
Thanks @MartyG-RealSense, here is the version info:
```
[ INFO] [1617872191.992358528]: RealSense ROS v2.2.22
[ INFO] [1617872191.992476906]: Built with LibRealSense v2.35.2
[ INFO] [1617872191.992515224]: Running with LibRealSense v2.44.0
```
I see. After I match the versions I'll inform the result. Thanks.

bigbellmercy, 2021-04-09:
@MartyG-RealSense After I matched the versions as guided as below, the build error disappeared. So, the issue is solved. Thank you!
```
[ INFO] [1617954168.562245881]: RealSense ROS v2.2.23
[ INFO] [1617954168.562349351]: Built with LibRealSense v2.43.0
[ INFO] [1617954168.562393710]: Running with LibRealSense v2.43.0
```

MartyG-RealSense, 2021-04-09:
Great news @bigbellmercy - thanks very much for the update!

MartyG-RealSense, 2021-04-15:
Case closed due to successful outcome and no further comments received.

closed 2021-04-15 by MartyG-RealSense