realsenseai/realsense-ros #1797: Use method2 to install realsense-ros in windows get problem

Twilight89, 2021-04-08:
Hi~ I try to install  realsense-ros in windows, and I have installed SDK2.0 in windows. When I run catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release , it get errors:
NMAKE : fatal error U1077: “P:\VisualStudio2017\VC\Tools\MSVC\14.16.27023\bin\Hostx64\x64\cl.exe”: 返回代码“0x2”
Stop.
NMAKE : fatal error U1077: “P:\VisualStudio2017\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe”: 返回代码“0x2”
Stop.
NMAKE : fatal error U1077: “P:\VisualStudio2017\VC\Tools\MSVC\14.16.27023\bin\HostX64\x64\nmake.exe”: 返回代码“0x2”
Stop.
Invoking "nmake" failed
(Inside 返回代码=return code)
By the way, in windows I can't run this command:
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
because no grep and tail command in windows.
Can you help me? I have got into big trouble.
Hope for your reply soon. Thanks~

MartyG-RealSense, 2021-04-08:
Hi @Twilight89   Yes, SDK 2.43.0 and ROS wrapper 2.2.23 are a compatible match.

As a starting point in investigating your case, may I confirm please if you have installed **ROS** on Windows before you began installing the RealSense ROS wrapper (2.2.23).  

https://wiki.ros.org/Installation/Windows

Twilight89, 2021-04-08:
@MartyG-RealSense 
Thanks for your reply~ 
I am pretty sure that I have installed ros on windows.
For more detail:
[image]
I'm sorry there are Chinese words inside, I don't know how to transfer to english.
Hope reply, sincerely.

MartyG-RealSense, 2021-04-08:
Which version of ROS are you using please?  There are separate wrappers for ROS1 and ROS2 with different instructions.

**ROS1 (Kinetic, Melodic, Noetic**
https://github.com/IntelRealSense/realsense-ros

**ROS2 (Dashing, Eloquent, Foxy)**
https://github.com/IntelRealSense/realsense-ros/tree/ros2

If you are using ROS1 then you would use the **2.2.23** ROS1 wrapper with SDK 2.43.0.

If you are using ROS2 then you would use the **3.1.5** ROS2 wrapper with SDK 2.43.0.

Twilight89, 2021-04-09:
@MartyG-RealSense 
Sorry for the delay and thanks for your reply. 
I have installed ros1 melodic on windows, which use the 2.2.23 ROS1 wrapper with SDK 2.43.0. In this part I think there is no problem.
[image]
So I'm so confused about my problem and search all issues about windows, but no one seems to me.
Really need your help. Thanks~

MartyG-RealSense, 2021-04-09:
The link below has a guide to installing ROS on Windows and running roscore.

https://janbernloehr.de/2017/06/10/ros-windows

Your roscore launch output in the image above is almost the same as the roscore launch output shown in an image in the guide in the section titled **Run a simple test**.  So that would suggest to me that your ROS Melodic installation on Windows is okay.

So let's look again at the **Method 2** process for installing RealSense ROS wrapper 2.2.23.

You confirmed in your first message in this discussion that you have installed the librealsense SDK, which is required for using the RealSense ROS wrapper.

Are you using the **Windows Command Prompt** terminal interface to enter the commands to set up the ROS wrapper as described in the guide linked to below?

https://github.com/IntelRealSense/realsense-ros/issues/1366#issuecomment-687628860

The Command Prompt is the interface that can be found by typing the word **command** into the text box at the bottom of the Windows screen, next to the corner button where the PC is shut down.

Twilight89, 2021-04-10:
@MartyG-RealSense 
Hi, 
I am sure that I have installed ros melodic on windows because I can run turtlesim demo. And I didn't use WSL which is windows subsystem linux.

In fact, if you just use "cmd" to run, there will be mistake that couldn't find ros and c++ compiler. 

So I create one like bellow, just like a command window in ubuntu.

[image]

In the red rectangle:
C:\Windows\System32\cmd.exe /k "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\Tools\VsDevCmd.bat" -arch=amd64 -host_arch=amd64 && c:\opt\ros\melodic\x64\setup.bat
I think it tells that this file is "C:\Windows\System32\cmd.exe", and it will use c++ compiler in "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\Tools\VsDevCmd.bat". And set environment for ros by "c:\opt\ros\melodic\x64\setup.bat".

So I think this part is also no problem. I pretty curious that can you guys run this repo(realsense-ros) on windows ?
Because In you tutorial for installing on windows, there is a command "git checkout git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1" which can not run on windows. And in [this issue](https://github.com/IntelRealSense/realsense-ros/issues/1366#issuecomment-687628860), this guy also met this problem, but the solution is that giving up on windows.

I am very sad about this problem, exhaust me a lot, spending whole day but can't fix it.
Hope for your help, sincerely.

MartyG-RealSense, 2021-04-10:
I recall a case where a RealSense ROS user bypassed the grep problem by replacing the checking equation with simply the value of the ROS wrapper version that they were using.   For example in your case:

**git checkout 2.2.23**

https://github.com/IntelRealSense/realsense-ros/issues/1322#issuecomment-670988626

Twilight89, 2021-04-10:
@MartyG-RealSense 
Hi, 
Thanks for reply. I ran this command git checkout 2.2.23：
[image]

But subsequently run catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release, it got error same as my second message.  

Maybe git checkout 2.2.23 is not the solution.  
Could you install this repo on windows to see whether this problem would happen to you ?
Sincerely.

MartyG-RealSense, 2021-04-10:
I am not set up on my computer to perform an installation myself.  Let's check through the installation steps in the instructions.

Before entering the catkin_make command, have you cloned the ROS dependency **ddynamic_reconfigure** into your catkin_ws/src folder from the link below:

https://github.com/pal-robotics/ddynamic_reconfigure/tree/kinetic-devel

Also, after you entered the git checkout 2.2.23 command, did you then input **cd ..** as mentioned in the instructions to navigate one directory backwards before entering the **catkin_init_workspace** command?

Twilight89, 2021-04-10:
@MartyG-RealSense 
1. I am pretty sure that I have clone ddynamic_reconfigure. Because at first I did't clone this one, it got another error, so I read the README.MD carefully. And I find this procedure. 
[image]
2. I have run catkin_init_workspace. So in src file, there is a CMakeLists.txt. 
3. Yep, I know catkin_make clean and catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release should run in catkin_ws file. 
So, these parts seem no problem.
I have checked these procedure for 3-4 times.
Sincerely.

MartyG-RealSense, 2021-04-10:
Other RealSense ROS users have had problems with setting the **-DCATKIN_ENABLE_TESTING=False** term.  One person got past the error by using True to permit testing.

Twilight89, 2021-04-10:
Thanks for advice.
But it didn't work for me. It just got the same error. And I directly run catkin_make, also got the same error.
Does anyone install realsense-ros on windows successfully?

MartyG-RealSense, 2021-04-10:
I do not personally recall a case where someone has clearly, without any room for doubt, confirmed that they succeeded in setting up the RealSense ROS wrapper on Windows and could not find such a case by researching it.  As it is supported in the installation instructions though, I have to assume that it is possible.

Twilight89, 2021-04-10:
@MartyG-RealSense Ok, I don't think I have enough time to solve this weird problem.

Twilight89, 2021-04-10:
Ok, I don't think I have enough time to solve this weird problem.
If anyone meets this problem and has a work around, please @ me.
Thanks again for your patience and time.

MartyG-RealSense, 2021-04-10:
You are very welcome @Twilight89

MartyG-RealSense, 2021-04-16:
Case closed due to no further comments received.  

If future readers of this case have advice for @Twilight89 then please feel free to leave a comment.

closed 2021-04-16 by MartyG-RealSense

TobiasVishchers, 2021-06-25:
I had the same problem (Win10, ROS Melodic, SDK 2.45 and ROS wrapper 2.3.0).
I still cant install the package, but was able to get one step further than @Twilight89.
As shown in Twilight89s screenshot there were some undeclared identifiers "RS2_OPTIONS_...".
Those are **missing if you install via vcpkg but are availabe if you install the SDK** per [binary installer](https://github.com/IntelRealSense/librealsense/releases/download/v2.45.0/Intel.RealSense.SDK-WIN10-2.45.0.3212.exe).

So I copied the include folder from the sdk (C:\Program Files (x86)\Intel RealSense SDK 2.0\include)
and replaced the one from vcpkg (C:\opt\ros\melodic\x64\tools\vcpkg\installed\x64-windows\include)
(dont rosdep install, because that will override vcpkg again with the wrong files)

CMake gets to 100% now but fails while linking the shared library realsense2_camera.dll

[image]

TobiasVishchers, 2021-06-25:
"vcpkg list" tells me that it installed realsense2:x64-windows 2.34.0.
Could it be a solution to use an older ROS wrapper?
There doesn't seem to be a Wrapper version which targets sdk 2.34.0 though.
Build 2.2.12 supports SDK v2.35.2 and Build 2.2.13 supports SDK v2.33.1.

MartyG-RealSense, 2021-06-25:
@TobiasVishchers  My recollection from the time that these SDKs were released is that it was okay to use the wrapper for 2.33.1 with 2.34.0 because there were not significant changes between SDKs 2.33.1 and 2.34.0.

Having said that, I usually recommend avoiding SDK 2.34.0 because it had an issue that generated continuous timing errors for some RealSense users.  If you do not experience that error yourself then it should be okay to use it though.

https://github.com/IntelRealSense/librealsense/issues/6189#issuecomment-607918458