realsenseai/realsense-ros #1798: Cannot detect device

jacklu333333, 2021-04-08:
I am using Intel Realsense L515 for ros system. 
However, I kept getting 

`No RealSense devices were found!`
 
In my `lsusb`

> Bus 002 Device 028: ID 8086:0b64 Intel Corp. 
> Bus 002 Device 025: ID 05e3:0626 Genesys Logic, Inc. 
> Bus 002 Device 024: ID 05e3:0626 Genesys Logic, Inc. 
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
> Bus 001 Device 077: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
> Bus 001 Device 010: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
> Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The first device is the Intel realsense L515.

I use a raspberry pi as the processor and with an external hub with a power supply. Therefore, power should not be the problem.

And I set up my workspace by using
[librealsense](https://github.com/IntelRealSense/librealsense)
[realsense-ros](https://github.com/IntelRealSense/realsense-ros)

and I compile it with 
`catkin_make_isdolated`

run it with 
`roslaunch realsense2_camera rs_camera.launch`

RealSenseSupport, 2021-04-21:
Hi @jacklu333333 

Is there anyway you can run the RealSense Viewer app/tool and provide some information as to what it is recognizing as far as RealSense device being available on the system?

Is there another computer, maybe a Windows based system that you can plug into and see if it's recognized there?

jacklu333333, 2021-04-22:
Hi, I just solve the problem by reinstall the whole OS. I am sure that it wasn't the reason for the hardware because on my laptop it works fine, and I couldn't locate the reason why the error. Still, thank you for your help.

Best Regards.

closed 2021-04-22 by jacklu333333