realsenseai/realsense-ros #1799: Best and cheapest extended USB 3.0 cable

ranjitkathiriya, 2021-04-08:
Hello there,

I need a 3.0 USB to type C cable for my intel real sense D435i. Can you please! suggest some links where I can buy It?

I am living in Ireland.

MartyG-RealSense, 2021-04-08:
Hi @ranjitkathiriya   Based on known support for RealSense and past feedback from RealSense users, **Newnex** and **OCP** are worth researching.

Because a high quality cable is important when using a long cable with RealSense cameras, purchasing a cheap cable can be a false economy as it may have performance degradation or no connection at all when transmitting over cable length of 2 meters or more.

**Newnex**
https://www.newnex.com/realsense-3d-camera-connectivity.php

https://youtube.com/watch?v=GLQgR1jT04M

**OCP**
https://www.ocp.com/news/intel-realsense-d435-and-d415-camera-cables/

ranjitkathiriya, 2021-04-09:
Thanks for reply @MartyG-RealSense

closed 2021-04-09 by ranjitkathiriya