realsenseai/realsense-ros #1800: issue about librealsense install by rosdep

JaehyunShim, 2021-04-09:
Hi, 

If I used rosdep to install ibrealsense2 (https://github.com/IntelRealSense/realsense-ros/blob/foxy/realsense2_camera/package.xml#L20), and got the following error code.
```sh
[realsense2_camera_node-1] [WARN] [1617945021.422944987] [RealSenseCameraNode]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-1] [ERROR] [1617945021.422971398] [RealSenseCameraNode]: The requested device with  is NOT found. Will Try again.
```

which I didn't see when I installed it following instruction
https://github.com/IntelRealSense/realsense-ros/tree/foxy#step-2-install-librealsense2

Maybe there has been changes in the librealsense2 code so it has to be rebloomed? 

Thanks!

RealSenseSupport, 2021-04-23:
@JaehyunShim Do you mean you get the error when install by rosdep? Or you got the error when run ros launch after installation by rosdep ? Looking forward to your reply. Thanks!

JaehyunShim, 2021-04-23:
@RealSenseSupport 

The latter one.

RealSenseSupport, 2021-04-25:
@JaehyunShim Did you still have the full log if you run ros launch after installation by rosdep? Or did you notice the librealsense version is the same or not for two installation methods?

JaehyunShim, 2021-04-25:
@RealSenseCustomerSupport  I do not have the log now but I tried both ways and that is what I got. You can just follow the instruction you wrote, install the packages and run them, then you will see the bug.

RealSenseSupport, 2021-04-27:
@JaehyunShim Thanks for your reply! Will have a try on my side.

JaehyunShim, 2021-04-28:
@RealSenseSupport Thanks a lot!

RealSenseSupport, 2021-05-18:
@JaehyunShim Any update from your side? Thanks!

JaehyunShim, 2021-05-18:
@RealSenseSupport 

Thank you for your reply. I do not think I can spend more time on this issue. If the bug still exists, I believe someone else will open a new thread and ask for help. Thank you for your support.

closed 2021-05-18 by JaehyunShim