realsenseai/realsense-ros #1803: L515 unstable

NorwegianPi, 2021-04-09:
Hello,

I have some issues with the L515. 
I am not able to integrate it rtabmap_ros along side my T265. At first i suspected that it was a config issue, but I was able to get hold of another L515 to test with. 

The new camera worked with RTABMAP strait away without any changes to the config.

Both cameras are running FW: 01.05.04.01 (Updated using windows SDK)

The new camera worked as expected for 1 day, the next day, i some times got connection error but re-connecting the camera seemed to do the trick. 
Today I am however not able to get a stable connection with rtabmap_ros (same cables and ports). 

I get it to update the map 60 or so times before it seems to stop.

If i try to relaunch rtabmap without stopping thr realsense node, I get this message:
 
```
[ WARN] [1617886497.652308693]: /rtabmap_l515/rgbd_sync: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap_l515/rgbd_sync subscribed to (exact sync):
   /l515/color/image_raw \
   /l515/aligned_depth_to_color/image_raw \
   /l515/color/camera_info
```

When it is not working at all i some times get this warning from realsense_ros:
```
[ WARN] [1617973602.106265942]: No stream match for pointcloud chosen texture Process - Color
```

I have checked these topics and data is published to them.

If realsense_ros is restarted rtabmap resumes map update for a short while.

I have similar setup and configuration for D435 and T265 which is stable.

Any suggestions on how to debug this?

| System   | |
| ------------- | ------------- |
| Ubuntu 18.04  | Intel NUC  |
| RTABMAP | built from source  |
| Real Sense | built from source  |
| IntelNuc | built from source  |
| ROS Melodic | apt  |

**Realsense Launch File**
```xml
<launch>
  <arg name="device_type_t265"    		default="t265"/>
  <arg name="device_type_l515"    		default="l515"/>
  
  <arg name="serial_no_t265"    			default="948422110768"/>
  <arg name="serial_no_l515"    			default="f0190011"/>
  
  <arg name="t265_ns"              			default="t265"/>
  <arg name="l515_ns"              			default="l515"/>

  <arg name="tf_prefix_t265"         default="$(arg t265_ns)"/>
  <arg name="tf_prefix_l515"         default="$(arg l515_ns)"/>

  <arg name="initial_reset"             default="false"/>
  <arg name="enable_fisheye"            default="false"/>

  <arg name="color_width"               default="1280"/>
  <arg name="color_height"              default="720"/>

  <arg name="depth_width"               default="640"/>
  <arg name="depth_height"              default="480"/>

  <arg name="ordered_pc"                default="false"/>
[... 793 lines cut ...]
[ INFO] [1617973860.491001089]: /rtabmap_l515/rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1617973860.491016178]: /rtabmap_l515/rtabmap: subscribe_odom_info = false
[ INFO] [1617973860.491026454]: /rtabmap_l515/rtabmap: subscribe_user_data = false
[ INFO] [1617973860.491038907]: /rtabmap_l515/rtabmap: subscribe_scan = false
[ INFO] [1617973860.491053988]: /rtabmap_l515/rtabmap: subscribe_scan_cloud = false
[ INFO] [1617973860.491076496]: /rtabmap_l515/rtabmap: subscribe_scan_descriptor = false
[ INFO] [1617973860.491090275]: /rtabmap_l515/rtabmap: queue_size    = 10
[ INFO] [1617973860.491103178]: /rtabmap_l515/rtabmap: approx_sync   = true
[ INFO] [1617973860.491223937]: Setup depth callback
[ INFO] [1617973860.553273372]: 
/rtabmap_l515/rtabmap subscribed to (approx sync):
   /t265/odom/sample \
   /l515/color/image_raw \
   /l515/aligned_depth_to_color/image_raw \
   /l515/color/camera_info
[ INFO] [1617973861.025705883]: rtabmap (1): Rate=0.20s, Limit=0.000s, Conversion=0.0050s, RTAB-Map=0.3133s, Maps update=0.0008s pub=0.0002s (local map=1, WM=1)
[ INFO] [1617973861.285572424]: rtabmap 0.20.9 started...
...
...
...
[ INFO] [1617973888.569313459]: rtabmap (82): Rate=0.20s, Limit=0.000s, Conversion=0.0014s, RTAB-Map=0.3086s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)
[ INFO] [1617973888.945341897]: rtabmap (83): Rate=0.20s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.3183s, Maps update=0.0001s pub=0.0000s (local map=1, WM=1)

then nothing...
```

mirkocomparetti-synesis, 2021-04-15:
Are you using the RaspberryPi to handle the camera?
I'm facing an issue with the same camera in ros and the raspberrypi4 and I'm wondering if you are using the same hardware https://github.com/IntelRealSense/realsense-ros/issues/1804

NorwegianPi, 2021-04-15:
@mirkocomparetti-synesis No. I am running it on a Intel NUC with Ubuntu 18.04.

NorwegianPi, 2021-04-15:
New observation: one of the L515s are detected as USB 2.1 or 3.2 depending on orientaton of the cable. This happens with multiple cables and only on one of my L515s

mirkocomparetti-synesis, 2021-04-16:
> @mirkocomparetti-synesis No. I am running it on a Intel NUC with Ubuntu 18.04.

Thanks for the confirmation!

NorwegianPi, 2021-05-04:
@RealSenseSupport any ideas what might be causing this issue?

RealSenseSupport, 2021-05-06:
Hi @NorwegianPi 

Looking at your new observation with one of the cameras showing differences based on the orientation of the USB connector, I suggest submitting for RMA either on the intelrealsense.com webstore or distributor that you may have purchased the camera from.

closed 2021-06-09 by RealSenseSupport