realsenseai/realsense-ros #1804: L515 stopping after few seconds

mirkocomparetti-synesis, 2021-04-09:
Hello,
We have a raspberrypi 4 8GB connected to the camera with ubuntu and the libraries have been manually compiled. We installed ros noetic and we are trying the camera connection and this happens

```
$ roslaunch realsense2_camera rs_camera.launch align_depth:=true color_width:=1920 color_height:=1080 color_fps:=15
... logging to /home/user/.ros/log/d96ac9dc-992f-11eb-9dd8-df83f75e12bb/roslaunch-ubuntu-6694.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberry:43837/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 1080
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1920
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
[... 113 lines cut ...]
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static
```
and they disappear when `[camera/realsense2_camera-2] process has finished cleanly` happens.

From the log file reported, we have
```
$ cat /home/user/.ros/log/d96ac9dc-992f-11eb-9dd8-df83f75e12bb/camera-realsense2_camera-2*.log
[ INFO] [1617974741.789083722]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
[ INFO] [1617974741.809774677]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1617974741.838669482]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.
[ INFO] [1617974755.214000829]: Bond broken, exiting
```

Do you have a clue why the process is terminated after few seconds (<15s)?

Is there anything we are missing?

Thank you!

mirkocomparetti-synesis, 2021-04-12:
Just to add something: using ubuntu on a standard desktop machine works like a charm.
Is there something we are missing while building the libraries for arm?
We are using the ubuntu server version as we will use the camera data "remotely" for the raspberrypi; can this be the culprit?

mirkocomparetti-synesis, 2021-04-13:
Hello, @MartyG-RealSense
any idea on why this happens on the raspberry pi?
I did follow the instructions from [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) for the LTS distrubution of ubuntu (20.04) and compiled with `cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_BINDINGS:BOOL=ON -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false -DCHECK_FOR_UPDATES:BOOL=ON -DBUILD_SHARED_LIBS:BOOL=ON -DBUILD_NODEJS_BINDINGS:BOOL=OFF -DBUILD_NETWORK_DEVICE:BOOL=ON -DENFORCE_METADATA:BOOL=ON -DFORCE_LIBUVC:BOOL=ON`.

I do can't understand why the proces stops cleanly after few seconds.

Or maybe you can point me to some directions to understand.

Thank you,
Best

mirkocomparetti-synesis, 2021-04-15:
Sorry to tag you all @MartyG-RealSense @RealSenseCustomerSupport @RealSenseSupport 
Do you have any idea about this situation? We are stuck and cannot proceed with the development with the raspberry pi as it does not stay up for more than 10s.
If I run the rs-server on the same pi, it work fine and I can connect remotely with the viewer.

RealSenseSupport, 2021-04-27:
Hi @mirkocomparetti-synesis 

We need to look at this further.  You say that you're able to run the L515 rs-server and connect to the viewer it works fine.  On the surface nothing looks out of the ordinary.  It seems to be issue with building.

mirkocomparetti-synesis, 2021-04-28:
Hi @RealSenseSupport 
Yes, it might be a compilation error. Even if we build with those options
```cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_PYTHON_BINDINGS:BOOL=ON -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false -DCHECK_FOR_UPDATES:BOOL=ON -DBUILD_SHARED_LIBS:BOOL=ON -DBUILD_NODEJS_BINDINGS:BOOL=OFF -DBUILD_NETWORK_DEVICE:BOOL=ON -DENFORCE_METADATA:BOOL=ON -DFORCE_LIBUVC:BOOL=ON```
this is what we get at the end
[image]
and we cannot use python.
Your support is crucial for this.
Best

RealSenseSupport, 2021-05-10:
@mirkocomparetti-synesis Did you get the streaming stop issue if you run with [rs-depth](https://github.com/IntelRealSense/librealsense/tree/master/examples/C/depth) or [rs-color](https://github.com/IntelRealSense/librealsense/tree/master/examples/C/color) ?  
And also you can have a try with "align_depth:=false" and lower the resolution to see if the issue still existing.

mirkocomparetti-synesis, 2021-05-11:
@RealSenseSupport I didn't try rs-depth or rs-color, but rs-server works fine without stop. The problem lies in the python library that is not compiled and it is missing a loto of methods (like the pipeline method, see https://github.com/IntelRealSense/realsense-ros/issues/1804#issuecomment-828210420)

RealSenseSupport, 2021-05-18:
@mirkocomparetti-synesis Did you get chance to try? Looking forward to your update. Thanks!

mirkocomparetti-synesis, 2021-05-18:
I didn't have the chance to try in the last week. Anyway, the cmake flags you suggested are already active in my build set here https://github.com/IntelRealSense/realsense-ros/issues/1804#issuecomment-837954731, so those will not make a difference.
Please, suggest possible resolution values even if we need the maximum resolution and reducing it is not realy an option for us.

Our use case is the following RaspberryPi running a "server" application to stream the data over the network and a different computer running a python client application that will read those data and process them. What is your suggestion to achieve that?
ROS is an option for that (if it would have worked in the first place) but any other working and tested option will be fine for us (considering the client must run python code on the camera data).

RealSenseSupport, 2021-05-18:
@mirkocomparetti-synesis You issue seems similar to https://github.com/IntelRealSense/realsense-ros/issues/1758#issuecomment-840285347  Could you please try to configure  _pointcloud_texture_stream_ from RS_STREAM_COLOR (default) to RS_STREAM_ANY in the launch file?

For your use case, not sure if you have checked our white paper in https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras ? Please check if this helps.

mirkocomparetti-synesis, 2021-05-18:
Sure, we read that white paper and this is what we found out
1. The rpi image does not work properly
2. python does not support network camera
3. c++ examples didn't really compile in our tests
4. we didn't change the launch file, but the same ros node works in a ubuntu desktop machine, so I do not think is a launch file problem but it is mostly related to the build process on arm

mirkocomparetti-synesis, 2021-05-20:
@RealSenseSupport as suggested we did test with the `RS_STREAM_ANY` and `align_depth:=false`, with no luck
```
$ roslaunch realsense2_camera rs_camera.launch align_depth:=false color_width:=1280 color_height:=720 color_fps:=15
... logging to /home/ubuntu/.ros/log/762488a8-b974-11eb-bbe2-ffafe61ceab8/roslaunch-synesiscmr01-54716.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://test:43295/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 720
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 1280
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 74 lines cut ...]
[ INFO] [1621520289.041066725]: setupPublishers...
[ INFO] [1621520289.052115505]: Expected frequency for depth = 30.00000
[ INFO] [1621520289.065257349]: Expected frequency for color = 15.00000
[ INFO] [1621520289.077324513]: setupStreams...
[ INFO] [1621520289.101742303]: insert Depth to L500 Depth Sensor
[ INFO] [1621520289.102001616]: insert Color to RGB Camera
 20/05 14:18:09,172 WARNING [281472239501680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/05 14:18:09,223 WARNING [281472239501680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1621520289.226432812]: SELECTED BASE:Depth, 0
 20/05 14:18:09,275 WARNING [281472239501680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/05 14:18:09,326 WARNING [281472239501680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1621520289.379828312]: RealSense Node Is Up!
[ WARN] [1621520289.400927044]: 
[ WARN] [1621520289.401115302]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 20/05 14:18:09,883 WARNING [281472239501680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/05 14:18:10,227 WARNING [281472239501680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 20/05 14:18:13,982 WARNING [281472239501680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/05 14:18:15,234 WARNING [281472239501680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/05 14:18:16,486 WARNING [281472239501680] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/ubuntu/.ros/log/762488a8-b974-11eb-bbe2-ffafe61ceab8/camera-realsense2_camera-2*.log
```

mirkocomparetti-synesis, 2021-05-27:
@RealSenseSupport is this https://github.com/IntelRealSense/librealsense/issues/9014#issuecomment-841533559 the case? Are you sure about that as rs-server works for me?
Is the L515 supported somehow?

RealSenseSupport, 2021-05-27:
@mirkocomparetti-synesis Sorry. As the white paper mentioned that "The goal of this project is to offer an alternative solution that allows users to network existing Intel® RealSense™ depth cameras D415, D435 and D435i (and in future more Intel® RealSense™ cameras)", rs-server can't work for L515 yet.  

For your issue with ROS, I'll try to find one Raspberry setup to see if the issue can be reproduced. Will update your later. Thanks!

RealSenseSupport, 2021-06-02:
@mirkocomparetti-synesis I didn't get L515 stop streaming issue with RealSenseViewer on Raspberry based on SDK v2.45. Could you please also check if you get issue when use RealSenseViewer? 

Next I'll try to setup ROS environment and test if the issue can be reproduced.

mirkocomparetti-synesis, 2021-06-03:
Hello @RealSenseSupport,
No, realsense viewer seems to work. Though we do not need it but we need to stram data from the raspberry pi over a network to a client that will need to read those data in python.
Any suggestion is appreciated.
Thanks!

mirkocomparetti-synesis, 2021-06-09:
Hello @RealSenseSupport
Do you have any news on this?

Thanks!

RealSenseSupport, 2021-06-09:
@mirkocomparetti-synesis These days I checked some related github tickets and found that some users can run rs-server with L515. And did you check https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server to see if this is what you're looking for? Thanks!

mirkocomparetti-synesis, 2021-06-14:
Thank you @RealSenseSupport 
Is rs-server with the L515 officially supported? I appreciate someone that managed to use it but I'd like to have an official statement.
I'll look at the link but reading the first part it states 

> Ubuntu 18.04 on an UpBoard but has also been tested on an Intel NUC

 which is far from our hardware and software (Ubuntu 20.04 on raspberry pi 4)

mirkocomparetti-synesis, 2021-06-14:
So the example has some weird things
```
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
ERROR: Could not find a version that satisfies the requirement pyrealsense2 (from versions: none)
ERROR: No matching distribution found for pyrealsense2
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
Collecting cron_descriptor
  Downloading cron_descriptor-1.2.24.tar.gz (22 kB)
Building wheels for collected packages: cron-descriptor
  Building wheel for cron-descriptor (setup.py) ... done
  Created wheel for cron-descriptor: filename=cron_descriptor-1.2.24-py3-none-any.whl size=37775 sha256=24abbd113e9a0eb50035b8a7355b9130ed686b449c8aa1e00f2634793541f379
  Stored in directory: /root/.cache/pip/wheels/7b/06/5f/2164c55afa10479505033599409f980cdca4daf096f41c40a1
Successfully built cron-descriptor
Installing collected packages: cron-descriptor
Successfully installed cron-descriptor-1.2.24
ERROR: Missing user or command in system cron line.
cron job set to run Every minute
```
As expected, it fails to download the pyrealsense2, as it does not exist for arm and we built it manually.
Also, in the manually built pyrealsense2, the pipeline object does not get compiled with arm, and that tool is the key for the ethernet example (as it requires the camera), as we requested here https://github.com/IntelRealSense/librealsense/issues/6820

Can you comment on all this?
Thanks

RealSenseSupport, 2021-07-21:
@mirkocomparetti-synesis Sorry for the late response. After double check, L515 is not currently supported in rs-server. Team is working on this and any update will let you know.

RealSenseSupport, 2021-09-17:
@mirkocomparetti-synesis Sorry to inform you that there's no plan to implement rs-server for L515 now. Any other question from your side? Thanks!

RealSenseSupport, 2021-10-11:
@mirkocomparetti-synesis Any other questions about this ticket? Please note that the ticket will be closed in another 7 days if we  don't get update from you. Thanks!