realsenseai/realsense-ros #1805: Can not change Image shape and size

Shunichi09, 2021-04-12:
I noticed that from version 2.2.22, the default image size became (height: 720, width: 1280) from (height: 480, width: 640) when I run this launch file

```launch
<launch>
    <group>
        <include file="$(find realsense2_camera)/launch/rs_camera.launch" >
            <arg name="align_depth" value="true" />
            <arg name="filters" value="disparity,spatial,temporal,decimation,hole_filling"/>
       </include>
    </group>
</launch>
```

Therefore, I would like to change my image size to old style (height: 480, width: 640), but /camera/image_raw/camera_info showed the shape was still (height: 720, width: 1280). 
How to fix this problem ??

``` launch
<launch>
    <group>
        <!-- https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md -->
        <include file="$(find realsense2_camera)/launch/rs_camera.launch" >
            <arg name="fisheye_width"       default="640"/>
            <arg name="fisheye_height"      default="480"/>
            <arg name="enable_fisheye"      default="false"/>
            <arg name="depth_width"         default="640"/>
            <arg name="depth_height"        default="480"/>
            <arg name="enable_depth"        default="true"/>
            <arg name="infra_width"        default="640"/>
            <arg name="infra_height"       default="480"/>
            <arg name="enable_infra1"       default="true"/>
            <arg name="enable_infra2"       default="true"/>
            <arg name="color_width"         default="640"/>
            <arg name="color_height"        default="480"/>
            <arg name="enable_color"        default="true"/>
            <arg name="align_depth" value="true" />
            <arg name="filters" value="disparity,spatial,temporal,decimation,hole_filling"/>
       </include>
    </group>
</launch>
```

I'm using realsense d435i and melodic.
Thank you for your help.

MartyG-RealSense, 2021-04-12:
Hi @Shunichi09   A discussion in the link below describes the changes from RealSense ROS wrapper 2.2.22 onward in regard to profiles.

https://github.com/IntelRealSense/realsense-ros/issues/

To set a custom configuration you should specify three parameters for each stream type that you are configuring: Width, Height and FPS.

Shunichi09, 2021-04-13:
Thank you for your comments !! 
I understood and I can change my image size by defining the fps as well as other parameters.

closed 2021-04-13 by Shunichi09

MartyG-RealSense, 2021-04-13:
Excellent news @Shunichi09 - thanks for the update!