realsenseai/realsense-ros #1808: Dropped frames when using network camera

sachin-pri, 2021-04-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 IP Device | 
| Firmware Version                   | 2.42.0 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0                                     | 
| Platform                                 | PC, Raspberry Pi  |
| SDK Version                            |  2.42.0                        | 

I have been trying to integrate the results of this tutorial (https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras) with ROS.

On completing the above tutorial, I am able to reliably view the data in the realsense-viewer with the camera configured to stream at 640x480x30fps.

I was able to modify realsense2_camera with the help of the comment: https://github.com/IntelRealSense/realsense-ros/issues/1501#issuecomment-727532956 by creating a network device and adding it to the context. I also had to make a few other modifications such as commenting out lines of code that were specific to USB devices alone. On doing so, I am currently able to run realsense2_camera and view the depth and color data in Rviz. However, I only get an average of 2Hz when I run rostopic hz on the camera/depth/image_rect_raw, and 10Hz on camera/color/image_rect_raw, despite having no such issues when viewing the data with the realsense-viewer as mentioned above. Are there any known reasons for why this might be happening?

MartyG-RealSense, 2021-04-13:
Hi @sachin-pri   I researched your question extensively.  The link below provides what I would consider the best starting point for investigating your problem.  In that case, involving a Pi slave and a laptop master on a local network, the Pi user found that altering image quality improved their publishing rate significantly.

https://answers.ros.org/question/289300/very-low-image-publish-rate-from-raspicam_node-over-wifi/

For RealSense ROS, the relevant setting to adjust image quality may be **/camera/image/compressed/jpeg_quality**

http://wiki.ros.org/image_transport#Parameters

sachin-pri, 2021-04-13:
Hi @MartyG-RealSense, thank you for your response. I modified the /camera/depth/image_rect_raw/compressed/jpeg_quality and the /camera/color/image_raw/compressed/jpeg_quality from 80 down to 10 using the dynamic_reconfigure method via rqt_reconfigure. I'm currently trying to stream the data on what I believe to be uncompressed topics (camera/depth/image_rect_raw and camera/color/image_raw topics), and I don't see any change in publishing frequency on them after making the quality modification. Does trying to modify the compression quality imply that I should instead work with compressed topics of type sensor_msgs/CompressedImage? My camera node seems to die as soon as I run rostopic hz on /camera/depth/image_rect_raw/compressed, so I'm not sure if this is the type of topic that I should be looking at either.

MartyG-RealSense, 2021-04-14:
Do the images in RViz look as though they are publishing at only 2Hz or are they moving at at 30FPS-like speed despite what rostopic hz says?

sachin-pri, 2021-04-14:
They look like they are publishing at 2Hz in Rviz as well. 2Hz is the average publishing rate; what happens is that there are long periods with no messages coming in, followed by a very short burst of frames. This is consistent when viewed both via rostopic hz and rviz.

MartyG-RealSense, 2021-04-15:
You are the first RealSense user that I am aware of who has adapted the ethernet system to ROS network protocols, so there is no reference material available to compare your setup with an established template.

Given that you are able to stream across the network at the correct speed when using the standard configuration in the white paper, it is probable that the problem is not related to the ethernet latency described in the paper.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-5-1-latency

The paper also suggests a pair of commands to run on the client-side that may increase network performance.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-4-3-tuning-linux-network-stack

sachin-pri, 2021-04-19:
Hi @MartyG-RealSense, I understand, thanks for the support. I don't plan to work on this for another few weeks. I'll add a comment if I have an update. Thanks!

MartyG-RealSense, 2021-04-20:
Okay, thanks very much @sachin-pri for the update - I will keep this case open for a further time period.

closed 2021-06-20 by MartyG-RealSense