realsenseai/realsense-ros #1809: D435i bias? Faulty hardware or expected?

shuhaowu, 2021-04-13:
I noticed that when I put the D435i (tilted in an angle), the magnitude of the acceleration is frequently over 9.81. The data below is averaged with 5 seconds worth of data:

```
+--------------------------+-----------+-----------+-----------+-----------+
| Frame (1202 data pts)    | a_x       | a_y       | a_z       | |a|       |
+--------------------------+-----------+-----------+-----------+-----------+
| camera_imu_optical_frame |  -0.30136 |  -9.60785 |  +2.90106 | +10.04081 |
| camera_link              |  +0.28893 |  +0.03001 |  +0.95688 | +10.04080 |
```

Here's a longer term plot (data averaged every 5 seconds):

[image]

It took more than half an hour to settle into a steady-state value, perhaps due to temperature effects. However, it never settled around 9.8, and instead settles around 9.9, which is about 1% of a difference. I can't seem to find the error specifications of the imu from the [datasheet (page 68)](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf). I'm wondering if I'm missing something, or perhaps I'm not specifying some configuration correctly?

MartyG-RealSense, 2021-04-13:
Hi @shuhaowu  When considering the accuracy of the D435i IMU's vertical acceleration, the value to focus on is Y Accel, which seems to be represented as a_y on your table.  A y-acceleration value of around 9.6 is what I would expect to see from the raw data of a D435i IMU.

A characteristic of an uncalibrated IMU is that the y-acceleration value may not be able to settle at zero even if the camera is stationary on a flat surface.

The IMU hardware component of the D435i does not have built-in internal calibration.  So the RealSense SDK provides an IMU calibration software tool. 

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration

Once a calibration of the IMU has been performed with the above tool then it can be written by the tool to the camera hardware.  This enables the **RealSense Viewer** tool to unlock an option in its Motion Module settings to correct the raw IMU data by adjusting for the known error variables in the IMU component.  This enables the y-acceleration value to be adjusted to be closer to the ideal target value of 9.81 instead of 9.6 or lower.

This feature is described and illustrated in the link below.

https://github.com/IntelRealSense/librealsense/issues/4751#issuecomment-527138458

************************

When accessing the IMU with **ROS** though, I believe that the raw uncorrected IMU data is used, resulting in the raw y-acceleration of 9.6 in your table.

shuhaowu, 2021-04-13:
> When accessing the IMU with ROS though, I believe that the raw uncorrected IMU data is used, resulting in the raw y-acceleration of 9.6 in your table.

That's interesting, I've actually performed that calibration and wrote the calibration result to the camera. Is there a way to access the calibrated result via ROS as that's how I am using the RealSense?

Also, the 9.6 value is also because the realsense is currently mounted with a pitch angle of about 16 deg. If I left the camera flat, the value settles around a value of 9.9 like the magnitude I showed above.

MartyG-RealSense, 2021-04-13:
I do not know of a way to use corrected IMU values in ROS.  I researched the subject but could not find any method for doing so.

shuhaowu, 2021-04-17:
I just did some test and I can see the value returned by realsense-ros matches up to the values I see from the realsense viewer with motion correction enabled. So it seems like the corrected IMU values are returned by the realsense ros node. I think I need to re-run my calibration after the device temperature has stabilized, as opposed to calibrating it when I started it from a cold state. I'll try that and report back on this issue.

MartyG-RealSense, 2021-04-24:
Hi @shuhaowu   Do you have another update that you can provide about your case, please?  Thanks!

shuhaowu, 2021-04-24:
I did a preliminary test a few days ago but it didn't seem to have helped with the drift. I did it quickly so it's possible that I did it wrong, so I'm waiting until I have sometime within the next day or two to try it again.

MartyG-RealSense, 2021-04-24:
Thanks very much for the update.  I will keep this case open for a further time period.  Good luck with your tests!

closed 2021-05-10 by MartyG-RealSense