realsenseai/realsense-ros #1811: Left IC2 Config error - Brand New D455

rajeshmanpat, 2021-04-13:
Timestamp: 1618307542026.293701
Severity: Error
Description: Left IC2 Config error

Received this error. The Camera worked find for a few days, then one day, just stopped. Tried everything. Using the standard cable that came with the Camera. Port is secured. Tried updating the firmware. The 2D cam just says No Frames received.

Please help.

MartyG-RealSense, 2021-04-14:
Hi @rajeshmanpat  It sounds as though the camera can be detected if you are able to start the stream in order for it to provide the message **No Frames Received**.  Is this correct please?  

Also, which stream gave this message - depth or color - and which program gave the message.  Was it the RealSense Viewer?

MartyG-RealSense, 2021-04-20:
Hi @rajeshmanpat  Do you require further assistance with this case, please?  Thanks!

rajeshmanpat, 2021-04-21:
Hi @MartyG-RealSense. That is correct. I checked with both realsense viewer and the ros wrapper. The message I am getting is  "No Frames Received" on both RGB color window as well as depth. Only the Motion Module seems to be working.

MartyG-RealSense, 2021-04-22:
Do you have problems with RGB and depth if you do not activate the Motion Module?  

There is a known issue where, if three streams are enabled simultaneously (RGB, Depth, Motion Module) then errors such as **No Frames Received** on the RGB stream can occur.  It does not matter which order the three streams are activated in, because enabling the Motion Module is always the trigger for the problem.  Whilst there are workarounds for this in C++ and Python programming, there is no fix for this issue in the RealSense Viewer.

rajeshmanpat, 2021-04-29:
Yes, RGB and depth also show the same message. Irrespective of which one I stream first.

MartyG-RealSense, 2021-04-29:
Do you mean that you receive No Frames Received on the RGB or the depth even if the Motion Module is not activated?

MartyG-RealSense, 2021-05-05:
Hi @rajeshmanpat   Do you require further assistance with this case, please?

If you do need further assistance: do you receive No Frames Received on the RGB or the depth even if the Motion Module is not activated?

rajeshmanpat, 2021-05-09:
Hi Marty, yes I receive the message "No Frames Received" even when Motion Module is not activated.

MartyG-RealSense, 2021-05-09:
Thanks very much for the confirmation.  Are you using Windows or Ubuntu, please?

MartyG-RealSense, 2021-05-09:
Could you try launching the SDK's **rs-capture** example program please, by using the instructions in the link below on Windows or by inputting **rs-capture** into the Ubuntu terminal.

https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

When used with a D455, this program should display RGB, depth and IMU streams as individual stream panels in the same window.  Can RGB and depth display correctly in this program please or does it only show the IMU streams?

rajeshmanpat, 2021-05-12:
I tried this both on windows and Ubuntu. On windows, the exe open a black window and closes without any further display. On Ubuntu I got the following message: "RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x557ea171f0): Frame didn't arrive within 15000"

MartyG-RealSense, 2021-05-12:
Since the Viewer and rs-capture both rely on OpenGL graphics, it would be useful to test a non-graphical example that provides only a text printout to test whether it is a graphics-related issue.  If it were related to graphics then a text-based example should be able to receive frames and print the information.

On Windows, could you run the **rs-hello-realsense** example program that provides a simple text readout of distance, please?

[image]

rajeshmanpat, 2021-05-13:
I ran the rs-hello-realsense example program, got this message "RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x5594ecc780):Frame didn't arrive within 15000"

MartyG-RealSense, 2021-05-13:
I went back to the start of the case and noticed the words "port is secured".  Are you using a screw-lock USB cable, please?  If you are, there was a recent D455 case (posted almost the same day as your case) where the camera's behavior changed negatively if the cable was screwed in to the small holes either side of the USB port on the side of the camera but the camera worked fine when the cable was unscrewed.

https://github.com/IntelRealSense/librealsense/issues/8830

rajeshmanpat, 2021-05-14:
No, I am not using any screw lock USB cable. I am using the standard cable that came with the D455 from Intel. I even changed the cable to D435i which is similar. I am getting same error in both the cables.

MartyG-RealSense, 2021-05-14:
When errors occur in the Viewer, have you tried going to the **More** option and selecting **Reset Hardware** from its menu to perform a hardware reset of the camera?

[image]

rajeshmanpat, 2021-05-15:
Yes, I have tried this a few times. This did not help.

MartyG-RealSense, 2021-05-15:
As it has been a month since the start of this case and the camera has been thoroughly tested during this investigation, I will seek advice from Intel about your problem.  Thanks again for your patience!

rajeshmanpat, 2021-05-15:
Thanks for your assistance. Hoping to get a replacement under warranty
soon. Our project is stuck at the moment.

-- 
Regards,

*Rajesh Manpat | iFuture Robotics | CEO*
 - Autonomous Mobile Robots for Smart Logistics
Web: http://www.ifuturerobotics.com | Mobile : +91 - 9844728454.

"*Infantry win battles,* *logistics win wars**!*"

DISCLAIMER: This email (including any attachments) is intended for the sole
use of the intended recipient/s and may contain material that is
CONFIDENTIAL COMPANY INFORMATION. The information may be bound by any Non
Disclosure Agreements signed at an organizational level and must not be
shared with any 3rd party without consent.

MartyG-RealSense, 2021-05-16:
Intel advised that you perform the official RMA return process with wherever you purchased the camera from.

sivaeinfo, 2021-05-18:
Actually i have tried to use the dope package in https://github.com/yehengchen/DOPE-ROS-D435

I have installed require packages and dependencies.
In this, they have given two command.
1.roslaunch realsense2_camera rs_rgbd.launch
2. roslaunch dope dope.launch

This gives following error

MartyG-RealSense, 2021-05-18:
Hi @sivaeinfo   Could you re-post the above question as a new case on the RealSense ROS GitHub by visiting the link below and clicking the green **New Issue** button please?  Thanks!

https://github.com/IntelRealSense/realsense-ros/issues

MartyG-RealSense, 2021-05-29:
Hi @rajeshmanpat   Do you have an update about this case that you can provide, please?

MartyG-RealSense, 2021-06-04:
Case closed due to no further comments received.

closed 2021-06-04 by MartyG-RealSense

DaphnaNanovel, 2024-08-25:
> Hi @rajeshmanpat It sounds as though the camera can be detected if you are able to start the stream in order for it to provide the message **No Frames Received**. Is this correct please?
> 
> Also, which stream gave this message - depth or color - and which program were you using? Was it the RealSense Viewer or the RealSense ROS wrapper?

Hi Marty,

I'm facing a similar issue with my D405 camera on Ubuntu (both x86 and arm). I've attached some screenshots that illustrate the problem I'm encountering.

Could you please provide some guidance on how to troubleshoot and resolve this issue? Any suggestions or potential solutions would be greatly appreciated.

Thanks,
Daphna
[image]
[image]
[image]

MartyG-RealSense, 2024-08-25:
Hi @DaphnaNanovel  Your D405 camera's serial number shows in the provided RealSense Viewer log as ffffffffffff.  This serial is incorrect and indicates that there is a problem.

A Left IC2 Config Error can mean that a cable inside the camera that joins two circuit boards together has a problem, such as the ends of the cable not being fully seated in the connecting ports on the boards.

In a paste case though it has also occurred when the USB cable had some damage caused by it being bent.

Are you using the official 1 meter USB cable that was supplied with the D405 camera, please?