realsenseai/realsense-ros #1813: Plans to migrate "/enable" service to ROS2?

anaelle-sw, 2021-04-15:
Hello,

In ROS1, we use the ["/enable" servic ](https://github.com/IntelRealSense/realsense-ros/blob/4424c59914ecfd0db39f28267ead96a4a5d3ccbe/realsense2_camera/src/realsense_node_factory.cpp#L264) in order to enable or disable a camera at runtime. The service internally calls the function `BaseRealSenseNode::toggleSensors` [here](https://github.com/IntelRealSense/realsense-ros/blob/4424c59914ecfd0db39f28267ead96a4a5d3ccbe/realsense2_camera/src/base_realsense_node.cpp#L177).

We would also be interested in using this service in ROS2, but it is supported yet.
I am aware of parameters like `enable_pointcloud` or `enable_depth`, etc... The problem with these is that they can only be set at launch time, but not at runtime. 

So I am wondering: Is there any plan to migrate the "/enable" service to ROS2? Is there any alternative today to enable/disable cameras at runtime without having to kill/re-launch the associated nodes? The function `toggleSensors` doesn't exist on the ROS2 branches of this repo, is there a specific reason for it? 

Thanks for helping!

doronhi, 2021-04-18:
There are plans to allow enabling/disabling specific sensors, pointcloud and other post-processing blocks in runtime.
The work is temporarily on hold but hopefully will be carried on in the near future.

RealSenseSupport, 2021-05-12:
@anaelle-sw The latest release https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.6 implemented this feature. Please have a try. Thanks!

RealSenseSupport, 2021-05-18:
@anaelle-sw Did you get chance to try the latest release https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.6 ? Any update from your side? Thanks!

anaelle-sw, 2021-05-18:
Hello, sorry for late reply. Yes, the release is working fine for us, thanks. I am closing this

closed 2021-05-18 by anaelle-sw