realsenseai/realsense-ros #1814: ROS2 Dashing debian package missing

joelbudu, 2021-04-19:
It seems the ROS2 Dashing debian package is missing (as seen in [https://index.ros.org/r/realsense2_camera/#dashing](https://index.ros.org/r/realsense2_camera/#dashing). The installation instructions on the ros2 branch however stipulates that this is availabale [https://github.com/IntelRealSense/realsense-ros/tree/ros2#installation-instructions](https://github.com/IntelRealSense/realsense-ros/tree/ros2#installation-instructions).

Is it possible for the dashing debian package to be released as well?

MartyG-RealSense, 2021-04-19:
Hi @joelbudu  The links are apparently broken at the time of writing this.

Would it be possible for you to install ROS2 Dashing with Debian packages using the link below?

https://docs.ros.org/en/dashing/Installation/Ubuntu-Install-Debians.html

A similar Debian package page for ROS2 Foxy is here:

https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html

joelbudu, 2021-04-19:
I'm actually referring to this command:
```sudo apt-get install ros-$ROS_DISTRO-realsense2-camera```

if $ROS_DISTRO=dashing you get this:
```E: Unable to locate package ros-dashing-realsense2-camera```

MartyG-RealSense, 2021-04-19:
I was referring to how the installation links for ROS2 in the instructions section are currently broken.  So it could be related to your problem with the **sudo apt-get** command.

[image]

joelbudu, 2021-04-19:
Can you please check with the availability of this debian package?
ros-dashing-realsense2-camera
cc: @doronhi

MartyG-RealSense, 2021-04-19:
@joelbudu   Yes, @doronhi the RealSense ROS wrapper developer is the appropriate person to refer the question to.

jkeshav-bvignesh, 2021-04-20:
I get the same error for the foxy debian package as well.

I use the official ros:foxy docker image. When I run `sudo apt-get install ros-$ROS_DISTRO-realsense2-camera`, I also get the `E: Unable to locate package ros-foxy-realsense2-camera` error.

doronhi, 2021-04-20:
I apologize.
The debian packages for all versions, besides Melodic perhaps, are missing.
I hope to get them all fixed for the next release of both ROS1 and ROS2, in 1-2 weeks.

MartyG-RealSense, 2021-05-04:
Hi @doronhi   Do you have an update that you can provide about the missing Debian packages please?  Thanks!

MartyG-RealSense, 2021-05-15:
Case closed due to no further comments received.

closed 2021-05-15 by MartyG-RealSense

SaltyHash, 2021-06-06:
Hey there @doronhi,

Looks like this is still an issue, and applies to ros-eloquent-realsense2-camera as well.  Any updates?

(My situation specifically:  NVIDIA Jetson Nano with Ubuntu 18.04 and ROS2 Eloquent installed via apt.  ros-*-realsense2-camera packages are nowhere to be found.)

doronhi, 2021-06-07:
As ROS2-Eloquent is EOL, no more sync events are being done by the ROS maintainers for Eloquent and it is no longer possible to release new debian packages for it. As far as I can tell your options are either to build realsense2_camera from source - Eloquent is still supported, or to move to either Dashing or Foxy (Or, pretty soon, to Galactic).

WonsokYoo, 2021-07-28:
@doronhi  Hi, I'm using foxy on 20.04 but when I type "sudo apt-get install ros-foxy-realsense2-camera", "E: Unable to locate package ros-foxy-realsense2-camera" are returned. Any ideas?

colinator, 2021-09-20:
This issue still exists, for both dashing and foxy. Shall we assume that intel is no longer supporting these? The pr says intel will continue to support at least some realsense cameras, and the software with it, but obviously these packages have been broken for a long time, so maybe they've abandoned them...