realsenseai/realsense-ros #1815: L515 Jetson nano Bond broken

jacklu333333, 2021-04-19:
I'm using jetson nano with realsense L515 through docker. The image version I use is ros:melodic.
I also experiment with ros:noetic and ros:kinetic. All the result shows that it was started successfully and then shut down.
I use the default launch file 

`roslanch realsense2_camera rs_camera.launch initial_reset:=true`

The output is as follow:

```
root@JetsonNano:/opt/ros/melodic/share/realsense2_camera/launch# roslaunch realsense2_camera rs_camera.launch initial_reset:=true

... logging to /root/.ros/log/b8968e60-a11f-11eb-a361-42776fa411cd/roslaunch-JetsonNano-1999.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://JetsonNano:36089/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
[... 94 lines cut ...]
 19/04 14:59:24,259 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1618844364.322354189]: insert Depth to L500 Depth Sensor
[ INFO] [1618844364.322640133]: insert Color to RGB Camera
 19/04 14:59:24,794 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 19/04 14:59:25,044 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 19/04 14:59:25,283 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 19/04 14:59:25,428 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 19/04 14:59:25,749 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/04 14:59:25,799 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 19/04 14:59:25,851 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1618844365.866237718]: SELECTED BASE:Depth, 0
 19/04 14:59:25,902 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[camera/realsense2_camera-3] process has finished cleanly
log file: /root/.ros/log/b8968e60-a11f-11eb-a361-42776fa411cd/camera-realsense2_camera-3*.log
[ INFO] [1618844365.954726584]: RealSense Node Is Up!
[ WARN] [1618844365.994102719]: 
[ WARN] [1618844365.996584291]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 19/04 14:59:26,804 WARNING [546442297728] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

The output of the log file is as follow:
```
[ INFO] [1618844334.872696953]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
[ INFO] [1618844365.574920894]: Bond broken, exiting
```

RealSenseSupport, 2021-05-03:
Hi @jacklu333333 

Are you able to connect the L515 to a system outside of docker container and outside of ROS environment?  Are you able to see and use the camera under RealSense Viewer?  I want to rule out other issues that might cause bond to be broken like any HW or other related issues with the configuration/OS/containers that you're using.

jacklu333333, 2021-05-04:
I manage to solve this issue by reinstall the whole system and set up the docker container from scratch. But I could identify exactly what is the reason for the error. 
Thank you for your help.

closed 2021-05-04 by jacklu333333

tmd9936, 2021-05-10:
The same error is happening to me, so
Can you tell me the command used to run the container?

jacklu333333, 2021-05-10:
@tmd9936 
Hi,
This is the cmd I use
`docker run --name myros --gpus all --net=host --env="DISPLAY" --volume="$HOME/.Xauthority:/root/.Xauthority:rw" --privileged -v /dev:/dev -v /tmp/.X11-unix:/tmp/.X11-unix -it ros:melodic`

In my case, I run docker in the same way both times, it might not be the reason for this error.