realsenseai/realsense-ros #1817: L515 Intrinsic Parameters

MBaranPeker, 2021-04-21:
hi, I am trying to get camera intrinsic parameters by subscribing camera_info topic. But I am facing with a problem, whenever I change resolution in the rs_camera.xml file the parameters doesn't change. It only shows intrinsic parameters for 1280*720 resolution. When ros_wrapper runs I can see it runs with new resolution but Intrinsic parameters are always the same.
I will be so happy if anybody can help me on this. Thanks

RealSenseSupport, 2021-05-07:
Hi @MBaranPeker 

Does this only happen in ROS?  You're looking for RGB intrinsic I'm assuming.  The L515 camera supports 3 RGB resolutions 1920x1080, 1280x720, 960x540.  Are you trying some other resolution?
Running rs-enumerate-devices -c will provide you with intrinsic information.

doronhi, 2021-05-09:
Which ROS distribution do you use please?
When I try I see different values in the *camera_info* topic when I run with different resolutions.
Are you sure you manage to change the resolution? Take a look at [librealsense #8971](https://github.com/IntelRealSense/librealsense/issues/8971)

MBaranPeker, 2021-05-13:
@doronhi thank you for your answer. Melodic. Actually I have followed librealsense before cahnging the resolution did not work anyway i handeled that in another way

closed 2021-05-26 by RealSenseSupport