realsenseai/realsense-ros #1818: How to get ride of this warning messenger-libusb.cpp, ds5-options.cpp and ds5-options.cpp:86

ranjitkathiriya, 2021-04-21:
Hey there,

I have seen this warning when I run a real sense node. Is there any way to remove these warnings? I have checked in Rviz. It is working perfectly, but these warnings often come when I am testing nodes. Is it possible to remove it if there is an easy fix then only? Otherwise, I am okay with these warnings. 

```
21/04 12:10:55,729 WARNING [140462651553536] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 12:10:55,779 WARNING [140462651553536] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 12:10:55,830 WARNING [140462651553536] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 12:10:55,880 WARNING [140462651553536] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 12:10:56,014 WARNING [140462651553536] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 12:10:56,065 WARNING [140462651553536] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 12:10:56,116 WARNING [140462651553536] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 12:10:56,622 ERROR [140461939357440] (ds5-options.cpp:86) Asic Temperature value is not valid!
 21/04 12:10:57,623 ERROR [140461939357440] (ds5-options.cpp:86) Asic Temperature value is not valid!
```

MartyG-RealSense, 2021-04-21:
Hi @ranjitkathiriya  If the **control_transfer returned** warnings only occur at launch and the launch is completed successfully, or appear one or two times a minute after launch, then they can be safely ignored.

If they are being generated continuously then it can indicate that there is a serious communication problem with the device.

They are not a message that you can remove, though **Doronhi** the RealSense ROS developer has spoken in the link below about the possibility of removing them in future.

https://github.com/IntelRealSense/realsense-ros/issues/1162#issuecomment-624129592

ranjitkathiriya, 2021-04-21:
> appear one or two times a minute after launch, then they can be safely ignored.
Yes, I see them in a minute or two, SO I will ignore it.

ranjitkathiriya, 2021-04-21:
> If they are being generated continuously then it can indicate that there is a serious communication problem with the device.

When I run the node at that time it is generated and then after 30 sec or 1 or 2 min in between they are coming, but I can tell that my project is working fine. Do you think that it may be a communication problem? If yes, How can I solve it any steps about that for checking?

MartyG-RealSense, 2021-04-21:
If you are receiving the messages at a rate of only one or two per minute and your project is working fine then the messages can be safely ignored.

ranjitkathiriya, 2021-04-21:
Yes! It is not generating often; I am relaxed now. It is generating every 1-2 minutes. I am closing this case.

Thanks! for helping me out...

closed 2021-04-21 by ranjitkathiriya