realsenseai/realsense-ros #1820: Intel Realsense SDK 'librealsense' vs librealsense2

williamx97, 2021-04-22:
Hi there,

I was following method 2 here: https://github.com/IntelRealSense/realsense-ros on a linux machine and need to build everything from source. 

Should I install the Realsense SDK 2.0 ```librealsense``` in ```~/ ``` or ```~/catkin_ws/src``` for Method 2 Step 1? 

If I install ```librealsense``` to ```~/ ``` should I also install ```librealsense2``` into ```~/catkin_ws/src``` when installing ```realsense-ros``` into  ```~/catkin_ws/src``` ?

Or should I only install ```realsense-ros``` into ```~/catkin_ws/src``` ? Won't that cause a dependencies issue as it depends on ```librealsense2```?

What's the difference between ```librealsense``` and ```librealsense2``` ?

MartyG-RealSense, 2021-04-22:
Hi @williamx97 'librealsense2', also known as RealSense SDK 2.0, is the current RealSense SDK, whilst 'librealsense' refers to the original librealsense SDK that is no longer supported and is now referred to as Legacy Librealsense.  So you should use the librealsense2 SDK with the RealSense ROS wrapper if you are using one of these camera models: SR300, 400 Series, L515 or T265.  

Legacy Librealsense does not support modern RealSense models and instead supports the F200, R200, SR300 and ZR300 models.

You should not install librealsense2 into the **~/catkin_ws/src/** folder.  This folder is for the RealSense ROS wrapper.

If you plan to also build librealsense2 from source code on Linux then the instructions for doing so are in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

You can alternatively build librealsense2 from Debian packages if you wish.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

The librealsense2 SDK should be built first and then the RealSense ROS wrapper after that.

williamx97, 2021-04-22:
Thanks for the response Marty. Does it matter if I download the tar file for the SDK from here: https://github.com/IntelRealSense/librealsense/releases/tag/v2.43.0 rather than from here: https://github.com/IntelRealSense/librealsense2-release/releases/tag/release%2Fkinetic%2Flibrealsense2%2F2.43.0-2 ? Or are both essentially the same?

williamx97, 2021-04-22:
and I'm assuming I'll need the dependencies listed here: http://wiki.ros.org/realsense2_camera on the right under 'dependencies' installed in ```~/catkin_ws/src ``` except for ```librealsense2``` ?

MartyG-RealSense, 2021-04-22:
I am not familiar with **release/kinetic/librealsense2/2.43.0-2**.  But in general, if you are going to use a tar.gz file then I recommend always obtaining it from the librealsense SDK releases page - https://github.com/IntelRealSense/librealsense/releases/

If you are installing librealsense and the ROS wrapper fresh then you can do so for the newer **2.44.0** librealsense version, which should be matched with the new ROS wrapper version **2.2.24** that was released today.

**librealsense SDK 2.44.0**
https://github.com/IntelRealSense/librealsense/releases/tag/v2.44.0

**ROS wrapper 2.2.24**
https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.24

In regard to dependencies, you should ensure that the ROS package **ddynamic_reconfigure** is installed.   The Method 2 instructions for building the RealSense ROS wrapper from source include information about how to do this.

williamx97, 2021-04-22:
Cheers Marty :) Thanks for the quick responses, closing the question

closed 2021-04-22 by williamx97