realsenseai/realsense-ros #1821: Set auto exposure upper limit

yorumun, 2021-04-23:
Is it possible to limit the RGB exposure when Auto Exposure is enabled?
I want to reduce blur when the subject is moving at night.

Thanks.

MartyG-RealSense, 2021-04-23:
Hi @yorumun  If you are using ROS, you could try using a **Region of Interest (ROI)** for the RGB auto-exposure.  The documentation link below states that this can be done with **/camera/rgb_camera/auto_exposure_roi/**

https://dev.intelrealsense.com/docs/ros-wrapper#section-d-400-stereo-depth-cameras

The link below provides an example of setting the RGB ROI in ROS with a rosparam in the launch file.

https://github.com/IntelRealSense/realsense-ros/issues/900#issuecomment-547261460

You should also be able to set the RGB ROI dynamically with rqt_reconfigure during runtime if you need different ROI settings for day and night.

yorumun, 2021-04-23:
Hi @MartyG-RealSense  Thanks for the reply.
Sorry for the lack of explanation.

I am using ROS and I am using dynamic_reconfigure to set the ROI.
However, I am having trouble with blurring.
So I would like to set the upper limit of the exposure value.
Is there any good way to do this?

MartyG-RealSense, 2021-04-23:
I understood your meaning and suggested an RGB ROI as a means of controlling the pixel intensity whilst auto-exposure is enabled.

In regard to reducing blurring, could you tell me which camera model you are using please?

yorumun, 2021-04-23:
I understand. Thank you very much.

I am using a D455.

MartyG-RealSense, 2021-04-23:
You could try setting **Auto-Exposure Priority** to false. When auto-exposure is enabled and auto-exposure priority is disabled, a constant FPS should be enforced instead of the FPS being able to vary depending upon epxosure.

https://github.com/IntelRealSense/realsense-ros/issues/1349#issuecomment-681695259

yorumun, 2021-04-26:
I tried that setting, but it caused blurring.
I set image size is 1280 x 720 and FPS is 30.

Can I set maximum exposure time value?
Dose "RS2_OPTION_AUTO_EXPOSURE_LIMIT" option disable in D455?

MartyG-RealSense, 2021-04-26:
The documentation for **RS2_OPTION_AUTO_EXPOSURE_LIMIT** states that its default value is '0', which means that the full exposure range is used by default.  

https://intelrealsense.github.io/librealsense/doxygen/rs__option_8h.html#a8b9c011f705cfab20c7eaaa7a26040e2ac11001971364b90b542941e51ebb911a

yorumun, 2021-04-27:
I understand.
I want to change the value of the maximum exposure time.

So I would like to know if it is possible to change this value in realsense-ros.

If not, I would like to know if I can change "RS2_OPTION_AUTO_EXPOSURE_LIMIT" directly.
When I tried to change the value with the following code, `RuntimeError: object doesn't support option #85` occurred.

```
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

profile = pipeline.start(config)

sensor = profile.get_device().first_color_sensor()

sensor.set_option(rs.option.enable_auto_exposure, True)
sensor.set_option(rs.option.auto_exposure_limit, 100)
```

MartyG-RealSense, 2021-04-27:
It looks as though **RS2_OPTION_AUTO_EXPOSURE_LIMIT** is a depth sensor option rather than a color sensor one.

https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-device.cpp#L982

I cannot think of a way to enforce limits on **color** exposure whilst auto-exposure is enabled.

MartyG-RealSense, 2021-05-09:
Case closed due to no further comments received.

closed 2021-05-09 by MartyG-RealSense

patrickwasp, 2022-07-01:
@yorumun did you ever find a work around for this?