realsenseai/realsense-ros #1822: Can't change FPS for d455

JeremieBourque1, 2021-04-23:
Hi,

I am trying to lower the fps for the depth and color modules of my d455 but the parameters don't have any effect.

### Launch file
```xml
<?xml version='1.0'?>
<launch>
  <include file="$(find realsense2_camera)/launch/rs_camera.launch" >
    <arg name="camera" value="d455"/>
    <arg name="depth_fps" value="5"/>
    <arg name="color_fps" value="5"/>
    <arg name="enable_depth" value="true"/>
    <arg name="enable_pointcloud" value="true"/>
    <arg name="enable_confidence" value="false"/>
    <arg name="align_depth" value="false"/>
  </include>
</launch>
```

### Output
```
[ INFO] [1619190346.698119323]: Initializing nodelet with 12 worker threads.
[ INFO] [1619190346.927939960]: RealSense ROS v2.2.23
[ INFO] [1619190346.928037427]: Built with LibRealSense v2.42.0
[ INFO] [1619190346.928092797]: Running with LibRealSense v2.43.0
[ WARN] [1619190346.928147581]: ***************************************************
[ WARN] [1619190346.928186786]: ** running with a different librealsense version **
[ WARN] [1619190346.928223282]: ** than the one the wrapper was compiled with!   **
[ WARN] [1619190346.928259156]: ***************************************************
[ INFO] [1619190346.962458112]:  
[ INFO] [1619190347.047070159]: Device with serial number 035322250587 was found.

[ INFO] [1619190347.047162515]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2 was found.
[ INFO] [1619190347.047196645]: Device with name Intel RealSense D455 was found.
[ INFO] [1619190347.047966904]: Device with port number 2-1 was found.
[ INFO] [1619190347.048036763]: Device USB type: 3.2
[ INFO] [1619190347.055575300]: getParameters...
[ INFO] [1619190347.107258253]: setupDevice...
[ INFO] [1619190347.107291094]: JSON file is not provided
[ INFO] [1619190347.107309458]: ROS Node Namespace: d455
[ INFO] [1619190347.107337338]: Device Name: Intel RealSense D455
[ INFO] [1619190347.107347794]: Device Serial No: 035322250587
[ INFO] [1619190347.107361404]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video2
[ INFO] [1619190347.107374978]: Device FW version: 05.12.12.100
[ INFO] [1619190347.107387866]: Device Product ID: 0x0B5C
[ INFO] [1619190347.107399689]: Enable PointCloud: On
[ INFO] [1619190347.107411248]: Align Depth: Off
[ INFO] [1619190347.107420619]: Sync Mode: On
[ INFO] [1619190347.107523129]: Device Sensors: 
[ INFO] [1619190347.143571868]: Stereo Module was found.
 23/04 11:05:47,143 WARNING [139997008107264] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1619190347.171083700]: RGB Camera was found.
[ INFO] [1619190347.172128038]: Motion Module was found.
[ INFO] [1619190347.172161383]: Add Filter: pointcloud
[ INFO] [1619190347.175112459]: num_filters: 1
[ INFO] [1619190347.175141238]: Setting Dynamic reconfig parameters.
[ WARN] [1619190347.274872092]: Param '/d455/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1619190347.294318533]: Done Setting Dynamic reconfig parameters.
[ INFO] [1619190347.295741324]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1619190347.298371252]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1619190347.298451767]: setupPublishers...
[ INFO] [1619190347.303098971]: Expected frequency for depth = 30.00000
[ INFO] [1619190347.354891687]: Expected frequency for color = 30.00000
[ INFO] [1619190347.390704475]: setupStreams...
[ INFO] [1619190347.392636866]: insert Depth to Stereo Module
[ INFO] [1619190347.392715369]: insert Color to RGB Camera
[ INFO] [1619190347.454037368]: SELECTED BASE:Depth, 0
[ INFO] [1619190347.476506171]: RealSense Node Is Up!
[ WARN] [1619190347.737503832]: 

```

The log says that both depth and color are expected to run at 30 fps and if I check the publishing frequency with rostopic hz I get about 15 Hz.

### My setup
OS: Ubuntu 18.04
SDK version: 2.43.0
ROS wrapper version: 2.2.23
d455 firmware version: 05.12.12.100

Any ideas how I can change the FPS?

Thanks

MartyG-RealSense, 2021-04-23:
Hi @JeremieBourque1  Since ROS wrapper version **2.2.22**, a new profile system was introduced where the default configuration of the camera is selected in the rs_camera.launch file if the launch profile is invalid.

When setting a custom configuration, you should now provide three factors for each stream type that you are customizing - its **height, width and FPS**.

Doronhi the RealSense ROS wrapper developer provides further information about this in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1751#issuecomment-798834284

In the log there is another problem with your ROs wrapper setup.

[image]

The above information indicates that you bult ROS wrapper 2.2.23 with librealsense SDK version 2.42.0 and then updated librealsense to version 2.43.0.  If you update the librealsense version then you should build the ROS wrapper again afterwards.

JeremieBourque1, 2021-04-23:
That fixed the problem, thank you!

closed 2021-04-23 by JeremieBourque1