realsenseai/realsense-ros #1823: Dashing StaticTransformBroadcaster late subscription

daziFR, 2021-04-26:
Hi,

I'm testing of 3.1.5 single wrapper supporting Dashing, Eloquent and Foxy. 
There is an issue with late subscriber (rviz), with warning for fixed frame: "No tf data. Actual error: Frame [map] does not exist". Fixed frame cannot be changed.

More information is already given in the: https://github.com/IntelRealSense/librealsense/issues/8863

There was a solution for this problem for old, obsolete ROS2 wrapper supporting Dashing.

Can you comment on this? I've tried to integrate this old fix and it does work, but maybe there is another approach.

Thank You.

MartyG-RealSense, 2021-04-26:
Thanks very much @daziFR for your report.

@doronhi   Could you comment on this case please?  Thanks!

doronhi, 2021-05-02:
Hi @daziFR , thanks for the elaborate description.
I think the [solution for this problem for old, obsolete ROS2 wrapper](https://github.com/intel/ros2_intel_realsense/pull/87) is already incorporated into this wrapper using the **tf_publish_rate** option. Try adding **tf_publish_rate:=1.0**

MartyG-RealSense, 2021-05-08:
Hi @daziFR  Do you require further assistance with this case, please?  Thanks!

daziFR, 2021-05-10:
Hi @MartyG-RealSense and @doronhi . I've tried setting the **tf_publish_rate:=1.0** and it works with late subscription. Topic can be closed. Thanks.

closed 2021-05-10 by daziFR

MartyG-RealSense, 2021-05-10:
Great to hear that you were successful @daziFR - thanks very much for the update!