realsenseai/realsense-ros #1824: Resource not found: rgbd_launch

GRF-Sunomikp31, 2021-04-26:
when I use "roslaunch realsense2_camera rs_rgbd.launch 
"  ;then:Resource not found: rgbd_launch
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/gipsy/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

My Ubuntu18.04 ROS melodic

GRF-Sunomikp31, 2021-04-26:
when I use :sudo apt-get install ros-melodic-rgbd-launch
error back:E: Unable to locate package ros-melodic-rgbd-launch

MartyG-RealSense, 2021-04-26:
Hi @GRF-Sunomikp31   Does it make a difference if you refresh the local packages cache first with the command below?

**sudo apt-get update**

GRF-Sunomikp31, 2021-04-27:
it doesn't work,but when i use sudo apt-get update,back like this:W: Target Packages (main/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target Packages (main/binary-i386/Packages) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target Translations (main/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target DEP-11 (main/dep11/Components-amd64.yml) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target DEP-11 (main/dep11/Components-all.yml) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target DEP-11-icons-small (main/dep11/icons-48x48.tar) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target DEP-11-icons (main/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target CNF (main/cnf/Commands-amd64) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target CNF (main/cnf/Commands-all) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target Packages (main/binary-amd64/Packages) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target Packages (main/binary-i386/Packages) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target Translations (main/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target DEP-11 (main/dep11/Components-amd64.yml) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target DEP-11 (main/dep11/Components-all.yml) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target DEP-11-icons-small (main/dep11/icons-48x48.tar) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target DEP-11-icons (main/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target CNF (main/cnf/Commands-amd64) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1
W: Target CNF (main/cnf/Commands-all) is configured multiple times in /etc/apt/sources.list:75 and /etc/apt/sources.list.d/realsense-public.list:1

GRF-Sunomikp31, 2021-04-27:
I just solved the problem ,but it still happens above.

MartyG-RealSense, 2021-04-27:
Another RealSense user experienced these 'configured multiple times' errors.  Their solution was to input the command **sudo apt-get upgrade** after entering the update command.

**sudo apt-get update
sudo apt-get upgrade**

https://github.com/IntelRealSense/librealsense/issues/4902#issuecomment-533834500

GRF-Sunomikp31, 2021-04-27:
> Another RealSense user experienced these 'configured multiple times' errors. Their solution was to input the command **sudo apt-get upgrade** after entering the update command.
> 
> **sudo apt-get update sudo apt-get upgrade**
> 
> [IntelRealSense/librealsense#4902 (comment)](https://github.com/IntelRealSense/librealsense/issues/4902#issuecomment-533834500)

I solved this problem, but now I have new problems.  I use **roslaunch realsense2_camera rs_camera.launch ** and **roslaunch realsense2_camera rs_rgbd.launch  **  have some error：

[camera/realsense2_camera_manager-2] process has died [pid 4260, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/gipsy/.ros/log/07294b18-a732-11eb-aa94-505bc2cc4003/camera-realsense2_camera_manager-2.log].
log file: /home/gipsy/.ros/log/07294b18-a732-11eb-aa94-505bc2cc4003/camera-realsense2_camera_manager-2*.log

[camera/realsense2_camera_manager-2] process has died [pid 3338, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/gipsy/.ros/log/2d463264-a72f-11eb-aa94-505bc2cc4003/camera-realsense2_camera_manager-2.log].
log file: /home/gipsy/.ros/log/2d463264-a72f-11eb-aa94-505bc2cc4003/camera-realsense2_camera_manager-2*.log

MartyG-RealSense, 2021-04-27:
Can you provide your **roslaunch** logs here please?  Thanks very much.

GRF-Sunomikp31, 2021-04-27:
> Can you provide your **roslaunch** logs here please? Thanks very much.

like this？

[0m[ INFO] [1619514096.873941195]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:[0m
[0m[ INFO] [1619514096.876340615]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] could not connect to host [gipsy-INVALID:48469], waiting...[0m
[0m[ INFO] [1619514096.896787926]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.[0m
[0m[ INFO] [1619514107.248205761]: Bond broken, exiting[0m

MartyG-RealSense, 2021-04-27:
By roslaunch logs, I mean the information that is listed after the roslaunch instruction is entered.  Like this:

[image]

GRF-Sunomikp31, 2021-04-27:
this？

... logging to /home/gipsy/.ros/log/32ff08fa-a73b-11eb-812b-505bc2cc4003/roslaunch-gipsy-INVALID-2975.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gipsy-INVALID:45215/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 53 lines cut ...]
[ INFO] [1619515838.477686988]:  
[ INFO] [1619515838.541025340]: Device with serial number 040322070435 was found.

[ INFO] [1619515838.541113488]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0 was found.
[ INFO] [1619515838.541155604]: Device with name Intel RealSense D435I was found.
[ INFO] [1619515838.542340112]: Device with port number 2-3 was found.
[ INFO] [1619515838.542418146]: Device USB type: 3.2
[ INFO] [1619515838.579446888]: getParameters...
[ INFO] [1619515838.617097204]: setupDevice...
[ INFO] [1619515838.617136850]: JSON file is not provided
[ INFO] [1619515838.617146342]: ROS Node Namespace: camera
[ INFO] [1619515838.617157566]: Device Name: Intel RealSense D435I
[ INFO] [1619515838.617165676]: Device Serial No: 040322070435
[ INFO] [1619515838.617175891]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/video4linux/video0
[ INFO] [1619515838.617184562]: Device FW version: 05.11.01.100
[ INFO] [1619515838.617193290]: Device Product ID: 0x0B3A
[ INFO] [1619515838.617202621]: Enable PointCloud: Off
[ INFO] [1619515838.617215798]: Align Depth: Off
[ INFO] [1619515838.617229821]: Sync Mode: Off
[ INFO] [1619515838.617272300]: Device Sensors: 
[ INFO] [1619515838.619156182]: Stereo Module was found.
[camera/realsense2_camera_manager-2] process has died [pid 3024, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/gipsy/.ros/log/32ff08fa-a73b-11eb-812b-505bc2cc4003/camera-realsense2_camera_manager-2.log].
log file: /home/gipsy/.ros/log/32ff08fa-a73b-11eb-812b-505bc2cc4003/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/gipsy/.ros/log/32ff08fa-a73b-11eb-812b-505bc2cc4003/camera-realsense2_camera-3*.log

MartyG-RealSense, 2021-04-27:
Yes, that is correct.  Thank you very much.

I note from the log that there is a mismatch in your librealsense SDK version and ROS wrapper version.  If you are using ROS wrapper version **2.2.24** then this should be matched with librealsense SDK version **2.44.0** and not 2.42.0.  

This is because a particular ROS wrapper version should be matched as closely as possible with the librealsense version listed in that particular ROS wrapper version's release notes.

[image]

If you want to avoid building librealsense again, then the earlier ROS wrapper version **2.2.22** should be built.  This is the wrapper recommended for SDK 2.42.0.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.22

GRF-Sunomikp31, 2021-04-27:
Okay, I will try this and give feedback again, thank you very much!

GRF-Sunomikp31, 2021-04-27:
I use the  **realsense-ros-2.2.22**，but still error；

[ INFO] [1619518409.038548668]: Initializing nodelet with 8 worker threads.
[ INFO] [1619518409.323450400]: RealSense ROS v2.2.22
[ INFO] [1619518409.323502476]: Built with LibRealSense v2.42.0
[ INFO] [1619518409.323529925]: Running with LibRealSense v2.42.0
[ INFO] [1619518409.349885385]:  
[ INFO] [1619518409.563244931]: Device with serial number 040322070435 was found.

[ INFO] [1619518409.563317299]: Device with physical ID 2-3-2 was found.
[ INFO] [1619518409.563340469]: Device with name Intel RealSense D435I was found.
[ INFO] [1619518409.578035625]: Device with port number 2-3 was found.
[ INFO] [1619518409.578166814]: Device USB type: 3.2
[ INFO] [1619518409.582684167]: getParameters...
[ INFO] [1619518409.657259787]: setupDevice...
[ INFO] [1619518409.657280687]: JSON file is not provided
[ INFO] [1619518409.657293449]: ROS Node Namespace: camera
[ INFO] [1619518409.657303774]: Device Name: Intel RealSense D435I
[ INFO] [1619518409.657311850]: Device Serial No: 040322070435
[ INFO] [1619518409.657320677]: Device physical port: 2-3-2
[ INFO] [1619518409.657328981]: Device FW version: 05.11.01.100
[ INFO] [1619518409.657337116]: Device Product ID: 0x0B3A
[ INFO] [1619518409.657345028]: Enable PointCloud: Off
[ INFO] [1619518409.657352787]: Align Depth: On
[ INFO] [1619518409.657362033]: Sync Mode: On
[ INFO] [1619518409.657417964]: Device Sensors: 
[ INFO] [1619518409.671511360]: Stereo Module was found.
[ INFO] [1619518409.678439632]: RGB Camera was found.
[ INFO] [1619518409.678606128]: Motion Module was found.
[ INFO] [1619518409.678669051]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1619518409.678694564]: num_filters: 0
[ INFO] [1619518409.678721600]: Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/gipsy/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleE
[camera/realsense2_camera_manager-1] process has died [pid 11639, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/gipsy/.ros/log/3c258666-a740-11eb-a795-505bc2cc4003/camera-realsense2_camera_manager-1.log].
log file: /home/gipsy/.ros/log/3c258666-a740-11eb-a795-505bc2cc4003/camera-realsense2_camera_manager-1*.log

MartyG-RealSense, 2021-04-27:
If you are building the ROS wrapper from source code using Method 2 then the installation instructions refer to making sure that you have installed the dependency **ddynamic_reconfigure**

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

GRF-Sunomikp31, 2021-04-27:
I  used Method 2，but I don't know how to install this dependency, just follow this method two to install it again? I just tried it and it didn't work

MartyG-RealSense, 2021-04-27:
Could you try using the command below please to install ddynamic_reconfigure for Melodic:

**apt-get update apt-get install ros-melodic-ddynamic-reconfigure**

GRF-Sunomikp31, 2021-04-27:
now have new error:

... logging to /home/gipsy/.ros/log/a6c7b7b6-a75c-11eb-9340-505bc2cc4003/roslaunch-gipsy-INVALID-3894.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gipsy-INVALID:40595/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 84 lines cut ...]
[ INFO] [1619530196.331245341]: insert Color to RGB Camera
 27/04 09:29:56,383 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1619530196.406665540]: SELECTED BASE:Depth, 0
[ INFO] [1619530196.417727303]: RealSense Node Is Up!
 27/04 09:29:56,433 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[ WARN] [1619530196.570821098]: 
 27/04 09:29:56,572 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:56,622 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:56,673 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:56,723 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:56,775 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:56,826 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:56,926 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:57,078 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 27/04 09:29:57,885 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:57,936 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:58,087 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 09:29:58,138 WARNING [139988041905920] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-04-27:
If **control_transfer returned** warnings only occur during launch and then stop, or only appear once or twice a minute after launch, then they can be safely ignored.

If the warnings continuously generate after launch then this can indicate that there is a serious communication problem wth the device.

GRF-Sunomikp31, 2021-04-27:
I think it‘s the latter，but my rviz doesn't have any image.

MartyG-RealSense, 2021-04-27:
Does performance improve if you add an **initial_reset:=true** instruction onto the end of the roslaunch instruction to reset the camera at launch?

**roslaunch realsense2_camera rs_rgbd.launch initial_reset:=true**

GRF-Sunomikp31, 2021-04-27:
it doesn't work;

roslaunch realsense2_camera rs_rgbd.launch initial_reset:=true
... logging to /home/gipsy/.ros/log/95ce9abc-a773-11eb-942e-505bc2cc4003/roslaunch-gipsy-INVALID-3598.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://gipsy-INVALID:43459/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 77 lines cut ...]
[ INFO] [1619540045.742627385]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1619540045.745930023]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1619540045.749547114]: setupPublishers...
[ INFO] [1619540045.755105404]: Expected frequency for depth = 30.00000
[ INFO] [1619540045.795049876]: Expected frequency for color = 30.00000
[ INFO] [1619540045.811267397]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1619540045.823560523]: setupStreams...
[ INFO] [1619540045.834380277]: insert Depth to Stereo Module
[ INFO] [1619540045.834418434]: insert Color to RGB Camera
 27/04 12:14:05,847 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 12:14:05,897 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1619540046.000124482]: SELECTED BASE:Depth, 0
[ INFO] [1619540046.015345847]: RealSense Node Is Up!
[ WARN] [1619540046.059928194]: 
 27/04 12:14:06,060 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 12:14:06,111 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 12:14:06,162 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 12:14:06,290 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 27/04 12:14:06,797 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 12:14:06,847 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 12:14:06,999 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/04 12:14:07,050 WARNING [140349437994752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-04-27:
Next, please try setting a custom configuration of a 640x480 depth stream with 30 FPS speed:

**roslaunch realsense2_camera rs_rgbd.launch depth_width:=640 depth_height:=480 depth_fps:=30.0**

GRF-Sunomikp31, 2021-04-27:
I forgot to set the coordinate system on RVIZ, and now it succeeds without errors! ! ! ! ! It should be: sudo apt-get update apt-get install ros-melodic-ddynamic-reconfigure
It succeeded 
thank you very much!!!!!!!!

MartyG-RealSense, 2021-04-27:
You're very welcome.  :)  That's great news - thank very much for the update!

MartyG-RealSense, 2021-05-03:
Case closed due to solution achieved and no further comments received.

closed 2021-05-03 by MartyG-RealSense

jedichen121, 2021-05-04:
I was going through this entire thread but it turns out the fix for me to solve `Resource not found: rgbd_launch` is just

`sudo apt install ros-melodic-rgbd-launch`

Still wondering why this is a separate package, not something installed with realsense-ros.

MartyG-RealSense, 2021-05-04:
Hi @jedichen121  RGBD_launch is a ROS component and not specific to the RealSense ROS wrapper.

http://wiki.ros.org/rgbd_launch