realsenseai/realsense-ros #1825: roslaunch realsense2_camera rs_rgbd.launch  have some error;

GRF-Sunomikp31, 2021-04-27:
I installed the environment according to the github;but when i use :roslaunch realsense2_camera rs_rgbd.launch
feedback:
Resource not found: rgbd_launch
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/gipsy/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

when i use :roslaunch realsense2_camera rs_camera.launch
have some error :
[camera/realsense2_camera_manager-2] process has died [pid 2475, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/gipsy/.ros/log/c15f9d66-a700-11eb-a69d-505bc2cc4003/camera-realsense2_camera_manager-2.log].
log file: /home/gipsy/.ros/log/c15f9d66-a700-11eb-a69d-505bc2cc4003/camera-realsense2_camera_manager-2*.log

GRF-Sunomikp31, 2021-04-27:
I just tried to download and compile this package again,but complilations is wrong:    I use :catkin_make

-- ~~  traversing 8 packages in topological order:
-- ~~  - realsense2_camera_msgs (plain cmake)
-- ~~  - realsense2_description (plain cmake)
-- ~~  - test_pkg
-- ~~  - ddynamic_reconfigure
-- ~~  - realsense2_camera (plain cmake)
-- ~~  - learning_service
-- ~~  - learning_topic
-- ~~  - usb_cam
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.

MartyG-RealSense, 2021-04-27:
Closed as duplicate of https://github.com/IntelRealSense/realsense-ros/issues/1824

closed 2021-04-27 by MartyG-RealSense