realsenseai/realsense-ros #1826: How to reproduce pointcloud from recording?

victorll998, 2021-04-27:
Hi everyone,

I have been testing with ```rs_from_file.launch```. Now I can replay depth and color stream from the bag files that I recorded either through ```rs_record``` or ```realsense viewer```. But when I tried to reproduce point cloud from the d455 bag file using ```rs_from_file.launch``` , I got nothing. To be more exact, I got a warning " No stream match for pointcloud chosen texture Process - Color "

```
SUMMARY
========

PARAMETERS
 * /d455_left/realsense2_camera/accel_fps: 0
 * /d455_left/realsense2_camera/accel_frame_id: d455_left_accel_f...
 * /d455_left/realsense2_camera/accel_optical_frame_id: d455_left_accel_o...
 * /d455_left/realsense2_camera/align_depth: True
 * /d455_left/realsense2_camera/aligned_depth_to_color_frame_id: d455_left_aligned...
 * /d455_left/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d455_left_aligned...
 * /d455_left/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d455_left_aligned...
 * /d455_left/realsense2_camera/aligned_depth_to_fisheye_frame_id: d455_left_aligned...
 * /d455_left/realsense2_camera/aligned_depth_to_infra1_frame_id: d455_left_aligned...
 * /d455_left/realsense2_camera/aligned_depth_to_infra2_frame_id: d455_left_aligned...
 * /d455_left/realsense2_camera/allow_no_texture_points: False
 * /d455_left/realsense2_camera/base_frame_id: d455_left_link
 * /d455_left/realsense2_camera/calib_odom_file: 
 * /d455_left/realsense2_camera/clip_distance: -1.0
 * /d455_left/realsense2_camera/color_fps: 30
 * /d455_left/realsense2_camera/color_frame_id: d455_left_color_f...
 * /d455_left/realsense2_camera/color_height: 480
 * /d455_left/realsense2_camera/color_optical_frame_id: d455_left_color_o...
 * /d455_left/realsense2_camera/color_width: 640
 * /d455_left/realsense2_camera/confidence_fps: 30
 * /d455_left/realsense2_camera/confidence_height: 480
 * /d455_left/realsense2_camera/confidence_width: 640
 * /d455_left/realsense2_camera/depth_fps: 30
 * /d455_left/realsense2_camera/depth_frame_id: d455_left_depth_f...
 * /d455_left/realsense2_camera/depth_height: 480
 * /d455_left/realsense2_camera/depth_optical_frame_id: d455_left_depth_o...
 * /d455_left/realsense2_camera/depth_width: 640
 * /d455_left/realsense2_camera/device_type: 
 * /d455_left/realsense2_camera/enable_accel: False
 * /d455_left/realsense2_camera/enable_color: True
 * /d455_left/realsense2_camera/enable_confidence: True
 * /d455_left/realsense2_camera/enable_depth: True
 * /d455_left/realsense2_camera/enable_fisheye1: False
 * /d455_left/realsense2_camera/enable_fisheye2: False
 * /d455_left/realsense2_camera/enable_fisheye: False
 * /d455_left/realsense2_camera/enable_gyro: False
 * /d455_left/realsense2_camera/enable_infra1: False
 * /d455_left/realsense2_camera/enable_infra2: False
 * /d455_left/realsense2_camera/enable_infra: False
 * /d455_left/realsense2_camera/enable_pointcloud: True
 * /d455_left/realsense2_camera/enable_pose: False
 * /d455_left/realsense2_camera/enable_sync: True
 * /d455_left/realsense2_camera/filters: 
 * /d455_left/realsense2_camera/fisheye1_frame_id: d455_left_fisheye...
 * /d455_left/realsense2_camera/fisheye1_optical_frame_id: d455_left_fisheye...
 * /d455_left/realsense2_camera/fisheye2_frame_id: d455_left_fisheye...
 * /d455_left/realsense2_camera/fisheye2_optical_frame_id: d455_left_fisheye...
 * /d455_left/realsense2_camera/fisheye_fps: 30
 * /d455_left/realsense2_camera/fisheye_frame_id: d455_left_fisheye...
 * /d455_left/realsense2_camera/fisheye_height: 0
 * /d455_left/realsense2_camera/fisheye_optical_frame_id: d455_left_fisheye...
 * /d455_left/realsense2_camera/fisheye_width: 0
 * /d455_left/realsense2_camera/gyro_fps: 0
 * /d455_left/realsense2_camera/gyro_frame_id: d455_left_gyro_frame
 * /d455_left/realsense2_camera/gyro_optical_frame_id: d455_left_gyro_op...
 * /d455_left/realsense2_camera/imu_optical_frame_id: d455_left_imu_opt...
 * /d455_left/realsense2_camera/infra1_frame_id: d455_left_infra1_...
 * /d455_left/realsense2_camera/infra1_optical_frame_id: d455_left_infra1_...
 * /d455_left/realsense2_camera/infra2_frame_id: d455_left_infra2_...
 * /d455_left/realsense2_camera/infra2_optical_frame_id: d455_left_infra2_...
 * /d455_left/realsense2_camera/infra_fps: 30
[... 65 lines cut ...]
[ INFO] [1619503189.051453191]: Setting Dynamic reconfig parameters.
[ INFO] [1619503189.059235488]: Done Setting Dynamic reconfig parameters.
[ WARN] [1619503189.059376911]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: Z16
[ WARN] [1619503189.059447383]: Using default profile instead.
[ INFO] [1619503189.059855952]: depth stream is enabled - width: 640, height: 480, fps: 5, Format: Z16
[ WARN] [1619503189.059890554]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: None
[ WARN] [1619503189.059912491]: Using default profile instead.
[ INFO] [1619503189.060321779]: color stream is enabled - width: 640, height: 480, fps: 5, Format: RGB8
[ INFO] [1619503189.061302098]: setupPublishers...
[ INFO] [1619503189.063261523]: Expected frequency for depth = 5.00000
[ INFO] [1619503189.084820095]: Expected frequency for color = 5.00000
[ INFO] [1619503189.097839129]: Expected frequency for aligned_depth_to_color = 5.00000
[ INFO] [1619503189.111305855]: setupStreams...
[ INFO] [1619503189.111407815]: insert Depth to Stereo Module
[ INFO] [1619503189.111447757]: insert Color to RGB Camera
[ INFO] [1619503189.111620735]: SELECTED BASE:Depth, 0
[ INFO] [1619503189.111860884]: RealSense Node Is Up!
[ WARN] [1619503189.113693673]: 
 27/04 15:59:49,113 WARNING [140200667764480] (playback_device.cpp:198) Playback device does not provide a matcher
 27/04 15:59:49,151 WARNING [140200676157184] (playback_device.cpp:198) Playback device does not provide a matcher
[ INFO] [1619503192.151847262]: publishing unordered pointcloud.
[ WARN] [1619503192.951852379]: No stream match for pointcloud chosen texture Process - Color
```

I am not sure if I made a mistake or the base node doesn't support point cloud reproduction. If so, Is there any way to reproduce pointcloud from the recording? Thank you in advance.

MartyG-RealSense, 2021-04-27:
Hi @victorll998   It was stated by Doronhi the RealSense ROS developer in a past case (in the link below) that the warning **No stream match for pointcloud chosen texture Process - Color** can occur due to a frame drop.

https://github.com/IntelRealSense/realsense-ros/issues/588#issuecomment-457890773

I note in your launch log that the color stream FPS is initially set at 30 FPS but the camera configuration was decided by the launch process to be invalid and color was set to 5 FPS instead.  

I also note that your log indicates that you have a mismatch between your librealsense SDK version and ROS wrapper version.  For SDK 2.44.0, the wrapper version **2.2.24** should be used instead of 2.2.23.  This is because it is important to match the wrapper version as closely as possible with the specific librealsense version recommended in a wrapper's release notes.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.24

[image]

doronhi, 2021-04-29:
I think it is related to #1778 
Waiting for a fix in librealsense2

victorll998, 2021-05-03:
@MartyG-RealSense thank you for your responses. How careless I am to not notice the mismatch of version and now I have fixed that.

Now I also notice the fps has been reset by the launch process. Does that mean I need to know the fps and resolution too before I load the recording using ```rs_from_file```?  The default values in '''rs_from_file``` are all set to 0 which is very confusing. Shouldn't it make more sense when getting the resolution and fps from the existing video_stream_ profile?

MartyG-RealSense, 2021-05-03:
What is the full roslaunch instruction that you are using when attempting to produce a point cloud from the bag data, please?  Is it this:

**roslaunch realsense2_camera rs_from_file.launch filters:=pointcloud**

victorll998, 2021-05-03:
yes, I am using " rs_from_file.launch" filters:=pointcloud rosbag_filename:=globalfilepath

MartyG-RealSense, 2021-05-08:
Hi @victorll998  I apologize for the delay in responding further.  Do you still have the **No stream match for pointcloud chosen texture Process - Color** please?

victorll998, 2021-05-12:
@MartyG-RealSense 

I have done some more tests on Nano Tx2 and unfortunately this issue is still there. Now I have switched to intel NUC i7 to eliminate any possible frame drops. I observed that other than frame drop if the first frame in depth and color stream doesn't match in timestamp it would cause this issue too. But I only have this issue when running it on 2.2.44. Now I am not sure if this issue comes along with the version. What are your thoughts?

MartyG-RealSense, 2021-05-12:
Now that the RealSense ROS wrapper **2.3.0** for use with librealsense SDK version **2.45.0** is available, is it possible for you to test with these versions to see whether **No stream match for pointcloud chosen texture Process - Color** continues to be present, given that you only experience the problem on the previous ROS wrapper **2.2.24**

doronhi, 2021-05-12:
As far as I know, the causing issue that first appeared in librealsense2 v2.43.0, reported in #1778, is not fixed yet.

MartyG-RealSense, 2021-05-24:
Hi @victorll998   Bearing in mind the comment of Doronhi above, do you require further assistance with this case please?  Thanks!

victorll998, 2021-05-27:
Thank you @MartyG-RealSense 

I hope this issue would be fixed soon. Meanwhile, I would like to if there is any way that I could change time tag of the recording when I playback recording using sim_time.

MartyG-RealSense, 2021-05-27:
I researched your **sim_time** question.  As far as I can tell, modifying the timing of use_sim_time from its default '0' value would require use of the **/clock** topic.

http://wiki.ros.org/Clock#Using_Simulation_Time_from_the_.2Fclock_Topic

MartyG-RealSense, 2021-06-02:
Hi @victorll998  Do you require further assistance with this case, please?  Thanks!

victorll998, 2021-06-03:
@MartyG-RealSense 

Sorry for the late update. **use_sim_time** is be useful to sync topics when reading from one rosbag file. In my multi-cameras node case, I create a sudo **/clock** signal using ```frame.get_timestamp()``` from one of rosbag files and it allows all ros nodes to keep a common time tag that I want.

But there are two follow-up questions which I am eager to know the answers:

1.  Because point cloud reproduction issue merges since v2.43.0, I decided to trace back/downgrade to v2.42.0 from v2.45.0. I had a refresh install of Ubuntu 20.04 to avoid any chance of realsense packages not cleaning up. Long story short, I can replay pointcloud from rosbag files but t265 is not recognized/detected anymore on my system. Check [my latest comment in this thread](https://github.com/IntelRealSense/librealsense/issues/3361#issuecomment-853511370) more details

2. I wonder if there is any possibility that I can stop/resume the playback when using ```rs_from_file``

MartyG-RealSense, 2021-06-03:
Thanks very much @victorll998 for the update regarding your success!

In regard to your follow-up questions:

1.  I am not involved in T265 support, so I cannot provide advice on that particular subject unfortunately.

2.  You could maybe use **rosnode kill** to exit from the launched node.

https://github.com/IntelRealSense/realsense-ros/issues/1187#issuecomment-750512292

MartyG-RealSense, 2021-06-14:
Hi @victorll998  Regarding your second question **"I wonder if there is any possibility that I can stop/resume the playback when using rs_from_file"**: do you require further assistance with this please?  Thanks!

victorll998, 2021-06-14:
Hi @MartyG-RealSense Thank you for your consistent and useful help. I really appreciate that. I walked around a big circle to find out that  __rosnode kill__ is the most effective method to stop ROS nodes. Before that I tried __roslaunch python api__ but it won't let me kill ROS nodes completely. 

Currently, I'm working on building a controller of realsense. I want to develop via ROS and C++ because most of my system's processing is on ROS. What I found is that realsense ROS has only little freedom to manipulate from the lower level. For example, it is difficult to detect and extract device/sensor states. If I develop via SDK, I could miss some useful ROS functionalities. I'm seeking somewhere between the two. I wonder if you have any suggestions on that.

MartyG-RealSense, 2021-06-14:
You can obtain some information from the **/diagnostics** topic that the RealSense ROS wrapper publishes, including ASIC, projector and motion module temperature readings.

If you are able to use ROS2, the **3.1.6** version of the ROS2 wrapper in May 2021 also added an **enable** service to toggle sensors on / off.

victorll998, 2021-06-14:
@MartyG-RealSense actually, I'm thinking if the __realsense ros wrapper__ has any heartbeat-like function/topics

MartyG-RealSense, 2021-06-14:
I suppose that the current FPS speed of a particular stream would be like a heartbeat, with a relatively fixed and consistent rate when healthy and demonstrating significant variations when unhealthy.  **rostopic hz** could display the publishing rate of a particular stream topic.

http://wiki.ros.org/rostopic#rostopic_hz

Outside of the RealSense ROS wrapper, the RealSense-using **robot_devkit** project has suggested a couple of performance monitoring tools for ROS2.

https://intel.github.io/robot_devkit_doc/pages/overview.html#id6

closed 2021-06-26 by MartyG-RealSense