realsenseai/realsense-ros #1827: depth_qos parameter not applied to depth pointcloud publisher QoS

anaelle-sw, 2021-04-28:
Hello,
I found out a problem with a publisher created without taking into account the parameter `<stream_type>_qos` given to the node.

## Setup
- D435
- Ubuntu 20.04, ROS2 Rolling
- realsense-ros repo forked from branch ros2

## Problem description
I am willing realsense-ros node to publish the D435 depth pointcloud not with the default QoS, but with the reliability set to best-effort (which is more adapted to sensor data exchange). So in the launch file, I set the parameter `depth_qos` to `'SENSOR_DATA'` and pass it to the node. Here are the interesting parts of the launch file:
```c
    enable_depth = LaunchConfiguration('enable_depth', default = 'true')
    depth_qos = LaunchConfiguration('depth_qos', default = 'SENSOR_DATA')
    enable_pointcloud = LaunchConfiguration('enable_pointcloud', default = 'true')

[...]

    realsense_node = Node(
        package = 'realsense2_camera',
        executable = 'realsense2_camera_node',
        name = 'realsense_node',
        parameters = [
            {...},
            {'enable_depth': enable_depth},
            {'depth_qos': depth_qos},
            {'enable_pointcloud': enable_pointcloud},
            {...}
           ]
        )

[...]
```
But when launching the node, and doing `ros2 topic info /points -v` (topic "/points" being the topic where realsense node publishes the depth pointcloud), I found out that the topic is advertised with reliable QoS, and not best-effort.

By looking at [realsense2_camera/src/base_realsense_node.cpp](https://github.com/IntelRealSense/realsense-ros/blob/0ea1b212b5e2b5f6ce67b74270c12cba9f126e68/realsense2_camera/src/base_realsense_node.cpp#L1022), I found out that, when setting `enable_depth = true` and `enable_pointcloud = true`, the publisher for the depth pointcloud is created without using the parameter `depth_qos`, but with default QoS (with a queue of 1).

The easy way to fix this just consists in passing the custom QoS for the stream when creating the publisher:
```c
_pointcloud_publisher = _node.create_publisher<sensor_msgs::msg::PointCloud2>(
     "depth/color/points", 
      rclcpp::QoS(
           rclcpp::QoSInitialization(
                qos_string_to_qos(_qos[stream]).history,
                qos_string_to_qos(_qos[stream]).depth),
           qos_string_to_qos(_qos[stream]));
```

## Additional information
* I had a look on the merged PR about adding QoS in realsense parameters list (#1597) . The problem is that the QoS is not applied in the case of publishers created with `_node.create_publisher`, but only if there are with `image_transport::create_publisher`. The depth pointcloud publisher is created via `_node.create_publisher`.

* Is there a specific reason to not apply the `depth_qos` parameter to the pointcloud publisher? If it is not and if you think the fix given above is satisfying, I can open a pull request.

* The example of fix I gave is only for the pointcloud publisher, but I guess the same thing happens with the others streams which parameter `<stream_name>_qos` parameter is settable but not applied: ["gyro/sample"](https://github.com/IntelRealSense/realsense-ros/blob/0ea1b212b5e2b5f6ce67b74270c12cba9f126e68/realsense2_camera/src/base_realsense_node.cpp#L1038), ["accel/sample"](https://github.com/IntelRealSense/realsense-ros/blob/0ea1b212b5e2b5f6ce67b74270c12cba9f126e68/realsense2_camera/src/base_realsense_node.cpp#L1043), and ["pose/sample"](https://github.com/IntelRealSense/realsense-ros/blob/0ea1b212b5e2b5f6ce67b74270c12cba9f126e68/realsense2_camera/src/base_realsense_node.cpp#L1048) topics are published with default QoS (and queue of 100). Should these also take the custom QoS into account?

Thanks for support!

MartyG-RealSense, 2021-04-28:
Hi @anaelle-sw  The first point that should be made is that the RealSense ROS2 wrapper wrapper does not currently support ROS2 Rolling.  At the time of writing this, it supports Dashing, Eloquent and Foxy.

@doronhi the RealSense ROS wrapper developer will know more about the **depth_qos** setting than I do, so I will tag @doronhi into this case to highlight your questions.

In the meantime, would it be possible please to try setting depth_qos in a roslaunch instruction instead of inside the launch file to see whether it makes a positive difference?  For example:

**ros2 launch realsense2_camera rs_launch.py depth_qos:=SENSOR_DATA**

anaelle-sw, 2021-04-28:
Thanks for your quick answer!

> the RealSense ROS2 wrapper wrapper does not currently support ROS2 Rolling. At the time of writing this, it supports Dashing, Eloquent and Foxy.

Yes, that's right, sorry I forgot to mention this. We need to use ROS2 Rolling because of some specific features, and actually realsense-ros runs just fine with Rolling in our case. About this issue, I don't think it is due to a ROS distro mismatched, since the custom QoS is just not applied when creating the publisher [here](https://github.com/IntelRealSense/realsense-ros/blob/0ea1b212b5e2b5f6ce67b74270c12cba9f126e68/realsense2_camera/src/base_realsense_node.cpp#L1022)

> In the meantime, would it be possible please to try setting depth_qos in a roslaunch instruction instead of inside the launch file to see whether it makes a positive difference?

Yes I tried this too, but it doesn't work, the pointcloud is also published with reliable QoS policy.

MartyG-RealSense, 2021-04-28:
**SYSTEM_DEFAULT** became the default QoS setting from ROS2 wrapper 3.1.5 onwards.  Before that, the default was apparently SENSOR_DATA according to the discussion in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1703

It would be interesting to install ROS2 wrapper 3.1.4 (the one before the QoS default was changed) and see whether it makes a difference when using the standard unedited launch-file **ros2 launch realsense2_camera rs_launch.py**

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.4

As mentione above though, @doronhi is the best person to respond to your specific questions about QoS.

doronhi, 2021-04-29:
Hi @anaelle-sw 
I think that the "points" topic deserves a parameter of its own, separated from the "depth_qos" parameter, say "pc_qos". Besides that, I agree the QoS for all topics should have a QoS parameter to set them.
I would gladly accept a PR that enables modifying the QoS for the rest of the topics as suggested.
Thanks.

MartyG-RealSense, 2021-05-10:
Hi @anaelle-sw  As this case is associated with the PR that you kindlky created, I have added an **Enhancement** tag to this case to signify that it should be kept open whilst the PR is active.  Thanks!

MartyG-RealSense, 2022-03-19:
As the QoS pull request associated with this case was merged into the ROS wrapper on May 20 2021, I will close the case.  Thanks again @anaelle-sw

closed 2022-03-19 by MartyG-RealSense