realsenseai/realsense-ros #1828: D435 tf data not published. Error: Time is out of dual 32-bit range

williamx97, 2021-04-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.12.12.100 | 
| Operating System & Version |  Raspbian GNU / Linux 9 | 
| Kernel Version (Linux Only)    |  4.14.79                                        | 
| Platform                                 | Raspberry Pi 3B+ |
| SDK Version                            |  2.44.0                        | 
| Language                            |  ROS Kinetic (Using Realsense ROS Wrapper 2.2.24)                    | 
| Segment			|  Robot                       | 

### Issue Description
Hi there,

I have installed librealsense 2.44.0 and ROS Wrapper 2.2.24 however there is no tf data being published when I run ``` roslaunch realsense2_camera rs_camera.launch ``` :

``` roslaunch realsense2_camera rs_camera.launch ``` output:

<details>

<summary>Output (click to expand)</summary>

```
pi@raspberrypi:~/turtle_ws $ roslaunch realsense2_camera rs_camera.launch
... logging to /home/pi/.ros/log/e80e4d8e-a81a-11eb-920c-b827eb203c41/roslaunch-raspberrypi-14072.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:41201/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
[... 547 lines cut ...]
[ INFO] [1619612810.049815802]: insert Depth to Stereo Module
[ INFO] [1619612810.050942991]: insert Color to RGB Camera
 28/04 13:26:50,182 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1619612810.233781374]: SELECTED BASE:Depth, 0
[ INFO] [1619612810.383696761]: RealSense Node Is Up!
[ WARN] [1619612810.429232008]:
[ERROR] [1619612810.431461489]: An error has occurred during frame callback: Time is out of dual 32-bit range
[ERROR] [1619612810.491946904]: An error has occurred during frame callback: Time is out of dual 32-bit range
 28/04 13:26:50,559 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/04 13:26:50,610 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/04 13:26:50,661 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/04 13:26:50,711 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/04 13:26:50,762 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/04 13:26:50,812 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/04 13:26:50,863 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/04 13:26:50,915 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/04 13:26:50,965 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/04 13:26:51,016 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 28/04 13:26:51,067 WARNING [1757410336] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

</details>

I suspect this error may be due to the ```Time is out of dual 32-bit range``` error? Was this error preventing tf data being published or can I ignore it?

MartyG-RealSense, 2021-04-28:
Hi @williamx97  Could you try adding the instruction below to the end of your roslaunch instruction to see whether it resolves the **Time is out of dual 32-bit range** error on your Pi please?

**enable_sync:=true**

So the roslaunch instruction would look like this:

**roslaunch realsense2_camera rs_camera.launch enable_sync:=true**

williamx97, 2021-04-29:
Hey @MartyG-RealSense I've managed to fix the missing tf data issue, it turns out to be a ROS network settings issue. I am still getting the ```Time is out of dual 32-bit range ``` error, and the pointcloud appears to be missing in rviz:
[image]

The above was generated using the ```roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_sync:=true ``` command

<details> 
<summary> Console Output (Click to Expand) </summary>

```
pi@raspberrypi:~ $ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_sync:=true
... logging to /home/pi/.ros/log/47192eac-a8ac-11eb-83ac-b827eb203c41/roslaunch-raspberrypi-2391.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.89:45269/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
[... 93 lines cut ...]
[ INFO] [1619674362.578417687]: insert Depth to Stereo Module
[ INFO] [1619674362.579145966]: insert Color to RGB Camera
 29/04 06:32:42,713 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1619674362.765435532]: SELECTED BASE:Depth, 0
[ INFO] [1619674362.916985197]: RealSense Node Is Up!
[ WARN] [1619674362.959342054]: 
[ERROR] [1619674362.960490957]: An error has occurred during frame callback: Time is out of dual 32-bit range
[ERROR] [1619674363.022399788]: An error has occurred during frame callback: Time is out of dual 32-bit range
 29/04 06:32:43,090 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 29/04 06:32:43,141 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 06:32:43,191 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 06:32:43,242 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 06:32:43,294 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 06:32:43,344 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 06:32:43,395 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 29/04 06:32:43,592 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 29/04 06:32:43,643 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 06:32:43,694 WARNING [1756361760] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/04 06:32:45,288 ERROR [1670378528] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

```

</details>

MartyG-RealSense, 2021-04-29:
I see in your launch log that you have a series of occurrences of the **control_transfer returned** warning.  If these are being generated only a small number of times at launch, or at a rate of one or two per minute after launch, then it can be safely ignored.  If **control_transfer returned** is generating continuously after launch though then it can indicate that there is a serious communication problem with the device.

How often is **control_transfer returned** generating for you, please?  And does adding **initial_reset:=true** to your roslaunch instruction to reset the camera at launch make any positive difference to the results?

MartyG-RealSense, 2021-05-05:
Hi @williamx97  Do you require further assistance with this case, please?  Thanks!

williamx97, 2021-05-05:
Hey @MartyG-RealSense my sincere apologies for the late response. I've attempted the command ```roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_sync:=true initial_reset:=true``` however I get the error I describe in https://github.com/IntelRealSense/librealsense/issues/8874 

<details> 
<summary> Console Output (Click to Expand) </summary>

```
pi@raspberrypi:~ $ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_sync:=true initial_reset:=true
... logging to /home/pi/.ros/log/9db35ee0-adca-11eb-8d71-b827eb203c41/roslaunch-raspberrypi-11050.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:36223/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
[... 88 lines cut ...]
[ WARN] [1620237177.675449722]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1620237177.675908993]: The requested device with serial number 045322072234 is NOT found. Will Try again.
 05/05 18:52:57,671 ERROR [1707078688] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 05/05 18:52:57,672 ERROR [1782576160] (sensor.cpp:526) acquire_power failed: failed to set power state
 05/05 18:52:57,675 WARNING [1782576160] (rs.cpp:306) null pointer passed for argument "device"
[ INFO] [1620237183.726992089]:  
[ WARN] [1620237183.768821592]: Device 1/1 failed with exception: failed to set power state
 05/05 18:53:03,765 ERROR [1715467296] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 05/05 18:53:03,765 ERROR [1782576160] (sensor.cpp:526) acquire_power failed: failed to set power state
[ERROR] [1620237183.769294561]: The requested device with serial number 045322072234 is NOT found. Will Try again.
 05/05 18:53:03,768 WARNING [1782576160] (rs.cpp:306) null pointer passed for argument "device"
[ INFO] [1620237189.820442702]:  
 05/05 18:53:09,858 ERROR [1765798944] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[ WARN] [1620237189.862847933]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1620237189.863243453]: The requested device with serial number 045322072234 is NOT found. Will Try again.
 05/05 18:53:09,859 ERROR [1782576160] (sensor.cpp:526) acquire_power failed: failed to set power state
 05/05 18:53:09,862 WARNING [1782576160] (rs.cpp:306) null pointer passed for argument "device"

```

</details>

flipping the USB C side of the D435 seems to resolve the issue most cases. I suppose it has something to do with the source compilation flag ```-DFORCE_RSUSB_BACKEND=true ``` when I compiled ```librealsense``` ?

I'm also not getting the ```control_transfer returned``` error message after startup when running ```roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_sync:=true``` im only getting it at launch

MartyG-RealSense, 2021-05-05:
No worries about the delay in responding.  Thanks for the update!

Given that the **Failed to set power state** error occurs after reset, does removing **initial_reset:=true** from the roslaunch instruction make this error disappear?

This issue may be related to the specification of the Pi 3B model, which is more modest compared to the more powerful Pi 4 (e.g USB 2 only and 1 GB of memory on Pi 3B, compared to USB 3 and more memory on Pi 4).

If the **control_transfer returned** warnings do not occur after launch then it is safe to ignore the ones that occurred during launch.

williamx97, 2021-05-06:
If I remove ```initial_reset:=true''' from a second roslaunch command after getting the ```RS2_USB_STATUS_BUSY ```error it unfortunately does not make the error disappear unfortunately:

<details> 
<summary> Console Output (Click to Expand) </summary>

```
pi@raspberrypi:~ $ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_sync:=true initial_reset:=true
... logging to /home/pi/.ros/log/9bf60580-ae56-11eb-a92c-b827eb203c41/roslaunch-raspberrypi-1094.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:41475/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
[... 222 lines cut ...]
 06/05 11:35:39,283 ERROR [1782576160] (sensor.cpp:526) acquire_power failed: failed to set power state
 06/05 11:35:39,286 WARNING [1782576160] (rs.cpp:306) null pointer passed for argument "device"
[ INFO] [1620297345.343036950]:  
 06/05 11:35:45,383 ERROR [1749021728] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[ WARN] [1620297345.387242638]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1620297345.387638155]: The requested device with  is NOT found. Will Try again.
 06/05 11:35:45,384 ERROR [1782576160] (sensor.cpp:526) acquire_power failed: failed to set power state
 06/05 11:35:45,387 WARNING [1782576160] (rs.cpp:306) null pointer passed for argument "device"
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
pi@raspberrypi:~ $ rs-enumerate-devices
 06/05 11:35:57,877 ERROR [1940063312] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
Could not create device - failed to set power state . Check SDK logs for details
 06/05 11:35:57,878 ERROR [1996306016] (sensor.cpp:526) acquire_power failed: failed to set power state

```

</details>

Only after reversing the USB-C side of the D435 does the error disappear. However I still get the ```Time is out of dual 32-bit range``` error mentioned previously.

MartyG-RealSense, 2021-05-06:
My experience from past cases is that if **enable_sync:=true** does not resolve the **Time is out of dual 32-bit range** warning on its own then using both enable_sync:=true and initial_reset:=true in the roslaunch instruction can resolve it.  You have already been using both though.

Considering that you are using Raspberry Pi, there was once a case with this problem on a Pi that was resolved by a combination of switching from Raspbian to Ubuntu and using a stable power supply.

https://github.com/IntelRealSense/realsense-ros/issues/1178

closed 2021-05-24 by MartyG-RealSense